esp8266/Arduino #1993: How to send favicon?

digimeas, 2016-05-03:
Hi!

I'm trying to respond to a favicon.ico GET with
>String dta = F("data:image/x-icon;base64,AAAB \<snip> AAAAA");
 server.send(200, "image/x-icon",  dta.c_str() );

On my android phone over wifi, chrome says bad image.  But when I create a html file with
>\[image] AAAAA" /> 

on my desktop or the phone, chrome shows the image fine.  

I've tried various combinations of "x-icon", "png" (for png files), etc. without success. The string is 1217 bytes long, so should be sent in one chunk.

What am I doing wrong?  Thanks
Doug

liquidfalcon, 2016-05-03:
Your browser is expecting the favicon file to be served as an icon file, not as a base64 encoded blob. If you serve the same file, but send the raw binary data instead, it should load without issue.

digimeas, 2016-05-04:
Hi @liquidfalcon. Thanks for your reply.

(Among the many, many, many things) I don't understand why, if I compose this
>  <!DOCTYPE html>
\<html>
  \<head>
    \<link rel="shortcut icon" type="image/x-icon" href="data:image/x-icon;base64,AAAB \<snip> AAAA"/>
  \</head>
  \<body>
    Image?\<brk>
    \[image] AAAA"/>
    \<brk>image!
  \</body>
\</html>

and serve it as `server.send(200,  "text/html",  dta.c_str());` from my ESP, both the favicon and the \<img> are displayed correctly, but serving the favicon in base64 doesn't work.

How, in ESP-speak, do I send the "raw binary data"? Load up a uint8_t* with a SPIFF read?
Would it still be a `server.send(200, "image/x-icon", char_array)`?

Thanks
Doug

liquidfalcon, 2016-05-04:
It works in place on an HTML page because you explicitly define the image as a base64 encoded ico file, and it happily supports that, since otherwise you couldn't actually embed images in pages. However, when it requests an image directly, it isn't aware that your content is encoded differently - It will be looking for the ico image format, and when it sees anything else that isn't the ico header and data, it'll reject it. Generally when serving files that are encoded, you'd inform the browser of it's encoding type with the header `Content-Encoding: XYZ`, but, to my knowledge base64 isn't one of these supported formats, and is only supported in the unique case of embedded images within html.

To send it raw, yes, reading it into an unsigned integer array would work just fine. There's also a function within `ESP8266WebServer` called `streamFile`, which would save you piping it directly.

digimeas, 2016-05-05:
@liquidfalcon, Worked like a charm! Even a code sample in FSBrowser.ino.

I'd assumed that the browser would examine the data stream somehow, 4CC-like, but now I think I get it :-)

Thanks again
Doug

closed 2016-05-10 by igrr