esp8266/Arduino #1995: COM1 not available

raghavendramanandi, 2016-05-04:
Hello guys,

I am trying to flash my ESP8266 01 with CP2102. When I hit Upload, I am getting the below error:

_error: cannot access COM1  

error: espcomm_open failed 
error: espcomm_upload_mem failed_

When I try to execute the AT command, as soon as I select Serial Monitor in Arduino IDE, I see below error:

_Arduino: 1.6.8 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 4M (3M SPIFFS), ck, Disabled, None"

Sketch uses 219,463 bytes (21%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 31,140 bytes (38%) of dynamic memory, leaving 50,780 bytes for local variables. Maximum is 81,920 bytes.
error: cannot access COM1

error: espcomm_open failed
error: espcomm_upload_mem failed
Board at COM1 is not available

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
_

Can any one let my know what could be the problem?

Thanks,

WereCatf, 2016-05-05:
Use another COM-port, then. Should be right there down the menu.

raghavendramanandi, 2016-05-05:
[image]

Except this 2 options, I don't see anything.

TheAustrian, 2016-05-05:
If you haven't used the CP2102 before, you might need to install a driver for it.

0won, 2016-05-06:
Same issue, besides being on COM3. Hardware is ESP-01 via FTDI232 (FT232RL). Double checked all wiring, COM3 recognized as FTDI driver version 2.12.16.0 and FT_Prog shows everything's OK. I'm so far down the troubleshooting list that I'm updating Java.

igrr, 2016-05-10:
@0won Since the issue is with the IDE I'm closing this ticket. Feel free to join our gitter chat if you need help though.

closed 2016-05-10 by igrr

raghavendramanandi, 2016-05-11:
@igrr @TheAustrian @0won , I had not installed the drivers for CP2102. Its working fine now. Thanks guys!

raghavendramanandi, 2016-05-11:
I used the drivers available at http://www.silabs.com/products/mcu/Pages/USBtoUARTBridgeVCPDrivers.aspx , if anybody face the same problem.

arshidkv12, 2018-01-06:
Change permsion http://playground.arduino.cc/linux/all#Permission