esp8266/Arduino #1997: 2.2.0 Exception (3):

tablatronix, 2016-05-05:
### Basic Infos
Using wifimanager library, on credential save, I get exceptions

#### Hardware
Hardware:			nodemcu v1 12e
Core Version:      	2.2.0 stable

### Description

Problem description

### Settings in IDE

Module:  			nodemcu v1 12e
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	80Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

```C++

Decoding 49 results
0x401003e9: _umm_malloc at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1380
0x40230770: etharp_output at ?? line ?
0x40231808: ip_output_if_opt at ?? line ?
0x401004f4: calloc at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1682
0x401073cc: pvPortZalloc at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/heap.c line 33
0x40220d4d: wifi_param_save_protect_with_check at ?? line ?
0x401057d2: spi_flash_read at ?? line ?
0x40220e35: system_param_save_with_protect at ?? line ?
0x40221224: wifi_station_ap_number_set at ?? line ?
0x4022137f: wifi_station_ap_number_set at ?? line ?
0x4022140c: wifi_station_ap_number_set at ?? line ?
0x40221448: wifi_station_set_config at ?? line ?
0x40202896: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 505
0x402027b7: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 505
0x4020a200: wifi_fpm_set_wakeup_cb at ?? line ?
0x40208d41: String::getBytes(unsigned char*, unsigned int, unsigned int) const at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x402423d0: bignum_exptmod at ?? line ?
0x40208d41: String::getBytes(unsigned char*, unsigned int, unsigned int) const at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x40208d6c: String::indexOf(char, unsigned int) const at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x402423d0: bignum_exptmod at ?? line ?
0x40208d90: String::indexOf(char) const at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x402048b0: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x40209638: Print::write(unsigned char const*, unsigned int) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40208355: Print::write(char const*) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40241508: bignum_exptmod at ?? line ?
0x40208355: Print::write(char const*) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40220a9a: wifi_get_opmode at ?? line ?
0x402083a4: Print::println(__FlashStringHelper const*) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x4020276c: ESP8266WiFiSTAClass::status() at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 505
0x40206c4c: WiFiManager::connectWifi(String, String) at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 716
0x4020872c: String::changeBuffer(unsigned int) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x4020877b: String::reserve(unsigned int) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x402087ad: String::copy(char const*, unsigned int) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x402088b4: String::operator=(String const&) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x402010ae: delay at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_wiring.c line 53
0x4024153c: bignum_exptmod at ?? line ?
0x40207a4f: WiFiManager::startConfigPortal(char const*, char const*) at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 716
0x4010068c: free at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1731
0x40241560: bignum_exptmod at ?? line ?
0x40207b85: WiFiManager::autoConnect(char const*, char const*) at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 716
0x4010068c: free at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1731
0x40207bdc: WiFiManager::autoConnect() at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 716
0x40206a5f: WiFiManager::WiFiManager() at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 716
0x40202024: setup at Documents\projects\arduino\sketches\libraries\WiFiManager\examples\AutoConnectWithFeedbackLED/AutoConnectWithFeedbackLED.ino line 51
0x40201f70: configModeCallback(WiFiManager*) at Documents\projects\arduino\sketches\libraries\WiFiManager\examples\AutoConnectWithFeedbackLED/AutoConnectWithFeedbackLED.ino line 20
0x402091b4: loop_wrapper at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100718: cont_norm at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/cont.S line 109
```

tablatronix, 2016-05-05:
And testing similar code ( a PR 157 on the wifimanager repo )

```
Exception (3):
epc1=0x40100255 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40070bde depc=0x00000000

ctx: cont 
sp: 3fff0710 end: 3fff0b90 offset: 01a0
```

```C

Decoding 30 results
0x40100255: _umm_free at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1298
0x4010020c: _umm_free at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1285
0x402441c0: bignum_exptmod at ?? line ?
0x4010068c: free at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1731
0x40105b32: spi_flash_read at ?? line ?
0x401075c0: vPortFree at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/heap.c line 18
0x40221a2a: wifi_param_save_protect_with_check at ?? line ?
0x402441c0: bignum_exptmod at ?? line ?
0x40221a47: wifi_param_save_protect_with_check at ?? line ?
0x40221aad: system_param_save_with_protect at ?? line ?
0x40221e9c: wifi_station_ap_number_set at ?? line ?
0x40221ff7: wifi_station_ap_number_set at ?? line ?
0x40222084: wifi_station_ap_number_set at ?? line ?
0x402220c0: wifi_station_set_config at ?? line ?
0x40202888: ESP8266WiFiSTAClass::disconnect(bool) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 505
0x4010053d: _umm_realloc at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1489
:  (inlined by) realloc at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1707
0x4010053d: _umm_realloc at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1489
:  (inlined by) realloc at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1707
0x40221712: wifi_get_opmode at ?? line ?
0x4022253b: wifi_station_get_connect_status at ?? line ?
0x40202990: ESP8266WiFiSTAClass::status() at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 505
0x40209338: String::changeBuffer(unsigned int) at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.cpp line 519
0x402083bc: WiFiManager::startConfigPortal(char const*, char const*) at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 839
0x4010020c: _umm_free at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1285
0x4010068c: free at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1731
0x40208528: WiFiManager::startConfigPortal() at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 839
0x4020703f: WiFiManager::WiFiManager() at Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 839
0x40201fc4: setup at Documents\projects\arduino\sketches\libraries\WiFiManager\examples\ConfigOnSwitch/ConfigOnSwitch.ino line 35
0x40209d90: loop_wrapper at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100718: cont_norm at AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/cont.S line 109
```

marwijn, 2016-05-06:
I got the same exception with the program below:

```inline void handler (void)
{
  timer0_write(ESP.getCycleCount() + 500000); 
}

void setup() {
  Serial.begin(74880);
  noInterrupts();
  timer0_isr_init();
  timer0_attachInterrupt(handler);

  timer0_write(ESP.getCycleCount() + 160000);
  interrupts();

}

void loop() {
  Serial.println("Test");
  delay(1000);
}

igrr, 2016-05-06:
@marwijn you are missing `ICACHE_RAM_ATTR` attribute on the interrupt handler.

marwijn, 2016-05-06:
@igrr Thanks a lot never realised I needed it. This fixes my problem. So the original problem is probably not related.

tablatronix, 2016-05-06:
I turned debugging on full

```
*WM: Sent wifi save page
:rcl
:abort
*WM: Connecting to new AP
*WM: Connecting as wifi client...
there is no poison after the block. Expected poison address: 0x3fff3e90, actual data: 0x0 0x5 0x8d 0xfe

Panic C:\Users\alverson\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc\umm_malloc.c:851 check_poison_block

ctx: cont 
sp: 3fff1050 end: 3fff1590 offset: 01b0
```

tablatronix, 2016-05-06:
Seems like the same problems over and over #1856 #1699 #1675 

can reproduce similar problem with that https://github.com/esp8266/Arduino/issues/1699#issue-137113910 sketch, but my flash seems ok

```
Flash real id:   001640E0
Flash real size: 4194304

Flash ide  size: 4194304
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration ok.
```

tablatronix, 2016-05-06:
I had to install esptool.py again and use it for erasing flash to fix this.
Esptool.exe -ce was not doing anything

It works for a few loops then 

```
Joining network
[WIFI][APlistAdd] add SSID: myNetworkName
[WIFI] delete old wifi config...
there is no poison after the block. Expected poison address: 0x3fff1880, actual data: 0x0 0x5 0xff 0x3f

Panic C:\Users\alverson\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc\umm_malloc.c:851 check_poison_block

ctx: cont 
sp: 3fff02c0 end: 3fff0700 offset: 01b0

>>>stack>>>
3fff0470:  feefeffe feefeffe feefeffe 40100334  
3fff0480:  feefeffe 00000ca0 3ffef5f4 40100435  
3fff0490:  feefeffe feefeffe feefeffe 40100537  
3fff04a0:  feefeffe 00001000 00000484 401006f5  
3fff04b0:  00000484 00001000 000003fe 40107014  
3fff04c0:  40004b31 3fff04f0 0000001c 4021ad41  
3fff04d0:  4010571a 4021ae29 3fff13fc 000003ff  
3fff04e0:  000003fd 3fff0590 3fff13fc 000003fd  
3fff04f0:  ffffff01 55aa55aa 00000009 0000001c  
3fff0500:  0000001c 00000063 00000063 000003ff  
3fff0510:  4021b218 3fff13fc 3fff13fc 000000ff  
3fff0520:  00000001 3fff05b0 4021b373 00000008  
3fff0530:  3fff13fc 000000ff 3fff0590 00000000  
3fff0540:  3fff14bc 3fff05f1 00000001 4021b400  
3fff0550:  3fff0590 3fff13fc 3ffef618 fffffffe  
3fff0560:  3fff05b0 3fff7734 3fff13fc 3ffef618  
3fff0570:  4021b43c 00000000 00000000 6c6f2065  
3fff0580:  402024ec 63206966 69666e6f 2e2e2e67  
3fff0590:  3fff0000 40106e68 3ffef618 00000008  
3fff05a0:  3ffef618 00000029 3fff05f9 402036d0  
3fff05b0:  3fff0500 3fff05c0 00000008 0000002a  
3fff05c0:  3ffef618 00000008 60000000 0000007e  
3fff05d0:  6000001c 0000002b 3ffef4b4 00400000  
3fff05e0:  3ffe8780 53532064 3fff05c0 3fff05f0  
3fff05f0:  3fff088c 402025b0 fe000a65 fffffffe  
3fff0600:  3ffef618 3ffef4b4 00000000 40202283  
3fff0610:  3fff0650 3fff0620 00000008 401006d8  
3fff0620:  4f494420 3030000a 3fff083c 3fff13c4  
3fff0630:  3fff0670 3fff083c 00000018 4020238b  
3fff0640:  3ffe8874 3ffe84b4 3fff0620 3fff0650  
3fff0650:  3fff0690 0000000e 00000009 3fff083c  
3fff0660:  3ffef4a8 3ffe84b4 3ffef618 4020249d  
3fff0670:  3fff083c 3fff13c4 3ffef4a8 40202f61  
3fff0680:  3ffe8494 3ffef5f8 3ffef618 3ffef5f8  
3fff0690:  00400000 3ffef4a8 3ffef618 3ffef5f8  
3fff06a0:  00400000 3ffef4a8 3ffef618 40201d37  
3fff06b0:  feefeffe feefeffe feefeffe feefeffe  
3fff06c0:  00000002 feefeffe feefeffe feefeffe  
3fff06d0:  feefeffe feefeffe feefeffe 3ffef6d4  
3fff06e0:  3fffdad0 00000000 3ffef6cc 40203553  
3fff06f0:  feefeffe feefeffe 3ffef6e0 40100958  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
```

igrr, 2016-05-07:
Reproduced this, thanks for detailed report. Looks like an out-of-bound write in SDK code. Will dig deeper on Monday.

kentaylor, 2016-05-08:
@tablatronix tested WifiManager, found it to fail, provided the trace and @igrr was able to reproduce it. 

I believe this is the same problem that led me to finding WiFiManager flaky and which further led to 
 to my "PR 157 on the wifimanager repo". I believe https://github.com/tzapu/WiFiManager/pull/157 fixes the problem. I always thought there was an underlying issue in the Espressif SDK but assumed Espressif would be unlikely to fix it. It is fascinating to see that @igrr is now interested in fixing this on behalf of Espressif. I further think the issue in the Espressif SDK is subtle and it can also be fixed for Arduino without modifying the Espressif SDK by modifying the ESP8266WiFiSTAClass::begin method.

@tablatronix provides a trace of my version failing as well but the memory corruption could have occurred when running tzapu/WiFiManager and still have been there when later testing with https://github.com/kentaylor/WiFiManager/. I haven't had my version fail during testing and the fix I think was calling WiFi.disconnect(true) before calling WiFi.begin(new SSID, new password) as shown at https://github.com/kentaylor/WiFiManager/commit/3bb5fad6534b30e0a1f4e6138c746c828d4d820a#diff-bfe0a2d15de63f0ead6d59ff54dd518eR241 .

I found that calling Wifi.begin(existing SSID, existing password) never failed but calling WiFi.begin(new SSID, new password) or WiFi.begin() will occasionally cause the ESP8266 to fail to connect to WiFi until rebooted and rarely put it into a state that requires reflashing to fix the corrupted memory. Usually rebooting was enough. It was never repeatable for me but the probability of failure did seem to depend on where it was up to in connecting to the access point.

Most people hard code Wifi.begin(SSID, password) into their sketch so are effectively calling Wifi.begin(existing SSID, existing password) which never causes a problem. In this case the Espressif SDK functions wifi_station_set_config, wifi_station_set_config_current and wifi_station_connect are not called. See https://github.com/esp8266/Arduino/blob/e82b74eab2f04a065ac97b678669bb81a2cdd79f/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L135 . So the problem must be associated with calling these functions.

The ESP8266 will autoconnect to WiFi at startup using the stored SSID and password. Currently if Wifi.begin(new SSID, new password) is called while already connecting to WiFi then wifi_station_set_config or wifi_station_set_config_current will be called at an indeterminate time in the WiFi connection sequence. Wifi.begin(new SSID, new password) and WiFi.begin() call the Espressif SDK function wifi_station_connect() when it is already connecting. Mostly it doesn't matter but it seems that if wifi_station_connect() is called when it is at a particular point in the WiFi connection process it causes problems. 

Therefore the solution for Arduino is to call wifi_station_disconnect() prior to calling wifi_station_set_config or wifi_station_set_config_current and to not call wifi_station_connect() when it is already connecting to WiFi.

The problem is occurring because wifi_station_connect() is being called when it is not required and can be fixed in Arduino by only calling wifi_station_connect() when required. However, the Espressif SDK should also be modified so that calling wifi_station_connect() is never harmful even if called when not required.

tablatronix, 2016-05-08:
Another issue I referenced above had succesfully workaround by setting persistant to false avoiding the saves i am assuming.

kentaylor, 2016-05-09:
@tablatronix Yes, if persistent is false wifi_station_set_config_current is called which doesn't write data to flash according to @igrr in #1054. If persistent is true wifi_station_set_config is called which does write data to flash. So it seems setting persistent to false is likely to avoid corrupting the flash.

But WiFi also fails sometimes for me when calling WiFi.begin() and persistent is irrelevant in that case. WiFi.begin() calls the Espressif SDK function wifi_station_connect() see https://github.com/esp8266/Arduino/blob/e82b74eab2f04a065ac97b678669bb81a2cdd79f/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L181 and wifi_station_dhcpc_start(). I think calling these functions when it is already automatically connecting to WiFi can be harmful.

igrr, 2016-05-10:
Okay, the issue in SDK code has been found and will be fixed in the next release. In the meantime I will push a workaround which prevents buffer overrun.

closed 2016-06-23 by igrr

overtone1000, 2016-07-27:
I encountered this bug using version 2.3.0 of the esp8266 core with the Eclipse Arduino Plugin on an ESP-12 module (via Adafruit). I could not reproduce it on a Sparkfun Thing Dev. I noted the error when calling the `ESP8266WiFiMulti::run()` function, and the debug output was exactly as previously described. I was also able to elicit the error by calling the `WiFi::setOutputPower()` method.

I implemented the `WiFi.persistent(false)` workaround described by @tablatronix and @kentaylor, which resolved the issue. The sketch now runs on both devices, and all WiFi behavior is normal in both station and access point modes. Thanks for your help!

engrziarehman, 2016-11-14:
I am encountering this problem for my IoT project, there are eight different ESP modules connected with the same router and an mqtt broker. All of the ESP modules starts in station mode then connect with wifi and mqtt broker successfully, but after a while every ESP module get Exception (3) simultaneously, I have tried various things, to resolve this issue but no use. I am using sdk 1.5.4. I have ESP-01 and ESP-12 modules, each behave the same. I am using ESP8266WiFi, pubsubclient and Ticker libraries. 
My firmware is an MQTT to serial brigde, following is my example code. 

void setup(void){
	Serial.begin(SERIAL_BAUDRATE);
	Serial.setDebugOutput(true);

	WiFi.persistent(false);  //it was suggested in issue# 1997, but it didnot worked
	//WiFi.setAutoConnect(false);  //commented because esp was unable to reconnect after exception.
	//WiFi.setAutoReconnect(false); //commented because esp was unable to reconnect after exception.
	WiFi.printDiag(Serial);
	WiFi.mode(WIFI_STA);

	//mqtt client configuration
	mqttclient.setServer(MQTT_SERVER, MQTT_PORT);
	mqttclient.setCallback(_mqttIn_Port_Out);
}
void loop(void) {
        //if any one of them wifi or mqtt is not connected, keep trying.
	while(!WiFi.isConnected() || !mqttclient.connected()){
                if(!WiFi.isConnected(){
		         WiFi.disconnect(true);
			_connect_wifi();
		}
		if(!mqttclient.connected()) {
			ETS_GPIO_INTR_DISABLE();	//to prevent esp to get watchdog trigger.
			_connect_mqtt();
			ETS_GPIO_INTR_ENABLE();	
		}
	}
	mqttclient.loop();
}

kentaylor, 2016-11-15:
I guess they are all suffering a watchdog timer timeout in the same loop when they are trying to deal with a failed WiFi or MQTT broker or similar. It is not related to the issue in this thread. 

Looking briefly at your code though I can see some misunderstanding. The ESP will do it's best to keep the WiFi connection going in the background. It will reconnect if a connection is lost and reconnect if the WiFi went away but has now come back. You can't improve on this and if you try in user code you are likely to stuff it up. Don't call WiFi.persistent(false), don't call WiFi.disconnect(true) and especially don't call WiFi.setAutoConnect(false). WiFi.setAutoConnect(true) is the default state and will stay that way unless you've ever set it otherwise. The setting is persistent across reboots so if you ever set it false it will stay that way too.

The best you can do with the WiFi connection, once it's been properly configured, is check it's state and act accordingly.

engrziarehman, 2016-11-16:
Thank you for your reply, I have made some changes. persistence, autoconnect and reconnect enabled, and manual connection WiFi.begin is being called as:
	if(WiFi.SSID() != MY_SSID){
		WiFi.begin(MY_SSID, MY_PASSWORD);
	}
Apparently the problem is resolved, i have not faced exception (3) any more.

kentaylor, 2016-11-17:
Glad it was sorted. Just an extra point of no practical value
if(WiFi.SSID() != MY_SSID){
WiFi.begin(MY_SSID, MY_PASSWORD);
}

is a repeat of the logic in the WiFi.begin function. It does the (WiFi.SSID() != MY_SSID) logic in the function so if you leave it out, it will work the same.

tablatronix, 2016-11-17:
You gotta be careful with while loops, not all functions do implcit yields. If you dont yield wdt reset will occur.

engrziarehman, 2016-11-23:
@kentaylor, after two days without facing exception (3) unfortunately i have started facing problem again, all of my ESP modules gets exception at the same time, few of them were newly programmed i.e. persistence, autoconnect and reconnect were true as per your suggestion, while rest of them had old program i.e. shown above. but the exception behaviour is the same for every module. I have enable WiFi+Core debug output, that gives:
Mode: STA
PHY mode: N
Channel: 11
AP id: 0
Status: 1
Auto connect: 1
SSID (8): myRouter
Passphrase (10): aaaaa12345
BSSID set: 0
Connecting to
myRouter
sta config unchangedscandone
f 0, 
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 13
cnt 
............state: 5 -> 2 (fc0)
rm 0
wifi evt: 1
STA disconnect: 15
..reconnect
state: 2 -> 0 (0)
f r0, scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 13
cnt 

connected with myRouter, channel 11
dhcp client start...
wifi evt: 0
.ip:192.168.15.242,mask:255.255.255.0,gw:192.168.15.1
wifi evt: 3
.
Attempting MQTT connection...
[hostByName] Host: 192.168.15.149 is a IP!
:ref 1
:sent 57
:rn 4
:c0 1, 4
MQTT connected
:sent 56
:sent 22
:rn 5
:c0 1, 5
pm open,type:2 0
:sent 2
:rn 2
:c0 1, 2
:sent 2
:rn 2
:c0 1, 2
:sent 2
:rn 2
:c0 1, 2
:sent 2
:rn 2
:c0 1, 2
:sent 2
:rn 2
:c0 1, 2
:sent 2
:rn 2
:c0 1, 2
:sent 2
:rn 2
:c0 1, 2
:sent 2
:rn 2
:c0 1, 2
Fatal exception 3(LoadStoreErrorCause):
epc1=0x40213c5f, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40000003, depc=0x00000000

Exception (3):
epc1=0x40213c5f epc2=0x00000000 epc3=0x00000000 excvaddr=0x40000003 depc=0x00000000

ctx: sys 
sp: 3ffffc60 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe00:  3ffe9a25 40104417 3ffed4b0 4010113a  
3ffffe10:  40101e00 0000ac0f 40216ad7 3ffffed0  
3ffffe20:  3ffee4b0 3ffeded8 3ffee5d0 00000014  
3ffffe30:  40216eed 3ffffed0 3ffee000 3ffeded8  
3ffffe40:  3ffec250 00000001 40216f39 3ffffed0  
3ffffe50:  7fffffff 3ffea2dc 3ffea2dc 00000001  
3ffffe60:  00000001 3ffea2c4 3ffea2c4 00000001  
3ffffe70:  40213709 5c089eec 00000000 4000050c  
3ffffe80:  3fffc278 3ffec250 ffffffb2 3ffec260  
3ffffe90:  3ffed4b0 00000012 00000000 40217c98  
3ffffea0:  00000000 3fff12f4 ffffffb2 00000000  
3ffffeb0:  00000000 3ffef7e0 00000001 00000000  
3ffffec0:  3fffdab0 00000000 3fffd9d0 3ffef7cc  
3ffffed0:  00000000 00310b0b 00640100 00000055  
3ffffee0:  3ffec274 00000067 3ffec299 3ffec268  
3ffffef0:  3ffec291 3ffec274 3ffec27e 3ffec28b  
3fffff00:  00000000 3ffec2a4 3ffec2bc 00000000  
3fffff10:  00000000 00000000 00000000 00000000  
3fffff20:  00000000 00000000 00000012 00000000  
3fffff30:  3fff16c4 40217763 3ffed4b0 3fff12f4  
3fffff40:  00000000 3ffee5d0 3ffed4b0 3ffec250  
3fffff50:  3ffec250 0000008f 00000000 00000012  
3fffff60:  00000000 0000008f 4020c9e3 3ffed4b0  
3fffff70:  3ffec244 3fffdcc0 3ffe9ae0 3ffe9ae0  
3fffff80:  00000080 3ffed4b0 00000000 3ffef7e0  
3fffff90:  4020c2ff 3fffdab0 00000000 40204653  
3fffffa0:  3ffe9ae0 40000f49 3fffdab0 40000f49  
<<<stack<<<

kentaylor, 2016-11-25:
@engrziarehman your last post looks like a buffer overrun, write outside array bounds or similar to me. Try searching on Fatal exception 3(LoadStoreErrorCause). More importantly it is an application issue and unrelated to bugs in the Espressif library so I'd prefer you sought help in a different thread.

Ttech, 2016-11-29:
@kentaylor I don't mean to hijack this bug report, but you said above checking / reconnecting to Wifi wears out flash. Could you explain? I haven't found a source of why this is but I am very curious. Thank you.

engrziarehman, 2016-11-29:
I have used [WiFiClient](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiClient/WiFiClient.ino) Example code, I only changed SSID and Password, and I am still facing exception (3), along with previously installed ESPs with my own code, all of them are facing exception at the same time, and after tracing stack dump i have found that each esp with different firmware having exception at the same disassembly line, inside <ieee80211_alloc_challenge> i.e.:

402136fc <ieee80211_parse_beacon>:
402136fc:	c0c112        	addi	a1, a1, -64
402136ff:	066142        	s32i	a4, a1, 24
...
...
...
40213c8c:	320c      	movi.n	a2, 3
40213c8e:	f00d      	ret.n
40213c90:	ffdf31        	l32r	a3, 40213c0c <ieee80211_parse_beacon+0x510>
**40213c93:	030262        	l8ui	a6, a2, 3		<<<--EXCEPTION ADRESS**
40213c96:	010252        	l8ui	a5, a2, 1

kentaylor, 2016-11-29:
@Ttech, I did not say:

> checking / reconnecting to Wifi wears out flash

That would do no harm. WiFi Connection parameters are stored in a particular flash location. Calling WiFi.disconnect(true) will erase that location, then calling WiFi.begin(SSID,password) will write them there again. Doing that in a tight loop could easily end up with 100s of thousands of flash writes at the same location. Once the correct parameters are stored in flash the ESP will always do it's best to stay connected to WiFi. No need to call WiFi.begin and there is nothing you can do in user code to make connecting to WiFi any more likely. Better to just test if you are connected. Calling WiFi.begin(SSID, password) will not rewrite the flash when the values SSID and password are the same as those already stored, so there is no risk of wearing out the flash by doing that. Calling WiFi.begin() does not write to flash and I expect the bug is only seen in flash writes.

The bug in the Espressif library discussed in this thread has been explained by @igrr as a [buffer overrun](https://github.com/esp8266/Arduino/issues/1997#issuecomment-218077839). I know that this bug has lots of different symptoms. I see different symptoms reported and have had different behaviour myself, including bricking where WiFi.begin(SSID, password) no longer works at all. The buffer overrun will have the effect of writing different incorrect stuff to flash in different circumstances and I expect the differing symptoms depend on what was actually written in that instance. Once the flash is corrupted the only way to fix it is a flash erasure, uploading a new sketch does not erase that area of flash. There are a variety of ways to erase flash. I use [this sketch](https://github.com/kentaylor/EraseEsp8266Flash) which I always run when I get mysterious failures. Oddly, when you see people describing symptoms consistent with corrupted flash and suggest trying a flash erasure they usually ignore you. @engrziarehman should try flash erasure as the [last post](https://github.com/esp8266/Arduino/issues/1997#issuecomment-263609264) makes me think corrupted flash is a possibility.

It fascinates me that this devastating bug can have existed in the Espressif library for so long without it being detected previously. I also wonder how many ESPs are in commercial products. If you were selling ESP light bulbs, you would be getting random failure rates that would lead to reputational damage that I would have thought  would result in commercial failure. As far as I'm aware the bug is still in the Espressif library and @igrr has done

> a workaround which prevents buffer overrun.

when using the Arduino library. 

I'm still seeing [symptoms reported](https://github.com/tzapu/WiFiManager/issues/247) that are consistent with the bug but I don't know if they are using the  Arduino library that has the workaround.

Ttech, 2016-12-01:
@kentaylor Thank you so much for this detailed explanation. I actually found this because I was getting those errors when doing MQTT.connect(). I'll try the flash erasure to see if that fixes it as well. 

The reason I asked about connecting and disconnecting wifi was to try to provide lower power - only connect when I need to transmit data to the data logger. That idea seems problematic now though.

kentaylor, 2016-12-01:
@Ttech, disconnecting from WiFi is fine and reduces power. Call WiFi.disconnect(false) which doesn't erase flash, then WiFi.begin() when you want to reconnect. Deep sleep will reduce power consumption much more. Search around for explanations of how to do it. 

You can also avoid using the flash store by calling WiFi.persistent(false) prior to calling WiFi.begin(SSID,password). 

I'm wondering if corrupted flash is commonly associated with Exception (3). It looks likely to be the underlying [problem here](http://blog.flynnmetrics.com/uncategorized/esp8266-exception-3/).

engrziarehman, 2016-12-01:
@kentaylor  Thank you for your reply. My problem is not related with flash corruption, why! let me explain:
1- I have tried [erasing flash](https://github.com/kentaylor/EraseEsp8266Flash) but it didn't work.
2- I have various ESPs connected with the same router, each work fine but when the time of exception comes they all get together. it is very unlikely that each module get flash corruption, some of them are newly programmed for the first time with the WiFiClient example code still they are getting exception.
3- the exception is not at the time of WiFi.begin(ssid,password), it works but after few mins, or some time few hours, the exception appears. 
4- the disassembly shows that the problem is in the <ieee80211_parse_beacon> function, that sometime unable to parse WiFi beacon, and possibly it is a bug in sdk.

kentaylor, 2016-12-01:
@engrziarehman As 

> some of them are newly programmed for the first time with the WiFiClient example code still they are getting exception.

it is not flash wear and not flash corruption and because it also occurs with an example app, not an error specific to your application so I'd agree 

> possibly it is a bug in sdk

but it is a different bug to that identified by @igrr in this closed issue. I'd try opening a new issue in your case to try and get attention from @igrr or someone else from Espressif. 

It seems unusual though and while it may be a bug in the sdk it might only show up when the router does something odd so you could try a different router too.

tablatronix, 2016-12-01:
Create a new issue, enable debug output , make a minimal proof sketch, , post exception.

Also learn how to use gdb exception debugger there is even one you can add to the arduino ide.

engrziarehman, 2016-12-08:
@kentaylor I have changed the router, yes the problem is solved and it was observed on wifi analyser that router was missing its beacon at some time, this might be the reason i was getting all of the ESPs exception at the same time, and they are fine with a new router. but my question is still there that no matter what my router is doing odd, it should not have get the exception at <ieee80211_parse_beacon> function, this should be corrected in sdk.

tablatronix, 2016-12-08:
That is bizarre, what kind of router?

kentaylor, 2016-12-08:
A router issue seems plausible to me. The ESP has to parse a lot of [beacon fields](https://en.wikipedia.org/wiki/Beacon_frame#Components) and there could be an error in the parser.  Wikipedia [suggests a possibility](https://en.wikipedia.org/wiki/Service_set_(802.11_network)#Service_set_identifier_.28SSID.29) "A common, albeit incorrect assumption, is that an SSID is a string of human-readable characters (such as ASCII), terminated by a NUL character (as in a C-string)." 

The fix for corrupted flash is to fill it with NULL characters, so perhaps the failure to read correctly from corrupted flash is an expectation of a NULL terminated string and the absence of a NULL terminated string in a beacon could cause a similar problem. As the @engrziarehman devices only fail occasionally and all fail at the same time there would have to be some changed circumstance that caused odd beacons. That would be unusual, but they might not even be from the connected router as WiFi devices are supposed to scan for all visible beacons. Perhaps switching on another router would be enough.

The fact that corrupted flash can't be fixed by calling the function to delete WiFi credentials indicates the function that deletes WiFi credentials from flash is also erroneous. It ought to write NULLs to the whole area of flash used for storing credentials so there is more than one error in the Espressif library. Maybe the same author who relies on NULL termination of strings when he/she shouldn't, wrote the beacon parsing and the flash reading and writing code.

tablatronix, 2016-12-08:
It cannot be fixed with a sketch, you have to use esptool.py or esptool.exe ( I had no luck with the binary erasing for some reason) to erase the flash. Or you can use one of the GUI flashers.

It is a an out of bound write, or buffer overflow essentially, so the corruption has nothing to do with wifi credentials probably. But I have not actually looked at the fix to see.

kentaylor, 2016-12-08:
This [sketch](https://github.com/kentaylor/EraseEsp8266Flash) fixed my bricked devices. It writes NULL to the whole flash including wiping itself from flash, so it can only run once after uploading. I wiped the whole of flash because I didn't know where the WiFi credentials are stored. If I knew that I could read out that area and look at the nature of the corruption.

I'm confused where @tablatronix says "so the corruption has nothing to do with wifi credentials flash probably". It must be corrupted flash because it persists over rebooting and uploading new sketches. I understood the effect of the buffer overrun to be writing stuff to flash from memory that is beyond the intended memory buffer. This can happen when you rely on identifying the end of a string by a NULL character and that NULL character has been overwritten. I guess the end of the SSID string and password string in flash is identified by a NULL character and the absence of that character means extra stuff is read in when loading the WiFi credentials from flash. When we overwrite the whole flash with NULLs we put that terminating NULL character back. 

This post and the [previous](https://github.com/esp8266/Arduino/issues/1997#issuecomment-265851391) contains a lot of speculation so I will not be surprised if someone points out where they are wrong.

tablatronix, 2016-12-09:
You said the wifi credentials delete function, i did not realize there was a full flash erase sketch.

kentaylor, 2016-12-13:
@engrziarehman you might be interestred in this from the [Espressif SDK 1.4](http://bbs.espressif.com/viewtopic.php?f=46&t=2198) 

> Resolved Issues(Bugs below are eligible for Bug Bounty Program):
...
The ESP8266 maybe reset if connected to a special router.

It [isn't used](https://github.com/esp8266/Arduino/issues/2138) by Arduino yet.

engrziarehman, 2016-12-20:
Thank you for your reply! I had been thinking that this version of sdk is incorporated in arduino, it shows [1.5.4_16_05_20](https://github.com/esp8266/Arduino/blob/master/tools/sdk/version). If it is not, then yes! it may be!

tablatronix, 2017-02-08:
oddly enough I am still getting a similar issue.

```
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
Decoding 13 results
0x401003c8: check_poison_block at C:\Users\USER\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 834
0x401003c8: check_poison_block at C:\Users\USER\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 834
0x40209881: Print::write(char const*) at C:\Users\USER\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 188
0x4010047e: get_unpoisoned at C:\Users\USER\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 938
0x401008d4: free at C:\Users\USER\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1720
0x4020776d: WiFiManager::~WiFiManager() at C:\Users\USER\Documents\projects\arduino\sketches\libraries\WiFiManager/WiFiManager.cpp line 891
0x40201f9d: handlewifimanager(bool) at C:\Users\USER\Documents\projects\arduino\sketches\AutoConnectWithTimeout/AutoConnectWithTimeout.ino line 111
0x40201f12: espInfo(Stream&) at C:\Users\USER\Documents\projects\arduino\sketches\AutoConnectWithTimeout/AutoConnectWithTimeout.ino line 111
0x40202002: setup at C:\Users\USER\Documents\projects\arduino\sketches\AutoConnectWithTimeout/AutoConnectWithTimeout.ino line 111
0x4020a9d7: loop_wrapper at C:\Users\USER\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100970: cont_norm at C:\Users\USER\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109
```

tablatronix, 2017-02-12:
@igrr I noticed this workaround is still in master , was this not fixed in SDK yet?

tablatronix, 2017-11-03:
@igrr bump
FYI, this workaround is still in code in RC

```
            // workaround for #1997: make sure the value of ap_number is updated and written to flash
            // to be removed after SDK update
            wifi_station_ap_number_set(2);
            wifi_station_ap_number_set(1);
```

shivamkeshri, 2017-12-12:
If you are getting fatal error exception like this:

Exception (3):
epc1=0x401003e9 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4000cbd9 depc=0x00000000
In infinite loop in your serial monitor of arduino IDE  .

then goto this link download the software and follow the procedure and erase the flash memory to solve the error.

This does not solve fatal error that occurs due to your program but in case your device goes in such condition that it can’t be able to access program memory then it will work and try atleast one time to solve the problem.

This is the procedure to hard reset the nodemcu  

( https://www.youtube.com/watch?v=MHrm7axsImI&t=146s  )

Step :
Install latest python version in you pc.(https://www.python.org/downloads )
Open cmd prompt as administrator .
Go to c/program files or program files (x86)->python (your version)->Script. For this type (cd c/program files (x86)/python(your version)/Script) then press enter .
Now type (pip install esptool).
Now download ESPlorer  ( https://esp8266.ru/esplorer/ ) version(Download ESPlorer.zip (v 0.2.0-rc6)) and extract the file and open executable jar file .
Now goto nodemcu firmware site (https://github.com/nodemcu/nodemcu-firmware/releases ) and from  download file (nodemcu_float_0.9.6-dev_20150704.bin ) and copy this file into the c/program files (x86)/python(your version)/Script folder . 
Now in cmd prompt just type.
	esptool.py --port COM(your port no.) --baud 115200 erase_flash
And press enter.

Note : you can see your port no. into the device manager .

rippe, 2018-01-05:
@igrr , I saw a comment you made more than a year ago about (!!!!) about ICACHE_RAM_ATTR. I must say, I don't understan what to do with it or where to place it :-(

I need som assistance for sure :-)

My code below is quite a dirty one, and beside the crash problem I would like to put as muck of the web server code in the setup to aviod lagging the processor. The connection to the Wemos is made one time upon a start/restart. 

If I didn't give all the needed information, don't hestita to ask.

`// Including the ESP8266 WiFi library
#include "HLW8012.h"
#include <ESP8266WiFi.h>
//#include <OneWire.h>
//#include <DallasTemperature.h>

#include <TimeLib.h>
#include <SimpleTimer.h>

// Replace with your network details
const char* ssid = "xxxxx";
const char* password = "xxxxx";

// Web Server on port 80
WiFiServer server(80);

//OLED
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <SFE_MicroOLED.h>  // Include the SFE_MicroOLED library
#define NUMFLAKES 10
#define XPOS 0
#define YPOS 1
#define DELTAY 2
#define LOGO16_GLCD_HEIGHT 16
#define LOGO16_GLCD_WIDTH  16
#define PIN_RESET 255  //
#define DC_JUMPER 0  // I2C Addres: 0 - 0x3C, 1 - 0x3D
int R[3] = {0};
int xPos[3] ={34,34,34};
int tempXpos, levelXpos, DiagXpos;

MicroOLED oled(PIN_RESET, DC_JUMPER);  // I2C Example

const int interruptPin = D7;                         // the number of the interrupt pin
volatile int pwmValue = 0;

volatile unsigned long startTime = 0;               // timer for interruptpin 
volatile unsigned long stopTime = 0;                // timer for interruptpin  
volatile unsigned long connectionTime = 0;          // timer for the overall signal  

int countISR = 0;                                   
volatile unsigned int msTime[3] = {22,33,44};              

bool startUp = false;
bool fallingEdge = false;

unsigned int temperature;                           // 22 - 88 ms
unsigned int level;                                 // 22 - 88 ms
unsigned int diagnostic;                            // 22 - 88 ms
int level_mm;                // 20 - 80 mm
unsigned int periodTimeRecieved = 0;
unsigned int periodTime = 110;

//int LEDone = 31;
//int LEDtwo = 33;

/float levelRange[15] = {31.65,33.81,35.97,38.14,40.30,42.46,44.62,46.78,48.95,51.11,53.27,55.43,57.59,59.76};   //Level in [ms]
float levelRange[7] = {31.65,35.99,40.33,44.67,49.01,53.35,59.69};
float levelRange_mm[7] = {20,30,40,50,60,70,80}; 

[... 363 lines cut ...]
  oled.println("Diag ");
  oled.print(T[2]);
  oled.display();*/
}

void ReWriteScreen()
{
  //Put up the headlines Tmp, Lev and Dia on the display
  oled.clear(PAGE);
  oled.clear(ALL);
  oled.setFontType(1);
  //oled.setTextColor(WHITE);
  oled.setCursor(0,0);
  oled.println("Tmp ");
  //oled.print(T[0]);
  oled.setCursor(0,18);
  oled.println("Lev  ");
  //oled.print(T[1]);
  oled.setCursor(0,36);
  oled.println("Dia ");
  //oled.print(T[2]);
  oled.display();
}`

Best regards Richard

rippe, 2018-01-05:
Hi,

I just read somewhere to test the `xtensa-lx106-elf-addr2line -pfia -e path/to/build/directory/your-sketch.elf 0x4000bf1b.` The result is strangely enough:
'0x4000bf1b: ?? ??:0'

I must be doing this wrong...?
Br, Richard

devyte, 2018-01-05:
@rippe void Rising() => ICACHE_RAM_ATTR void Rising()
This is documented in the docs.
Please refer such questions to a community forum.

rippe, 2018-01-05:
@devyte, ooops, I'm so sorry... Don't use to post questions at all. Will do this in the future. 
Thank you for the answer anyway (much more basic than I thought and what I could google).

Best regards Richard

dontsovcmc, 2018-07-12:
I got exception when connecting to Wi-Fi that turned off =(

```
0x40213724: __ssputs_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfprintf.c line 180
0x4020fae0: _printf_i at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfprintf_i.c line 241
0x401004f4: calloc at ?? line ?
0x40106b98: pvPortZalloc at ?? line ?
0x4010577c: spi_flash_read at ?? line ?
0x40235d55: wifi_param_save_protect_with_check at ?? line ?
0x401063f1: __wrap_spi_flash_read at ?? line ?
0x40235e3f: system_param_save_with_protect at ?? line ?
0x40235e22: system_param_save_with_protect at ?? line ?
0x4023630c: wifi_station_ap_number_set at ?? line ?
0x401063f1: __wrap_spi_flash_read at ?? line ?
0x40236526: wifi_station_ap_number_set at ?? line ?
0x402365d6: wifi_station_ap_number_set at ?? line ?
0x4024d530: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x402365a5: wifi_station_ap_number_set at ?? line ?
0x4023660c: wifi_station_set_config at ?? line ?
0x402068e0: ESP8266WiFiSTAClass::disconnect(bool) at ?? line ?
0x40213700: __any_on at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/mprec.c line 1039
0x4020fae0: _printf_i at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfprintf_i.c line 241
0x4020eab8: Print::write(unsigned char const*, unsigned int) at ?? line ?
0x40209f39: Print::write(char const*) at ?? line ?
0x40209f4c: Print::print(char const*) at ?? line ?
0x4020562e: setup_ap(Settings&, SlaveData const&, float const&, float const&) at ?? line ?
0x40209f39: Print::write(char const*) at ?? line ?
0x4020eab8: Print::write(unsigned char const*, unsigned int) at ?? line ?
0x40209f39: Print::write(char const*) at ?? line ?
0x4020a075: Print::printNumber(unsigned long, unsigned char) at ?? line ?
0x40210248: sprintf at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/sprintf.c line 646
0x4020eab8: Print::write(unsigned char const*, unsigned int) at ?? line ?
0x4020eab8: Print::write(unsigned char const*, unsigned int) at ?? line ?
0x4020a0d3: Print::print(long, int) at ?? line ?
0x4020eab8: Print::write(unsigned char const*, unsigned int) at ?? line ?
0x40209f39: Print::write(char const*) at ?? line ?
0x40209f4c: Print::print(char const*) at ?? line ?
0x40202ebc: calculate_values(Settings&, SlaveData&, float*, float*) at ?? line ?
0x40205384: MasterI2C::getSlaveData(SlaveData&) at ?? line ?
0x40240000: etharp_output at netif/etharp.c line 950 (discriminator 1)
0x40209bdc: EspClass::wdtDisable() at ?? line ?
0x40204928: loop at ?? line ?
0x4020b048: loop_wrapper() at core_esp8266_main.cpp line ?
0x40100700: cont_norm at cont.S.o line ?
```

PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
SYSTEM: ESP8266 80MHz 80KB RAM (1MB Flash)
WifiManager 0.12

seinecle, 2018-11-07:
For those still struggling with this issue in 2018, this did the trick:

WiFi.persistent(false);
WiFi.disconnect(true);
WiFi.begin(ssid, password);

source: http://blog.flynnmetrics.com/uncategorized/esp8266-exception-3/

ta4ukoma, 2019-08-06:
> For those still struggling with this issue in 2018, this did the trick:
> 
> WiFi.persistent(false);
> WiFi.disconnect(true);
> WiFi.begin(ssid, password);
> 
> source: http://blog.flynnmetrics.com/uncategorized/esp8266-exception-3/

THX! thank you so much! That's working!

tablatronix, 2019-08-07:
Wow this is old, Probably need to look at this again

dahlheim2, 2019-08-22:
old yes, but helped me!  Thanks, seinecle!

dzid26, 2019-10-08:
> WiFi.persistent(false);
>  WiFi.disconnect(true);
> WiFi.begin(ssid, password);
But that means no WiFi manager :/ 

It is strange. I have bare ESP8266MOD and ESP-12F (Huzzah dev board) and first one works and the other one doesn't. And it worked before (with less code). 

Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (0):
Passphrase (0):
BSSID set: 0
We haven't got any access point credentials, so get them now
failed to connect, finishing setup anyway

Exception (3):
epc1=0x40216d98 epc2=0x00000000 epc3=0x00000000 excvaddr=0x401f832d depc=0x00000000

>>>stack>>>

tablatronix, 2019-10-08:
Yeah Not sure if the actual cause is determined, as the workaround is still in there.

```C++
        if(WiFi._persistent) {
            // workaround for #1997: make sure the value of ap_number is updated and written to flash
            // to be removed after SDK update
            wifi_station_ap_number_set(2);
            wifi_station_ap_number_set(1);
```

d-a-v, 2019-10-08:
@dzid26 can you try with latest git version of this core ?

4C3T, 2020-01-09:
I had this error, among others. I succeeded with installing the older version of the ESP8266 board manager, 2.1.0 to be exact, and also chose a lower baud rate to flash it and it worked flawlessly after that.
I'm not sure if this might help anyone else (Or if it's mentioned above. I read a few pages until I realized it actually said to use 9600 Baud on the back of the module, which the newest version couldn't deliver for some reason), but it did work finally.
The sketch I'm using is a sample code for AJAX. It tossed out heaps of errors, but now - Nothing.

kdkwarhead, 2020-03-01:
I got the same problem with my Di Mini Pro.
I Tried with lua firmware from https://nodemcu-build.com before audrino IDE. Didn’t like it much so I tried Arduino IDE instead. It worked fine until I tried “WiFi.begin()”. Total crash!

FIX:
In Arduino IDE:
Change tools/Erase flash to "All flash content" once
flash a sketch.
And then set it back to "only sketch"

I did it with upload speed 57600 but it should work with other speeds as well.
It was some leftover data from old firmware in my case.
Works perfectly now :)

svefro, 2022-11-17:
I have run in to a similar issue with most of my esp8266 devices in my house.
i get random reboots, sometimes 3 minutes sometimes 1 day uptime.
Fatal exception:4 flag:3 (Software Watchdog) epc1:0x4025c1ca epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000

stack trace mostly points to  pm_send_nullfunc, somthing with dhcp, wifi or dns.
nothing in my code. i save stack trace in eeprom on some devices and read it back after boot.

if i do a erase_flash with esptool.exe the problem seams to goes away. (i have done this with a few devices and they seem to be ok)

But i have more than 70-80 devices in my house. They control everything. heatpumps, water heater, floor heating. ventilation, water valves and more.
They all run the same firmware (2 versions 1Mb lite version and 2Mb full that contain all features) that i have developed since from about 2016-2017.

**I recently changed my wifi to Unifi and with that i changed my ssid and password. That is when it all started.**
At the same time did some changes to the firmware so i did not understand that it was the wifi that caused the issue.
i have been debugging ever since.

I have tried implementing ota firmware upload with [kentaylor](https://github.com/kentaylor) s sketch (https://github.com/kentaylor/EraseEsp8266Flash)
but i have had no luck getting it to run after flash is erased :p

**Has there been any progress on this topic since it was started?
I would like to be able to fix the corrupt flash without having to connect to every device and do a errace_flash+reload firmware.**

i'm currently using espressif8266@4.0.1 arduino core v3.0.2

checkFlashCRC and checkFlashConfig returns true;

i see in ESP.cpp that there is a function called rtcUserMemoryWrite.
can i use that to clear the system data? is wifi details stored here?
|<------system data (256 bytes)------->|<-----------------user data (512 bytes)--------------->|