esp8266/Arduino #2000: Password request using wemos D1 R2 / ESP8266

danielperez505, 2016-05-06:
----------------------------- Remove above -----------------------------

### Basic Infos
wireless communication with wemos D1 R2 using ESP8266, its  working perfectly when I connected  with the usb cable

#### Hardware
Hardware:			Wemos D1R2
Core Version:      	      ESP8266

### Description

Im getting a password request  to access a wemos D1 R2 type board password to upload a new sketch¨ but I havent set any pass. I would aprreciated any help!!

### Settings in IDE

### Sketch

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "Mohican";
const char* password = "2a3bf31c";

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 8266
  // ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  ArduinoOTA.setHostname("WemoS");

  // No authentication by default
  // ArduinoOTA.setPassword((const char *)"123");

  ArduinoOTA.onStart([]() {
    Serial.println("Start");
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

   pinMode(D5, OUTPUT);  // initialize onboard LED as output
}

void loop() {
   ArduinoOTA.handle();
  digitalWrite(D5, HIGH);   
  delay(500);                 
  digitalWrite(D5, LOW);
  delay(500);
}

igrr, 2016-05-06:
Which IDE version are you using?

danielperez505, 2016-05-06:
1.6.8

igrr, 2016-05-16:
I tried your sketch, modified WiFi connection code to suit my network setup, and OTA uploads went fine, didn't get any password request. I'm on OS X.
Which OS are you using, and which version of ESP8266 core (2.1.0, 2.2.0, 2.3.0-rc1)?

danielperez505, 2016-05-19:
Im using wemos d1r2 mini. http://www.wemos.cc/Products/d1_mini.html

igrr, 2016-05-19:
That's awesome, but I've asked about OS and ESP8266 Arduino core version.

OS is either Windows (XP, Vista, 7, 10, ...), Linux (which distro?), Mac OS (which version)?

To check ESP8266 Arduino core version — open Arduino IDE, go to Tools menu > Board > Boards Manager, scroll down to "ESP8266", see something like "version 2.2.0 installed". The number between "version" and "installed" is the version number i need :)

danielperez505, 2016-05-19:
Sorry, Im running win 10
ESP 8266 v2.1.0
arduino 1.6.8

igrr, 2016-05-19:
2.1.0 might have had some OTA issues indeed. Please try 2.2.0.

danielperez505, 2016-05-19:
Thanks man, how can Update it  can you pls help me?

igrr, 2016-05-19:
Same menu (boards manager), select new version from drop down box, click install.

danielperez505, 2016-05-19:
I did as you said but the problem still there =( any other suggestion?

igrr, 2016-06-02:
Did you flash the ESP with new version of the code, compiled with 2.2.0? Or did you just update the core without re-flashing the ESP?

mohsh86, 2016-06-10:
i got a similar issue when prompting for console, upload goes fine, console prompts for password, click enter (blank password) doesn't work

Running Ubuntu 16.04, Arduino 1.6.9, ESP 2.2

Nigel-B, 2016-06-17:
The same problem has developed here too. 
Running Cinnamon, Arduino 1.6.8, ESP 2.2
Strangely, I have used the same board a week ago with no problems. Only when trying it today did it start asking for a board password.
Any help would be gratefully received.

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "EE-BrightBox-mw7m7m";
const char* password = "hop-throw-neat";

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 8266
   ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
   //ArduinoOTA.setHostname("myesp8266");

  // No authentication by default
   //ArduinoOTA.setPassword((const char *)"123");

  ArduinoOTA.onStart([]() {
    Serial.println("Start");
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");Linux 
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();
}

danielperez505, 2016-06-17:
I had the same issue weeks ago. 
The issue is with photon. Here the solution. 
https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.md#password-protection.
In the script set a pass for you. 
The once you upload the code using serial communocation restart arduino ide.
The upload the script again using otra
You should be able to upload it
But some times you can't see the serial
Let me know if works for u

Nigel-B, 2016-06-26:
No it doesn't work for me.
It almost seems like there are two password layers or something similar.

danielperez505, 2016-06-26:
As I said for me  it only work to upload the sketch not to se the serial communication. 
Maybe be sure that you install phyton 2.x and not 3.x
Also that you run the ota example, then restart arruinó ide and select the esp as a port.

wi3, 2016-06-28:
i have also the same problem with ota serial communication 

windows 10
Arduino 1.6.8
ESP Package 2.3.0
Python 2.7.12
Hardware is Olimex 8266

With the default Arduino Example
thx in advance

mkeyno, 2016-07-22:
@igrr , i have also the same problem with ota serial communication

windows 7
Arduino 1.6.8 & 1.6.9
ESP Package 2.3.0
Python 2.7 
I've also check all password that I used in AP & station mode and "123" , even when I disable ```cpp ArduinoOTA.setPassword("admin");``` ,prompt request for password

Shane-Lester, 2017-01-25:
Same using a nodemcu v3
windows 10
Arduino 1.6.11
ESP Package 2.3.0
Python 2.7

I have uncommented the line with the password
 ie
  // No authentication by default
   ArduinoOTA.setPassword((const char *)"123");

and tried 123 but still get 
uable to connect: wrong password?

duncan-a, 2017-01-26:
I also have this problem:

Windows 10
Arduino 1.6.8
ESP8266 2.3.0
Board NodeMCU 09 (ESP-12)

danielperez505, 2017-01-26:
I found the issue Windows 10 blocked phyton  so yo have to check your fw

DuncanAmos, 2017-02-11:
"Windows 10 blocked phyton  so yo have to check your fw" isn't overly helpful to newbies and non-techies - can you provide some more details please?

mostlyflawles, 2017-02-15:
Hepp, it worked for me. But serial monitor doesn't work, but that's seems to be a known issue.

How to open the firewall depends on which firewall you are using.
I'm using Windows 7 and ESET and I had to add an rule under advanced setup to allow "c:\Python27\python.exe" to access my internal wifi domain.

http://lmgtfy.com/?q=open+firewall+port+windows+10

Shumatic, 2017-04-18:
Hi

I'am using a ESP-12E, and i don't know the flash size but,
I found out that changing flash size from 512K (64 SPIFFS) to 4M (1M SPIFFS)

devyte, 2017-10-17:
I can't reproduce this. Closing.

closed 2017-10-17 by devyte