esp8266/Arduino #2001: Q: How to reconfigure the station connection without disconnecting active clients from the soft ap

WilkoV, 2016-05-06:
# Basic Infos

#### Hardware
Hardware:			NodeMCU v1.0
Core Version:      	2.2.0

### Description

Hey,

during the setup phase I am running my NodeMCU 1.0 in station and soft AP mode. The idea is, that e.g. an app can connect to the soft ap, places via REST a credentials for the station and waits for the succes + ip or  fail messages. But whenever I call WiFi.begin(ssid, password) a second time the connection is closed by the ESP8266. 

I did this by using  AT commands in an previous project.

### Settings in IDE

Module:  			NodeMCU v1.0
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			?
Flash Frequency:	?
Upload Using:		SERIAL
Reset Method:		?

thewhiterabbit, 2016-05-16:
Could you please post the sketch you are running? I will look at it. I don't think you should call WiFi.begin() more than once in a sketch because that is the connection to the host router. I'm fairly certain that calling it multiple times will cancel the previous connections to the host router.

lukecyca, 2016-06-01:
> places via REST a credentials for the station and waits for the succes + ip or fail messages

Could you have your front-end make the request, then wait some time before making another request to see if it was successful? AFAIK this is how most home router web interfaces deal with this situation.

If necessary you could even explicitly close the connection after responding to the first request, forcing the client to open a new connection for the subsequent request.

devyte, 2017-10-17:
When you call WiFi.begin(), the station switches to the channel of the AP it's trying to connect to. When it does this, the SoftAP is dragged along for the channel change, because the ESP has a single radio. Therefore, the client sees that the SoftAP "disappeared".
The SoftAP needs to be set to the same channel of the AP that the Station is trying to connect to. That is the only work around available.
Closing.

closed 2017-10-17 by devyte