esp8266/Arduino #2003: is addServiceTxt mdns responder working on latest release?

kdsoo, 2016-05-06:
### Basic Infos

#### Hardware
Hardware:			Adafruit Huzzah esp8266
Core Version:      	2.2.0-rc1

### Description

addServiceTxt seems to be not populating mdns txt field.
It's weird because long ago I once tried this and confirmed working at that time.(populating "genuine arduino something....")
I even tried enableArduino() implemented in the library code but it doesn't work either.

void MDNSResponder::enableArduino(uint16_t port, bool auth){
  addService("arduino", "tcp", port);
  addServiceTxt("arduino", "tcp", "tcp_check", "no");
  addServiceTxt("arduino", "tcp", "ssh_upload", "no");
  addServiceTxt("arduino", "tcp", "board", ARDUINO_BOARD);
  addServiceTxt("arduino", "tcp", "auth_upload", (auth) ? "yes":"no");
}

I checked with avahi-browse command

kdsoo@kdsoo-pc:~$ avahi-browse -r _SeaHavenWiFiNode._tcp -p -t
+;wlan0;IPv4;SeaHavenWiFiNode-5ccf7f020a78;_SeaHavenWiFiNode._tcp;local
=;wlan0;IPv4;SeaHavenWiFiNode-5ccf7f020a78;_SeaHavenWiFiNode._tcp;local;seahavenwifinode-5ccf7f020a78.local;192.168.0.31;80;

and avahi-discover (GUI frontend for Linux) also checks out mdns txt field not populating.

Can anyone check this addServiceTxt working on 2.2.0-rc1?

kdsoo, 2016-05-07:
My fault.
Service name added and first argument of addServiceTxt should be match to populate txt field properly.

closed 2016-05-07 by kdsoo

VanTudor, 2020-10-02:
@kdsoo Can you please paste an example with a custom name? I'm trying to do something like 
`  MDNS.addService("somename", "_http", "tcp", mDNSPort);
  MDNS.addServiceTxt("_http", "tcp", "type", mDNSTXTServiceType);`
but it doens't work. I didn't quite get what you meant with the first argument comment.

kdsoo, 2020-10-02:
@VanTudor You should give the same service name to addServiceTxt() as you given as first argument of addService().

`		MDNS.addService("mjpeg", "tcp", 80);
		MDNS.addServiceTxt("mjpeg", "tcp", "endpoint", "/stream");

		MDNS.addServiceTxt("mjpeg", "tcp", "version", VERSION_HASH);

		MDNS.addServiceTxt("mjpeg", "tcp", "build", BUILD_TIME);

		MDNS.addServiceTxt("arduino", "tcp", "version", VERSION_HASH);

		MDNS.addServiceTxt("arduino", "tcp", "build", BUILD_TIME);
`