esp8266/Arduino #2004: linuxarm-xtensa-lx106-elf-g46f160f.tar.gz directories are mode 555 should be 755

barnabear, 2016-05-06:
I've been trying to get Arduino running on Raspbian with ESP8266 support. This is Raspberry Pi (linux on Arm 32 bit).

I managed to build arduino/Arduino from master source, I think it prompted me for a version and suggested 1.6.9, which I accepted.

I then put http://arduino.esp8266.com/stable/package_esp8266com_index.json into the File->Preferences Additional Board Manger URL's field.

Then I selected board manager and the ESP8266 boards. It then tried to install the support tools.

However it failed in creating the subdirectory tree for xtensa. It was trying to create the gcc directory, but the directory it was trying to create it in was mode 555 (not writeable).

So I looked in the .tar.gz it downloaded, and found all the directories to be mode 555.

The equivalent Linux Intel 64 bit .tar.gz has directories 755.

I think the point is that although the archive can be unpacked, you can't then recursively copy the directory tree into place.

Please could this .tar.gz be created with directory permissions of 755, and then I think it will work?

Thanks,

Barnabear.

barnabear, 2016-06-03:
I think you're going to need to change the hash as well.

igrr, 2016-06-12:
Fixed hash and size, thanks.

closed 2016-06-23 by igrr