esp8266/Arduino #2007: ESP-01 Stock in Flash state

mkeyno, 2016-05-07:
---------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-1?
Core Version:      	 2.1.0-rc2 & 2.2

### Description

hi dear @igrr , what your new library  do with ESP-01 module , I compile every library example and it works precisely after program uploaded but when I reset the module is going to flash mode whereas I pull up all pins 0 & 2 , this happened only on my ESP-01 module and other module work perfectly, kindly let me know is it softwaerely possible to module go to flash more after the reset
BR

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

chaeplin, 2016-05-07:
flash size of esp-01 : 512K(blue board color) or 1M(black board color).

mkeyno, 2016-05-07:
I aware of that , also set the black board to 64k but same problem , after upload the sketch and software reset it works but when I use hardware reset it stock in flash more

chaeplin, 2016-05-07:
I have tested both s/w and h/w reset my esp-01/git version. No flash mode.
I had a experience with s/w reset while GPIO 0 was in low, esp started.

mkeyno, 2016-05-08:
so what reason and what situation force the module to go  in flash mode ?

mkeyno, 2016-05-08:
for test I upload the empty sketch and empty folder , but the same behavior ? @chaeplin is there other any way to erase module flash ?

chaeplin, 2016-05-08:
@mkeyno 
Use https://github.com/themadinventor/esptool to erase flash.
./esptool.py --port /dev/tty.usbserial-XXXX erase_flash

If you have multi meter check voltage of GPIO 0 and 2.

igrr, 2016-05-08:
Also please open Arduino serial monitor, switch baud rate to 74880 there, and reset the module. Copy the boot message and post it here, please.

mkeyno, 2016-05-08:
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

igrr, 2016-05-08:
OK, this doesn't indicate the ESP is in boot mode. It says it has loaded whatever sketch there was in flash.

mkeyno, 2016-05-08:
first come this  
ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld

and then come last  boot message , how ever in this mode I don't need to press the flash button  to upload the sketch , that's why I thought my module stock in flash mode

also this is what I upoaded
void setup() 
{
 Serial.begin(115200);
  delay(500);
   Serial.println(F("\n\n\n Starting.....")); 
}

void loop() 
{

}

mkeyno, 2016-05-08:
also I might add in this situation IC module and 3.3 regulator draw much current and get hot

mkeyno, 2016-05-10:
hi @igrr  , I find somehow module flash chip is corrupted, maybe is due to upload the about 200kb files on module with 1Mb flash size , however my attempt to erase the chip with your instruction `esptool.py --port COM23 --baud 76800 erase_flash` is failed `A fatal error occurred: Invalid head of packet,` also  I've try to upload the empty folder data to replace it with old one , but it seems no successful, what is correct way to format the flash clean? if we use FSbrowser example continuously  add or remove new files in module , will is  corrupt the flash ?

mkeyno, 2016-05-10:
also can you direct me to the link that elaborate boot mods and its cause  clearly

mkeyno, 2016-05-10:
dear @chaeplin  can I have your comment for the following stack decoder result

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Starting.....

Soft WDT reset

ctx: cont 
sp: 3fff26f0 end: 3fff29f0 offset: 01b0

0x40105b1a: spi_flash_erase_sector at ?? line ?
0x40205dbc: spiffs_phys_wr at C:-\spiffs/spiffs_cache.c line 204
0x4020fbf8: EspClass::flashEraseSector(unsigned int) at C-/Esp.cpp line 451
0x4020fc01: EspClass::flashEraseSector(unsigned int) at C-Esp.cpp line 451
0x4021167b: spiffs_hal_erase(unsigned int, unsigned int) at C-/spiffs_hal.cpp line 178
0x40203ae3: spiffs_erase_block at C-\spiffs/spiffs_nucleus.c line 851
0x40210429: HardwareSerial::write(unsigned char) at C-/HardwareSerial.h line 121
0x40211508: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at C-/spiffs_hal.cpp line 102
0x4021164c: spiffs_hal_erase(unsigned int, unsigned int) at C-/spiffs_hal.cpp line 168
0x402026a8: SPIFFS_format at C-\spiffs/spiffs_hydrogen.c line 872
0x402112c8: SPIFFSImpl::begin() at C-/spiffs_api.cpp line 123
0x40211ca9: fs::FS::begin() at C-/FS.cpp line 182
0x40208733: setup at -/PIR_socket.ino line 561
0x4021080c: loop_wrapper at C-/core_esp8266_main.cpp line 43
0x40100114: cont_norm at C-/cont.S line 109

chaeplin, 2016-05-10:
@mkeyno Is there any function takes more than 2 sec in setup ?

mkeyno, 2016-05-11:
I don't  thinks so , just the 500 ms delay

void setup() 
{
   
  //Serial.begin(115200);
  Serial.begin(115200, SERIAL_8N1,SERIAL_TX_ONLY);
  delay(500);
  //Serial.setDebugOutput(true);
  Serial.println(F("\Starting....."));

 for(byte i=0;i<number_of_sensor;i++)
 {
  pinMode(INPUTS[i], INPUT_PULLUP);
 pinMode(OUTPUTS[i], OUTPUT); 
 digitalWrite(OUTPUTS[i],PIN_INI[i]);
 }

  
  pinMode(LED_Status, OUTPUT);
  
  SPIFFS.begin();
  EEPROM.begin(256); 

 
  DIR(); 
  restor_config(); //from eeprom
  
Serial.print("Slected FlashChipSize: ");Serial.println(formatBytes(ESP.getFlashChipSize()));
Serial.print("FlashChipRealSize: ");Serial.println(formatBytes(ESP.getFlashChipRealSize()));
Serial.print("SketchSize:");Serial.println(formatBytes(ESP.getSketchSize()));
Serial.print("FreeSketchSpace:");Serial.println(formatBytes(ESP.getFreeSketchSpace()));
Serial.print("FreeHeap:");Serial.println(formatBytes(ESP.getFreeHeap()));

Serial.print("\nChipId: ");Serial.println(ESP.getChipId());
Serial.print("CpuFreqMHz: ");Serial.println(ESP.getCpuFreqMHz());  
Serial.print("SdkVersion: ");Serial.println(ESP.getSdkVersion());
Serial.print("FlashChipSpeed: ");Serial.println(ESP.getFlashChipSpeed());
Serial.print("FlashChipMode: ");Serial.println(ESP.getFlashChipMode());

  
  Serial.println("\n\nDevice ID=" + preSetPass +"(" + preSetUser + ")");
  Serial.print("Mode=");          Serial.println(MyMode);
  Serial.print("Hostname=");      Serial.println(myHostname);
  Serial.print("SSID=");           Serial.println(SSID);
  Serial.print("Password =");      Serial.println(ssidPass);
  Serial.print("AP SSID=");        Serial.println(AP_SSID);
  Serial.print("AP Password =");   Serial.println(AP_ssidPass);
  Serial.print("reset_number =");  Serial.println(reset_number);
  Serial.print("Server IP Address =");  Serial.println(S_IP);
  update_description();
  
 if(MyMode[0]=='A' || reset_number>4 || SSID.length()<3 )
                            {
                               Serial.println("\t\t\t Entering the Access Mode");
                               WiFi.mode(WIFI_AP);
                               WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
                               String temp_name="Smart_Node_" +String(ESP.getChipId(), HEX);
                               if(AP_SSID.length()<3) {WiFi.softAP(temp_name.c_str(),"12345678"); AP_SSID=temp_name; }
                               else                   WiFi.softAP(AP_SSID.c_str(),AP_ssidPass.c_str());                                
                               delay(500); // Without delay I've seen the IP address blank
                               EEPROM_Write(String(0),76,77);   
                               PerfixID=WiFi.softAPIP().toString().substring(8);
                               Serial.print("AP IP address: ");Serial.println(WiFi.softAPIP());
                               dnsServer.setErrorReplyCode(DNSReplyCode::ServerFailure);
                               dnsServer.start(DNS_PORT, "*", apIP);  
                            }
else if(reset_number<5)
              {
                Serial.println("\t\t\t Entering the Station Mode");
                WiFi.mode(WIFI_STA);
                WiFi.begin(SSID.c_str(), ssidPass.c_str());             
                unsigned long startTime = millis();
                while (WiFi.status() != WL_CONNECTED && millis() - startTime < 10000)  {Serial.write('%');delay(500); } 
                if(WiFi.status() != WL_CONNECTED) {
                                reset_number++;
                        EEPROM_Write(String(reset_number),76,77);
                        ESP.restart();
                        }            
                 PerfixID=WiFi.localIP().toString().substring(8);
                 Serial.print("AP IP address: ");Serial.println(WiFi.localIP());
                if (!MDNS.begin(myHostname.c_str())) {
                        Serial.println("Error setting up MDNS responder!");
                        while(1) { 
                        delay(1000);
                        }
                      }
                Serial.println("mDNS responder started");
              }
 
  
 
  server.on("/process", process);    
  server.on("/update",      HTTP_POST, [](){ Serial.println("Upload post");    server.send(200, "text/plain", ""); /* Redirect("/setting.html");*/  }, handleFileUpload);
  server.on("/", handleRoot);
  server.on("/config.html", handleConfig);
  server.onNotFound(handleOther);
  
  server.begin();  Serial.println("\t\t\tHTTP server started");  
  
  const char * headerkeys[] = {"User-Agent","Cookie"} ; //here the list of headers to be recorded
  size_t hz = sizeof(headerkeys)/sizeof(char*); 
  server.collectHeaders(headerkeys, hz ); //ask server to track these headers
  
  webSocket.begin();
  webSocket.onEvent(webSocketEvent);
  
}

mkeyno, 2016-05-13:
its soled , if we upload the empty sketch folder twice with to flash size selection , its kindda restate itself and lives again

closed 2016-05-13 by mkeyno