esp8266/Arduino #2009: ESP8266 WDT WifiClientSecure

SiFly, 2016-05-09:
### Basic Infos

I use WifiSecureClient connecting  to a server which does **NOT** support **TLS_RSA_WITH_AES_256_CBC_SHA**. So it abruptly drops the connection (i.e. straight off sends a FIN)

The call from the connect verb does not return with a failure. This appears to cause a watchdog timeout. Given that it is possible to fake up  a FIN packet quite easily during the point of cipher negotiation this is surely quite an issue.

Given that more and more servers are going to stop dropping plain old SHA soon surely we should be moving to SHA256 ?

#### Hardware
Hardware:			ESP-12
Core Version:      	2.1.0

### Description

Problem description

### Settings in IDE

Module:  			NodeMCU ESP8266012
Flash Size: 		4MB
CPU Frequency:     80Mhz

3fff53f9 00000000 00000005 40205fc8
3fff25b0:  4020153a 00000000 3fff300c 40239bdb
3fff25c0:  00000034 00000000 3fff5254 4020604a
3fff25d0:  00000034 00000001 0000002f 0000002f
3fff25e0:  00020011 00000000 00000034 4023957c
3fff25f0:  4024f9b0 00000010 00000004 0000002f
3fff2600:  00000000 3fff53f9 3fff300c 40239fc1
3fff2610:  4377ab04 3fff4c0c 00000200 0000002f
3fff2620:  00000000 3fff53f9 3fff300c 4023a229
3fff2630:  3ffea2a4 3fff4c0c 3fff300c 40238bd0
3fff2640:  0000001c 4022370a 3fff2b0c 00000000
3fff2650:  00000000 3fff4c0c 3fff300c 4023a2b4
3fff2660:  000001bb 3fff13d0 3fff13d0 3fff16c0
3fff2670:  3fff16c4 3fff13d0 3fff2b0c 40205f62
3fff2680:  8e81321f 3fff13d0 4020b434 3fff13d0
3fff2690:  3ffe8fad 000001bb 3fff13d0 4020612a
3fff26a0:  3ffe9ca0 8e81321f 3ffe8fd0 00000274
3fff26b0:  3ffe8fad 000001bb 3fff13d0 40205375
3fff26c0:  3ffe9ca0 8e81321f 3ffe9ca0 8e81321f
3fff26d0:  3ffe8fad 3fff18d0 3fff18d0 4020c9dc
3fff26e0:  3fff18d0 3fff18d0 3fff115c 40206140
3fff26f0:  3ffe8e99 3fff18d0 3fff115c 40202dc6
3fff2700:  3fff1604 20bd14f6 3fff18d0 4020c9dc
3fff2710:  3ffeb8e4 00000000 3fff18d0 3fff2774
3fff2720:  3fff18d0 00000000 3fff115c 402028ef
3fff2730:  7fffffff 3ffeb8e4 3ffeb8e4 402038d8
3fff2740:  3fff1604 3fff1625 3fff1635 40202c30
3fff2750:  3fff300c 20bd14f6 00000000 3fff198c
3fff2760:  3fffdad0 000861c1 3fff2774 40202cac
3fff2770:  3ffe9120 3fff2770 3fff277c 00000000
3fff2780:  3ffe8f53 00000002 3ffe8f57 ffffffff
3fff2790:  400044ab 3fffc718 3fff2880 08000000
3fff27a0:  60000200 08000000 08000000 00000000
3fff27b0:  0000ffff 00000000 ffffff01 003fd000
3fff27c0:  4024f1c0 00000484 003fd000 00000030
3fff27d0:  00000001 00000013 3fff18d0 4020b2d9
3fff27e0:  3ffe8e99 4024cea8 3fff18d0 4020b2d9
3fff27f0:  3fff18ac 000006e1 000006e1 4010020c
3fff2800:  00000001 3ffe9841 40210067 00000001
3fff2810:  ffffffff 00000000 3ffeb191 00000008
3fff2820:  402100ba 3ffee8f0 3ffee8f0 3fff366c
3fff2830:  3ffef3fc 0000003c 401020f3 3ffee8f0
3fff2840:  00000000 4000444e 00000000 3ffeede4
3fff2850:  00000000 4000422e 60000200 3fff28fc
3fff2860:  00000100 40004aa0 00000100 003ff000
3fff2870:  00000100 3fff28e0 3fffc718 003ff000
3fff2880:  3fff18ac 0000052d 0000052d 4010020c
3fff2890:  0000001c 4024f1c0 3fff538c 401008c8
3fff28a0:  401060da 3fff28e0 000003ff 401075f0
3fff28b0:  3fff18ac 000003fe 000003fe 4010020c
3fff28c0:  3fff18ac 0000015f 0000015f 4010020c
3fff28d0:  3fff18ac 00000156 00000156 4010020c
3fff28e0:  3fff1660 00000000 3fff34d4 401008c8
3fff28f0:  3fff18ac 000000f5 000000f5 4010020c
3fff2900:  4022d29a 3fff14f0 3fff3bbc 3fff1660
3fff2910:  4022e0bf 00000001 3ffe8e9a 4020c904
3fff2920:  3fff2b5c 00000001 3ffe8e9a 4020c904
3fff2930:  3fff115c 0000000b 3ffe8f69 3fff198c
3fff2940:  3fffdad0 0000000c 3fff18d0 4020b2d9
3fff2950:  3ffe8e99 000861c1 3fff18d0 4020b2d9
3fff2960:  3ffe8f5e 00000000 3fff18d0 4020c9dc
3fff2970:  3ffe8f48 00000018 3fff2774 3ffe8f53
3fff2980:  feefeffe 000861c1 3fff1158 40202cfc
3fff2990:  3fffdad0 00000000 3fff1985 40203289
3fff29a0:  feefeffe feefeffe 3fff1985 4020b480
3fff29b0:  feefeffe feefeffe 3fff19a0 40100958
<<<stack<<<

Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL?
Reset Method:		nodemcu

igrr, 2016-05-16:
Could you please decode the stack dump and paste the result here? Thanks.
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

igrr, 2016-06-02:
@SiFly which server was that? We now have SHA256 digest in axTLS, i would like to test.

devyte, 2017-10-17:
@SiFly is this issue still valid with latest git?

SiFly, 2017-10-17:
this appears to be resolved

closed 2017-10-17 by SiFly