esp8266/Arduino #2018: don't use rtc mem data while connecting the ESP in Putty

mril, 2016-05-11:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-13 shield
Core Version:    

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Debug Messages
dont use rtc mem data
ets Jan 8 2013, rst cause :2, boot mode (3:6);

wdt reset 
load 0x40100000, len 28636, room 16 
tail 12
chksum 0x61
ho 0 tail 12 room 4 
load0x3ffe8000, len 1960, room 12
tail 12
chksum 0x86
ho 0 tail 12 room 4  load 0x3ffe87b0, len 14020, rooù 12
tail 8
chksum 0xfd
csum 0xfd
messages here
```

Hi all I'm using ESP 13 Shield, I connected the TX/RX/GND to a FTDI basic programmer, when I plug the esp I receive bunch of encoded characters  with the messages dont use rtc..., the message in the debug messages section change often, even sometimes I got only don't use rtc mem data.
Guys need help pls what do I have to do.
THKS

mril, 2016-05-13:
UPPPPPPP

mcauser, 2016-05-13:
I believe it is part of the latest Espressif firmware. I am getting the same on an ESP-12F running MicroPython 1.8.69

igrr, 2016-05-13:
Given that you are seeing this message, you must be using some firmware, not the one generated by this project. We have code in place which specifically works around displaying this message :)
If you are using something like AT firmware with an external Arduino AVR microcontroller, please direct your question to the ESP8266 community forum. This project is about using an ESP8266 instead of an AVR microcontroller.

closed 2016-06-02 by igrr