esp8266/Arduino #2020: Infinite Loop in MDNSResponder::_parsePacket

liquidfalcon, 2016-05-12:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	Git 

### Description

MDNSResponder::_parsePacket hits an infinite loop at the following code:

```
while (answer->next != 0) { 
   answer = _answers->next; 
} 
```

If I throw a counter in there and log it, it hits ~5500 iterations before the WDT fires. Additionally, putting in a simple if (iterations > 32) break; seems to stave off the crash, but a memory leak occurs each time I use MDNS. This happens the first time and every time I query services using MDNS.
### Settings in IDE

Module:  			ESP-12E - Built with `Core Development Module` and open source LWIP
Flash Size: 		4M
CPU Frequency:		80MHz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:	        Serial
Reset Method:		N/A

### Sketch

Simply calling  `MDNS.queryService(F("$MY_SERVICE"), "tcp");`  exhibits this behaviour. Additionally, I have approximately 10 other ESP-12 modules and other devices running similar code on this network, all implementing $MY_SERVICE.

### Debug

Stack trace:

```
Soft WDT reset

ctx: sys 
sp: 3ffffa60 end: 3fffffb0 offset: 01b0

>>>stack>>>
3ffffc10:  4021582d 3ffffe20 3fff0aa0 40215638  
3ffffc20:  64616f6c 30302d7a 35313030 9a003637  
3ffffc30:  d1bebb62 17722efd fde9550f af2e5c55  
3ffffc40:  eb90ff85 1a0e9f3f eedcd412 c8a072b0  
3ffffc50:  74169889 d092ed45 693767a6 c238e2cd  
3ffffc60:  00000005 00000000 00000020 40102032  
3ffffc70:  3ffea9a5 4010530f 3ffee408 e5415f8e  
3ffffc80:  40102d45 3ffee408 40220cab 00000000  
3ffffc90:  fffffff9 09373f8d 3ffeee58 40102ed4  
3ffffca0:  3ffeb244 00000000 00000000 00000001  
3ffffcb0:  00000000 09373f8d 40103312 00000100  
3ffffcc0:  7fffffff 3ffeb244 3ffeb244 00000001  
3ffffcd0:  00000001 40220153 3ffee138 3fff2dd7  
3ffffce0:  00000000 09373f8d 00000000 4000050c  
3ffffcf0:  3fffc278 401030ac 3fffc200 00000022  
3ffffd00:  3ffeb238 40246ab0 3fff348e 3fff3454  
3ffffd10:  402266b0 00000030 00000018 05ffffff  
3ffffd20:  61636f6c 3030006c 35313030 5f0c3637  
3ffffd30:  65776f50 6e655372 04726f73 7063745f  
3ffffd40:  636f6c05 00006c61 0000001b 3fff30d4  
3ffffd50:  3ffeb2f1 3ffeef94 00010000 00000030  
3ffffd60:  3fff3454 3fff2db0 3fff2da4 40246ffc  
3ffffd70:  00000001 00000000 00000020 40102032  
3ffffd80:  000000ac 00027ea7 000001ac 402480a0  
3ffffd90:  00000001 40104503 3ffeeb68 0000000c  
3ffffda0:  3ffffe74 3ffffe70 00000080 00000006  
3ffffdb0:  00000000 00000000 3fff2da4 40246d95  
3ffffdc0:  000005e0 00000000 40104131 3ffeeb20  
3ffffdd0:  0000003c 00000000 00000002 00000100  
3ffffde0:  3ffeeb20 40101bc6 00000001 00000000  
3ffffdf0:  00000000 40101993 00000000 3ffeeaf0  
3ffffe00:  00000005 00000000 00000020 40102032  
3ffffe10:  3ffea9a5 4010530f 3ffee3e0 00000016  
3ffffe20:  40102d45 3ffee3e0 3ffeeb20 00000014  
3ffffe30:  fffffff7 09374cb8 3ffeee58 6f6c2ed4  
3ffffe40:  006c6163 30303030 36373531 40100200  
3ffffe50:  3fff294c 3ffe9162 00000001 0000068c  
3ffffe60:  00008400 00000004 01ac0000 40109301  
3ffffe70:  00000000 00000001 00000050 00000001  
3ffffe80:  00000005 09374cb8 00000000 4000050c  
3ffffe90:  3fffc278 401030ac 3fffc200 3ffed20c  
3ffffea0:  3fff0aa0 3fff2924 3fff2c34 40215b6d  
3ffffeb0:  3fff2da4 3fff294c 3fff2c34 4021d76b  
3ffffec0:  40000f58 00000000 00000020 40214568  
3ffffed0:  3fff2da4 3fff294c 3fff2844 402427a0  
3ffffee0:  000014e9 00000000 3ffed1f8 3fff27d8  
3ffffef0:  000000e0 3fffdcb0 3ffefbc8 40107a84  
3fffff00:  00000000 00000000 3ffeb348 3fff27d0  
3fffff10:  3ffed1f8 3fff27d8 3fff294c 40247e20  
3fffff20:  00000002 3fff2da4 3fff2da4 3ffef550  
3fffff30:  00000000 3fff294c 0000001c 3fff2da4  
3fffff40:  3ffed1ea 00000000 3fff294c 402470cd  
3fffff50:  930101ac 0000014b 00000000 00000027  
3fffff60:  00000002 00000018 4022060f 3ffee3e0  
3fffff70:  3ffed1c4 3fffdcc0 3ffeaa98 3ffeaa98  
3fffff80:  40220582 3ffee3e0 00000000 3fff2e24  
3fffff90:  3fffdc80 00000000 3fff294c 40234623  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<
```

Decoded:

```
Decoding 41 results
0x4021582d: MDNSResponder::_parsePacket() at /home/danielm/arduino-esp8266/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp:360
0x40215638: MDNSResponder::_parsePacket() at /home/danielm/arduino-esp8266/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp:360
0x40102032: pp_post at ??:?
0x4010530f: lmacRxDone at ??:?
0x40102d45: trc_NeedRTS at ??:?
0x40220cab: pp_attach at ??:?
0x40102ed4: trc_NeedRTS at ??:?
0x40103312: wDev_ProcessFiq at ??:?
0x40220153: ppTxPkt at ??:?
0x4000050c: ?? ??:0
0x401030ac: wDev_ProcessFiq at ??:?
0x40246ab0: etharp_send_ip at /home/danielm/arduino-esp8266/hardware/esp8266com/esp8266/tools/sdk/lwip/src/netif/etharp.c:435
0x402266b0: ieee80211_ht_updateparams at ??:?
0x40246ffc: etharp_output at /home/danielm/arduino-esp8266/hardware/esp8266com/esp8266/tools/sdk/lwip/src/netif/etharp.c:995
0x40102032: pp_post at ??:?
0x402480a0: ip_output_if_opt at /home/danielm/arduino-esp8266/hardware/esp8266com/esp8266/tools/sdk/lwip/src/core/ipv4/ip.c:780
0x40104503: lmacRecycleMPDU at ??:?
0x40246d95: etharp_output_to_arp_index at /home/danielm/arduino-esp8266/hardware/esp8266com/esp8266/tools/sdk/lwip/src/netif/etharp.c:890
0x40104131: lmacIsIdle at ??:?
0x40101bc6: ppEnqueueRxq at ??:?
0x40101993: ppProcessTxQ at ??:?
0x40102032: pp_post at ??:?
0x4010530f: lmacRxDone at ??:?
0x40102d45: trc_NeedRTS at ??:?
0x40100200: _umm_free at /home/danielm/arduino-esp8266/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:1277
0x40109301: ?? ??:0
0x4000050c: ?? ??:0
0x401030ac: wDev_ProcessFiq at ??:?
0x40215b6d: MDNSResponder::update() at /home/danielm/arduino-esp8266/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp:360
0x4021d76b: std::_Function_handler<void (), std::_Bind<std::_Mem_fn<void (MDNSResponder::*)()> (MDNSResponder*)> >::_M_invoke(std::_Any_data const&) at /home/danielm/arduino-esp8266/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp:360
0x40000f58: ?? ??:0
0x40214568: UdpContext::_s_recv(void*, udp_pcb*, pbuf*, ip_addr*, unsigned short) at /home/danielm/arduino-esp8266/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp:360
0x402427a0: udp_input at /home/danielm/arduino-esp8266/hardware/esp8266com/esp8266/tools/sdk/lwip/src/core/udp.c:343
0x40107a84: pvPortMalloc at /home/danielm/arduino-esp8266/hardware/esp8266com/esp8266/cores/esp8266/heap.c:13
0x40247e20: ip_input at /home/danielm/arduino-esp8266/hardware/esp8266com/esp8266/tools/sdk/lwip/src/core/ipv4/ip.c:553
0x402470cd: ethernet_input at /home/danielm/arduino-esp8266/hardware/esp8266com/esp8266/tools/sdk/lwip/src/netif/etharp.c:1379
0x4022060f: pp_tx_idle_timeout at ??:?
0x40220582: pp_tx_idle_timeout at ??:?
0x40234623: ets_snprintf at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
```

igrr, 2016-06-02:
/cc @larsenglund

larsenglund, 2016-06-02:
Can you enable MDNS_DEBUG_RX and post the serial output here?

liquidfalcon, 2016-06-23:
Sorry for the slow response there gents, log output is below (Immediately after this log is the WDT fire and subsequent stack trace):

```
Reading answers RX: REQ, ID:0, Q:0, A:4, NS:0, ADD:0
 12 5f 50 6f 77 65 72 53 65 6e 73 6f 72 
found matching service: PowerSensor
 4 5f 74 63 70 
 5 6c 6f 63 61 6c 
type: 000c rdlength: 40
0e 4c 69 6e 65 7a 2d 30 30 30 30 31 36 33 34 0c 5f 50 6f 77 65 72 53 65 6e 73 6f 72 04 5f 74 63 70 05 6c 6f 63 61 6c 00 
 14 4c 69 6e 65 7a 2d 30 30 30 30 31 36 33 34 
 12 5f 50 6f 77 65 72 53 65 6e 73 6f 72 
found matching service: PowerSensor
 4 5f 74 63 70 
 5 6c 6f 63 61 6c 
type: 0010 rdlength: 0

 14 4c 69 6e 65 7a 2d 30 30 30 30 31 36 33 34 
 12 5f 50 6f 77 65 72 53 65 6e 73 6f 72 
found matching service: PowerSensor
 4 5f 74 63 70 
 5 6c 6f 63 61 6c 
type: 0021 rdlength: 28
 14 6c 69 6e 65 7a 2d 30 30 30 30 31 36 33 34 
linez-00001634
 14 6c 69 6e 65 7a 2d 30 30 30 30 31 36 33 34 
 5 6c 6f 63 61 6c 
type: 0001 rdlength: 4
All answers parsed, adding to _answers list..
Reading answers RX: REQ, ID:0, Q:0, A:4, NS:0, ADD:0
 12 5f 50 6f 77 65 72 53 65 6e 73 6f 72 
found matching service: PowerSensor
 4 5f 74 63 70 
 5 6c 6f 63 61 6c 
type: 000c rdlength: 40
0e 4c 69 6e 65 7a 2d 30 30 30 30 31 37 37 37 0c 5f 50 6f 77 65 72 53 65 6e 73 6f 72 04 5f 74 63 70 05 6c 6f 63 61 6c 00 
 14 4c 69 6e 65 7a 2d 30 30 30 30 31 37 37 37 
 12 5f 50 6f 77 65 72 53 65 6e 73 6f 72 
found matching service: PowerSensor
 4 5f 74 63 70 
 5 6c 6f 63 61 6c 
type: 0010 rdlength: 0

 14 4c 69 6e 65 7a 2d 30 30 30 30 31 37 37 37 
 12 5f 50 6f 77 65 72 53 65 6e 73 6f 72 
found matching service: PowerSensor
 4 5f 74 63 70 
 5 6c 6f 63 61 6c 
type: 0021 rdlength: 28
 14 6c 69 6e 65 7a 2d 30 30 30 30 31 37 37 37 
linez-00001777
 14 6c 69 6e 65 7a 2d 30 30 30 30 31 37 37 37 
 5 6c 6f 63 61 6c 
type: 0001 rdlength: 4
All answers parsed, adding to _answers list..
Reading answers RX: REQ, ID:0, Q:0, A:4, NS:0, ADD:0
 12 5f 50 6f 77 65 72 53 65 6e 73 6f 72 
found matching service: PowerSensor
 4 5f 74 63 70 
 5 6c 6f 63 61 6c 
type: 000c rdlength: 40
0e 4c 69 6e 65 7a 2d 30 30 31 32 31 33 33 38 0c 5f 50 6f 77 65 72 53 65 6e 73 6f 72 04 5f 74 63 70 05 6c 6f 63 61 6c 00 
 14 4c 69 6e 65 7a 2d 30 30 31 32 31 33 33 38 
 12 5f 50 6f 77 65 72 53 65 6e 73 6f 72 
found matching service: PowerSensor
 4 5f 74 63 70 
 5 6c 6f 63 61 6c 
type: 0010 rdlength: 0

 14 4c 69 6e 65 7a 2d 30 30 31 32 31 33 33 38 
 12 5f 50 6f 77 65 72 53 65 6e 73 6f 72 
found matching service: PowerSensor
 4 5f 74 63 70 
 5 6c 6f 63 61 6c 
type: 0021 rdlength: 28
 14 6c 69 6e 65 7a 2d 30 30 31 32 31 33 33 38 
linez-00121338
 14 6c 69 6e 65 7a 2d 30 30 31 32 31 33 33 38 
 5 6c 6f 63 61 6c 
type: 0001 rdlength: 4
All answers parsed, adding to _answers list..
Reading answers RX: REQ, ID:0, Q:0, A:4, NS:0, ADD:0
 12 5f 50 6f 77 65 72 53 65 6e 73 6f 72 
found matching service: PowerSensor
 4 5f 74 63 70 
 5 6c 6f 63 61 6c 
type: 000c rdlength: 40
0e 4c 69 6e 65 7a 2d 30 30 30 30 31 36 34 34 0c 5f 50 6f 77 65 72 53 65 6e 73 6f 72 04 5f 74 63 70 05 6c 6f 63 61 6c 00 
 14 4c 69 6e 65 7a 2d 30 30 30 30 31 36 34 34 
 12 5f 50 6f 77 65 72 53 65 6e 73 6f 72 
found matching service: PowerSensor
 4 5f 74 63 70 
 5 6c 6f 63 61 6c 
type: 0010 rdlength: 0

 14 4c 69 6e 65 7a 2d 30 30 30 30 31 36 34 34 
 12 5f 50 6f 77 65 72 53 65 6e 73 6f 72 
found matching service: PowerSensor
 4 5f 74 63 70 
 5 6c 6f 63 61 6c 
type: 0021 rdlength: 28
 14 6c 69 6e 65 7a 2d 30 30 30 30 31 36 34 34 
linez-00001644
 14 6c 69 6e 65 7a 2d 30 30 30 30 31 36 34 34 
 5 6c 6f 63 61 6c 
type: 0001 rdlength: 4
All answers parsed, adding to _answers list..

```

me-no-dev, 2016-07-28:
Can you guys test [this PR](https://github.com/esp8266/Arduino/pull/2333)?
Let me know if you still get resets
```
mDNS query _arduino._tcp
2 service(s) found
1: (192.168.254.182:8266) esp-test
2: (192.168.254.156:22) piduino1
mDNS query _http._tcp
2 service(s) found
1: (192.168.254.182:80) esp-test
2: (192.168.254.194:80) HP LaserJet CP1025nw
mDNS query _ipp._tcp
1 service(s) found
1: (192.168.254.194:631) HP LaserJet CP1025nw
mDNS query _afpovertcp._tcp
1 service(s) found
1: (192.168.254.49:548) Hristo's Time Capsule
mDNS query _smb._tcp
1 service(s) found
1: (192.168.254.49:445) Hristo's Time Capsule
mDNS query _ftp._tcp
0 service(s) found
```

devyte, 2018-01-03:
The referenced PR was merged.
Closing due to no feedback in over 2 months.

closed 2018-01-03 by devyte

devyte, 2018-11-26:
yes, the current mdns is broken. A rewrite was done in #5370 and it's integration into our core as an experimental replacement is WIP.

arihantdaga, 2018-11-26:
Ohh... @devyte  Really Thank you for confirming. Is there any temporary solution  or workaround ? We are sipping our products, and mdns is an essential component ?

devyte, 2018-11-26:
Only to try the new implementation to check if it works for you. The current code is broken in several dimensions simultaneously.