esp8266/Arduino #2022: HttpUpdate Example - HTTP_UPDATE_FAILD Error (-104): Wrong HTTP code

Zyxmn, 2016-05-13:
I have tried to implement the HttpServer Example albeit it with a few changes. Here is my sketch:

///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
/**
 * httpUpdate.ino
 *
 *  Created on: 27.11.2015
 *
 */

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

const char* ssid = "ssid";
const char* password = "password";

void setup() 
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(100);
  
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) 
  {
    delay(500);
    Serial.print(".");
  }
  
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  delay(500);

}

void loop() {
  
        t_httpUpdate_return ret = ESPhttpUpdate.update("https://rawgit.com/Zyxmn/esp8266_BlinkyBin/master/Blinky.cpp.bin");
        
        delay(500);
        switch(ret) {
            case HTTP_UPDATE_FAILED:
                Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
                Serial.println();
                Serial.println();
                Serial.println();
                break;

            case HTTP_UPDATE_NO_UPDATES:
                Serial.println("HTTP_UPDATE_NO_UPDATES");
                Serial.println();
                Serial.println();
                break;

            case HTTP_UPDATE_OK:
                Serial.println("HTTP_UPDATE_OK");
                Serial.println();
                Serial.println();
                Serial.println();
                break;
        }
        delay(500);
}

///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

I get the wrong http code error as mentioned in the title, I was wondering if you anyone can help me figure out how to resolve it. 

Thanks a lot in advance fellas! Appreciate the help.

igrr, 2016-05-13:
Which version of ESP8266 Arduino core are you using?

Zyxmn, 2016-05-13:
The Adafruit Huzzah Board

igrr, 2016-05-13:
I mean, which version of software did you install? 2.0.0, 2.1.0, 2.2.0, 2.3.0-rc1, etc?

Zyxmn, 2016-05-13:
Oh yeah sorry, its  2.2.0 .

igrr, 2016-05-13:
Could you please try running the same with 2.3.0-rc1? There were some issues with HTTPClient library which were fixed in 2.3.0-rc1.

Zyxmn, 2016-05-13:
Installed the 2.3.0-rc1 , the code wont compile now . says 

Board huzzah (platform esp8266, package esp8266) is unknown

Error compiling for board Adafruit HUZZAH ESP8266.

igrr, 2016-05-13:
Did you remove 2.2.0 before installing 2.3.0-rc1?
If not, you may need to close the IDE and remove the old package.
On Windows, it is under `Users\[username]\AppData\Roaming\Arduino15\packages\esp8266\hardware`.
On Mac, `/Users/$USER/Library/Arduino15/packages/esp8266/hardware`.
See #1514, #1387.

Zyxmn, 2016-05-13:
I did not remove the 2.2.0 before installing 2.3.0-rc1 . I manually deleted the 2.2.0 folder and installed the 2.3.0-rc1 like you said. The code compiles fine now. 

The new HTTP error it shows now is :+1: 
 HTTP_UPDATE_FAILED Error (-1): HTTP error: connection refused

igrr, 2016-05-13:
Which is probably the correct behaviour now, because you are loading HTTPS URL without specifying certificate fingerprint. Could you try the same, replacing `https` with `http` in your binary URL?

Zyxmn, 2016-05-13:
Tried the http version, the error is back to 
HTTP_UPDATE_FAILED Error (-104): Wrong HTTP code

How do I obtain a certificate fingerprint for my url ? might try https with the fingerprint certificate like in the example.

igrr, 2016-05-13:
Your server sends HTTP code 301, "Moved permanently". ESP8266HTTPUpdate library doesn't support HTTP redirects at this point. It expects an URL which points directly to the binary.

```
curl -v http://rawgit.com/Zyxmn/esp8266_BlinkyBin/master/Blinky.cpp.bin
*   Trying 104.18.62.176...
* Connected to rawgit.com (104.18.62.176) port 80 (#0)
> GET /Zyxmn/esp8266_BlinkyBin/master/Blinky.cpp.bin HTTP/1.1
> Host: rawgit.com
> User-Agent: curl/7.48.0
> Accept: */*
> 
< HTTP/1.1 301 Moved Permanently
< Date: Fri, 13 May 2016 11:52:44 GMT
< Content-Type: text/html; charset=utf-8
< Content-Length: 214
< Connection: keep-alive
< Set-Cookie: __cfduid=dd3dbfc868520226a7f6f7b3f6da9855c1463140364; expires=Sat, 13-May-17 11:52:44 GMT; path=/; domain=.rawgit.com; HttpOnly
< X-Content-Type-Options: nosniff
< X-Robots-Tag: none
< RawGit-Naughtiness: 0
< Access-Control-Allow-Origin: *
< Cache-Control: public, max-age=2592000
< Location: https://raw.githubusercontent.com/Zyxmn/esp8266_BlinkyBin/master/Blinky.cpp.bin
< Vary: Accept
< RawGit-Cache-Status: HIT
< CF-Cache-Status: HIT
< Expires: Sun, 12 Jun 2016 11:52:44 GMT
< Server: cloudflare-nginx
< CF-RAY: 2a25e7eb815a16b2-ARN
```

Zyxmn, 2016-05-13:
Thanks for that! Do you have any suggestions on servers I could use to get a direct link to the bin file?

igrr, 2016-05-16:
Well using the URL from `Location` header above (`https://raw.githubusercontent.com/Zyxmn/esp8266_BlinkyBin/master/Blinky.cpp.bin`) worked for me.

Zyxmn, 2016-05-16:
Could you please post the sketch you've used ? I  have tried both the http and https URLS, and it still does not seem to be working for me.

HTTPS : gives the connection refused error
HTTP : gives the wrong http code error.

I even tried the default httpServer example, changing on the ssid password and the url . The results are the same.

igrr, 2016-05-16:
Sketch was:
```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  WiFi.begin("testbox1", "testbox123");
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
  }
  Serial.println("Connected, updating");
  delay(1000);
  
  t_httpUpdate_return ret = ESPhttpUpdate.update("https://raw.githubusercontent.com/Zyxmn/esp8266_BlinkyBin/master/Blinky.cpp.bin", "", "21 99 13 84 63 72 17 13 B9 ED 0E 8F 00 A5 9B 73 0D D0 56 58");
}
void loop() {
}
```
Don't forget to power cycle the ESP after uploading this sketch, otherwise it will fail to update.

Output (with some debugging enabled from Tools menu):
```
rl.lúû|.ål‡|....å.lÏ.b|éÇ.Ï.íríbå.båÚnnûlnnú‚Ï.b.pÏélrlrlpÚn‡....Ç.l.åú...b.n‚|.lååéébåÚnnÓ.låél`.ê..nn.l`...nréíín...åú.l.ríín...åú.l¸Çnú.
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with testbox1, channel 6
dhcp client start...
ip:192.168.77.138,mask:255.255.255.0,gw:192.168.77.1
Connected, updating
[HTTP-Client][begin] url: https://raw.githubusercontent.com/Zyxmn/esp8266_BlinkyBin/master/Blinky.cpp.bin
[HTTP-Client][begin] host: raw.githubusercontent.com port: 443 url: /Zyxmn/esp8266_BlinkyBin/master/Blinky.cpp.bin
[HTTP-Client][begin] httpsFingerprint: 21 99 13 84 63 72 17 13 B9 ED 0E 8F 00 A5 9B 73 0D D0 56 58
please start sntp first !
State:.sending Client Hello (1)
State:.receiving Server Hello (2)
State:.receiving Certificate (11)
State:.receiving Server Hello Done (14)
State:.sending Client Key Exchange (16)
State:.sending Finished (16)
State:.receiving Finished (16)
[HTTP-Client] connected to raw.githubusercontent.com:443
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Security-Policy: default-src 'none'; style-src 'unsafe-inline''
[HTTP-Client][handleHeaderResponse] RX: 'X-XSS-Protection: 1; mode=block'
[HTTP-Client][handleHeaderResponse] RX: 'X-Frame-Options: deny'
[HTTP-Client][handleHeaderResponse] RX: 'X-Content-Type-Options: nosniff'
[HTTP-Client][handleHeaderResponse] RX: 'Strict-Transport-Security: max-age=31536000'
[HTTP-Client][handleHeaderResponse] RX: 'ETag: "9a1278562c6f247c7897ef76a9c45c9fdac4af21"'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: application/octet-stream'
[HTTP-Client][handleHeaderResponse] RX: 'Cache-Control: max-age=300'
[HTTP-Client][handleHeaderResponse] RX: 'X-GitHub-Request-Id: 67F5E015:68C0:272B0E7:57398E67'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 223408'
[HTTP-Client][handleHeaderResponse] RX: 'Accept-Ranges: bytes'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Mon, 16 May 2016 09:12:31 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Via: 1.1 varnish'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'X-Served-By: cache-itm7427-ITM'
[HTTP-Client][handleHeaderResponse] RX: 'X-Cache: HIT'
[HTTP-Client][handleHeaderResponse] RX: 'X-Cache-Hits: 1'
[HTTP-Client][handleHeaderResponse] RX: 'Vary: Authorization,Accept-Encoding'
[HTTP-Client][handleHeaderResponse] RX: 'Access-Control-Allow-Origin: *'
[HTTP-Client][handleHeaderResponse] RX: 'X-Fastly-Request-ID: e112eb24a13c03efc43b78037efdc30ce39c5226'
[HTTP-Client][handleHeaderResponse] RX: 'Expires: Mon, 16 May 2016 09:17:31 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Source-Age: 152'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 223408
sleep disable
ssl->need_bytes=16432 > 6859
pm open,type:0 0
Alert: close notify
[HTTP-Client][end] tcp is closed
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
@cp:0
ld
```

The `@cp:0` line indicates that the sketch was updated successfully.

Zyxmn, 2016-05-16:
Although my sketch doesn't seem to work, the sketch you have uploaded works perfectly.
I need to remember to put restart commands in pre and post ota sketches.

 Thanks for the help @igrr !

igrr, 2016-05-16:
ESP requires manual reset after UART upload, but not after OTA upload.

Your boot mode (`1,6`) indicates that either you didn't reset ESP after UART upload, or GPIO0 was low at the time when it got reset after OTA.

Zyxmn, 2016-05-16:
Not Resetting after UART upload was the reason.  OTA works perfectly now. Cheers!

closed 2016-05-16 by Zyxmn

bp28203, 2016-06-24:
Can anyone help me with this issue...

I ahve hosted the bin file on Microsft IIS server at this location  http://192.168.0.5:912/blinkredLedFast.bin.

If I try to access this link from browser I do get the download option. However if I use it in the code I get error as 

HTTP_UPDATE_FAILD Error (-1): HTTP error: connection refused

I tried replacing the below code

t_httpUpdate_return ret = ESPhttpUpdate.update("http://192.168.0.5:912/blinkredLedFast.bin","80");

with 
t_httpUpdate_return ret = ESPhttpUpdate.update("https://raw.githubusercontent.com/Zyxmn/esp8266_BlinkyBin/master/Blinky.cpp.bin", "", "21 99 13 84 63 72 17 13 B9 ED 0E 8F 00 A5 9B 73 0D D0 56 58");

and bin files get downloaded succeffuly.

kenpeter, 2016-08-03:
@igrr 

> ESP8266HTTPUpdate library doesn't support HTTP redirects at this point. It expects an URL which points directly to the binary.

There is a section called "Advanced updater" in https://esp8266.github.io/Arduino/versions/2.0.0/doc/ota_updates/ota_updates.html

Also in .arduino15/packages/esp8266/hardware/esp8266/2.3.0/doc/ota_updates/readme.md, line 342

It seems I can create a php script, then when hit this script, firmware will be downloaded, no need to hit the direct bin file in server. Am I correct?

Because I have a URL in laravel 5, which allows the firmware to be download dynamically. Currently, I am not able to make it working. In your comment above, esp update only allows direct hit to the bin file. Are you able to clarify?

igrr, 2016-08-03:
Sorry, i'm not familiar with php and laravel. Currently HTTP response has to contain the actual binary, not a redirect to another URL. I.e. response code has to be 20x and with binary file in the body of response.

kenpeter, 2016-08-03:
Hi,

This the url: https://toothfi.com/device/device/wifi_gateway/set_me_up/prod/get_firmware

1. I test this: curl -v https://toothfi.com/device/device/wifi_gateway/set_me_up/prod/get_firmware > out.txt
HTTP/1.1 200 OK and out.txt is the binary file.

2. Try the code below, error: HTTP_UPDATE_FAILD Error (-1): HTTP error: connection refused

```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  WiFi.begin("xyz", "xyz");
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
  }
  Serial.println("Connected, updating");
  delay(1000);

  t_httpUpdate_return ret = ESPhttpUpdate.update("https://toothfi.com/device/device/wifi_gateway/set_me_up/prod/get_firmware");

  Serial.println(F("\n---- ota update done ----\n"));
  switch(ret) 
  {
    case HTTP_UPDATE_FAILED:
      Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
	    break;
    case HTTP_UPDATE_NO_UPDATES:
      Serial.println(F("HTTP_UPDATE_NO_UPDATES"));
      break;
    case HTTP_UPDATE_OK:
      Serial.println(F("HTTP_UPDATE_OK"));
      break;
    default:
	    Serial.printf("Unexpected response code %d from ESPhttpUpdate.update\n",(int)ret);
	    break;
	
  }
}
void loop() {
}
```

3.The header I am using
```
// https://stackoverflow.com/questions/4345322/how-can-i-allow-a-user-to-download-a-file-which-is-stored-outside-of-the-webroot
  private function get_firmware_download($file) {
    if (file_exists($file)) {
      header('Content-Description: File Transfer');
      header('Content-Type: application/octet-stream');
      header('Content-Disposition: attachment; filename='.basename($file));
      header('Content-Transfer-Encoding: binary');
      header('Expires: 0');
      header('Cache-Control: must-revalidate, post-check=0, pre-check=0');
      header('Pragma: public');
      header('Content-Length: ' . filesize($file));
      ob_clean();
      flush();
      readfile($file);
      exit;
    }
    else {
      echo "no such file: $file";
      die;
    }
  }

```

igrr, 2016-08-03:
Since you're using HTTPS, you need to provide HTTPS certificate fingerprint as third argument to `update`.

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h#L78-L79

DavidFi1, 2016-12-01:
> igrr commented on Aug 3
> not a redirect to another URL. I.e. response code has to be 20x and with binary file in the body of response.

Can you please fix this issue?
We will be able to use google drive etc...
It could be very very useful !!!

Thank you.
David.

mugginsjm, 2017-03-08:
Hi maybe someone can help me. I have had some success with OTA but not with this sketch
I don't know what the problem is here?... I get:
thanks

[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate] - code: 200
[httpUpdate] - len: 360736
[httpUpdate] ESP8266 info:
[httpUpdate] - free Space: 2781184
[httpUpdate] - current Sketch Size: 363904
[httpUpdate] runUpdate flash...
sleep disable
[begin] roundedSize: 0x00059000 (364544)
[begin] updateEndAddress: 0x00300000 (3145728)
[begin] currentSketchSize: 0x00059000 (364544)
[begin] _startAddress: 0x002A7000 (2781184)
[begin] _currentAddress: 0x002A7000 (2781184)
[begin] _size: 0x00058120 (360736)
Fatal exception 0(IllegalInstructionCause):
epc1=0x40201014, epc2=0x00000000, epc3=0x400043da, excvaddr=0x00000000, depc=0x00000000

Exception (0):
epc1=0x40201014 epc2=0x00000000 epc3=0x400043da excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffefb30 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffefcd0: 00000000 00000000 00000000 401052a1 
3ffefce0: 4010644f 00000000 00000000 00000000 - See more at: http://www.esp8266.com/viewtopic.php?f=28&t=11660&p=63461#p63461

Cristianistrate, 2017-04-02:
Hi, I got "HTTP_UPDATE_FAILD Error (-1): HTTP error: connection refused"
I use " t_httpUpdate_return ret = ESPhttpUpdate.update("http://10.10.10.20",80,"/ota.php");"
The php will give me the right file.
on CHROME it says "only for ESP8266 updater!" but I got error on esp.
Please help.

pieman64, 2017-04-02:
@Cristianistrate doesn't the port need to be "80" rather than 80?
Are you able to provide details of the headers you are using in your sketch and your PHP script?

Cristianistrate, 2017-04-02:
Tried "80". same error.

kandev, 2017-04-28:
I guess the error is because you check for "HTTP_X_ESP8266_SKETCH_MD5" header.
Although it's written in all the examples around the net, it's never sent by ESP8266. Maybe it's bug, maybe I'm doing it wrong, but HTTP_X_ESP8266_SKETCH_MD5 header is missing on my ESPs.

davisonja, 2017-04-30:
It's supplied if you're using the git version.

cirdiam, 2017-05-06:
I can get OTA updates working using http, but not using https
I'm using Arduino IDE 1.6.9, ESP8266 cores 2.3.0, an ESP8266-12E

code is:
#include <Arduino.h>

#include <ESP8266WiFi.h> 
#include <ESP8266WiFiMulti.h> 

#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
#include <WiFiClientSecure.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti; //use this when using ESP8266WiFiMulti.h

const char* ssid = "...my ssid..";
const char* pass = "..my router password";
const char* host = "www.aquaswift.ca";  
const int httpsPort = 443;
const char* fingerprint = "fe a4 1f 46 64 35 19 6a d7 5a 08 a2 9d 20 62 24 6b e9 fd ef"; 

void setup() {

 //setup serial port
    USE_SERIAL.begin(115200);
    USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

  //connect to wifi
    WiFiMulti.addAP(ssid, pass); //use this when using ESP8266WiFiMulti.h
    
    // wait for WiFi connection
    Serial.println();    
    Serial.print("Waiting for wifi");    
    while (WiFiMulti.run() != WL_CONNECTED) { //use this when using ESP8266WiFiMulti.h
      Serial.print(".");
      delay(500);    
    }
    Serial.println();
    Serial.println("WiFi connected");
    Serial.println();
    Serial.print("WiFi connected.  IP address:");
    Serial.println(WiFi.localIP()); 

  //establish a connection to remote web server
    // Use WiFiClientSecure class to create TLS connection
    WiFiClientSecure client;
    Serial.print("connecting to ");
    Serial.println(host);
    if (!client.connect(host, httpsPort)) {
      Serial.println("connection failed");
      return;
    }
  
    if (client.verify(fingerprint, host)) {
      Serial.println("certificate matches");
    } else {
      Serial.println("certificate doesn't match");
    }
[... 393 lines cut ...]
aid 16
cnt 

connected with 2GHz, channel 6
dhcp client start...

======running flash config example gives the following output:
Flash real id:   001640E0
Flash real size: 4194304

Flash ide  size: 4194304
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration ok.

=============
I've tried it with 2 different ESP8266-12E chips - both give same results
I've tried running off fully charged 6V battery thru a 1amp 3.3v regulator
and running off bench supply set to deliver 1amp at 5v thru 1amp 3.3v regulator
Same problem every time.  But works fine if I use http instead of https.  almost seems like there's a bug in ESP8266httpUpdate, as it works with http, but not https.  Also I suspect the following is the key message in the data above: 

Fatal exception 29(StoreProhibitedCause):
epc1=0x4000e1b2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Anybody know what the fix is?