esp8266/Arduino #2024: Print.cpp strange random problems

PhilColbert, 2016-05-13:
Hi

Running a nodemcu at 160hmz, standard setup, randomly in some hours of running this pops up, I cant work out why :(

It happens totally randomly after executing the same code thousands of times.....

Any ideas?

Thanks

Code :-

    uint8_t sbuf[wlen];
    char cbuf[wlen];
    
    client.readBytes(sbuf, wlen);     

Problem on 2.2 :-

0x4020fab9: Print::printf(char const*, ...) at C:\Users\Fugazi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\cores\esp8266/Print.cpp line 88
0x4020b5e9: ClientContext::_consume(unsigned int) at C:\Users\Fugazi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x4020b64a: ClientContext::read() at C:\Users\Fugazi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
:  (inlined by) WiFiClient::read() at C:\Users\Fugazi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\libraries\ESP8266WiFi\src/WiFiClient.cpp line 224
0x4020fcb3: Stream::readBytes(char*, unsigned int) at C:\Users\Fugazi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\cores\esp8266/Stream.cpp line 68

Exception (28):
epc1=0x4020fb03 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: sys 
sp: 3fff1710 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fff18b0:  3fff1ecc 00000008 3fff18c0 4020fab9 <
3fff18c0:  3137325b 5d333131 49202d20 7220206e  
3fff18d0:  41646165 61446c6c 6e4f6174 69666957  
3fff18e0:  696f5020 3220746e 0a0d2061 00000a00  
3fff18f0:  3fff23f0 3fff479c 3fff5aa4 4020b5e9  
3fff1900:  3fff1940 3fff1910 00000008 4020b64a  
3fff1910:  00000000 00000000 00042309 00000003  
3fff1920:  00000412 000000a8 00000a8d 4020fcb3  
3fff1930:  3ffea8b0 00000000 3fff1910 3fff1940  
3fff1940:  3fff2f08 00000a8e 3fff2f14 3fff2e80  
3fff1950:  3fff2f08 00000a8e 3fff2f14 4020a5e7  
3fff1960:  00000000 00000000 00000000 00000000  
3fff1970:  00000000 00000000 00000000 00000000  
3fff1980:  00000000 00000000 00000000 00000000  
3fff1990:  00000000 00000000 00000000 00000000  
3fff19a0:  00000000 00000000 00000000 00000000  
3fff19b0:  00000000 00000000 00000000 00000000  
3fff19c0:  00000000 00000000 00000000 00000000  
3fff19d0:  00000000 00000000 00000000 00000000  
3fff19e0:  00000000 00000000 00000000 00000000  
3fff19f0:  00000000 00000000 00000000 00000000  
3fff1a00:  00000000 00000000 00000000 00000000  
3fff1a10:  00000000 00000000 00000000 00000000  
3fff1a20:  00000000 00000000 00000000 00000000  
3fff1a30:  00000000 00000000 00000000 00000000  
3fff1a40:  00000000 00000000 00000000 00000000  
3fff1a50:  00000000 00000000 00000000 00000000  
3fff1a60:  00000000 00000000 00000000 00000000  
3fff1a70:  00000000 00000000 00000000 00000000  
3fff1a80:  00000000 00000000 00000000 00000000  
3fff1a90:  00000000 00000000 00000000 00000000  
3fff1aa0:  00000000 00000000 00000000 00000000  
3fff1ab0:  00000000 00000000 00000000 00000000  
3fff1ac0:  00000000 00000000 00000000 00000000  
3fff1ad0:  00000000 00000000 00000000 00000000  
3fff1ae0:  00000000 00000000 00000000 00000000  
3fff1af0:  00000000 00000000 00000000 00000000  
3fff1b00:  00000000 00000000 00000000 00000000  
[... 179 lines cut ...]
3fff35d0:  00000000 00000000 00000000 00000000  
3fff35e0:  00000000 00000000 00000000 00000000  
3fff35f0:  00000000 00000000 00000000 00000000  
3fff3600:  00000000 00000000 00000000 00000000  
3fff3610:  00000000 00000000 00000000 00000000  
3fff3620:  00000000 00000000 00000000 00000000  
3fff3630:  00000000 00000000 00000000 00000000  
3fff3640:  00000000 00000000 00000000 00000000  
3fff3650:  00000000 00000000 00000000 00000000  
3fff3660:  00000000 00000000 00000000 00000000  
3fff3670:  00000000 00000000 00000000 00000000  
3fff3680:  00000000 00000000 00000000 00000000  
3fff3690:  00000000 00000000 00000000 00000000  
3fff36a0:  00000000 00000000 00000000 00000000  
3fff36b0:  00000000 00000000 00000000 00000000  
3fff36c0:  00000000 00000000 00000000 00000000  
3fff36d0:  00000000 00000000 00000000 00000000  
3fff36e0:  00000000 00000000 00000000 00000000  
3fff36f0:  00000000 00000000 00000000 00000000  
3fff3700:  00000000 00000000 00000000 00000000  
3fff3710:  00000000 00000000 00000000 00000000  
3fff372

ctx: sys 
sp: 3fff2fd0 end: 3fffffb0 offset: 01b0

PhilColbert, 2016-05-13:
I have just found out why this  happening, but cant understand it? 

I think its because the amount of data on a socket coming in is too big and must overflow a buffer ? The strange thing is the socket is constantly checked for data, usually it has a few bytes in the data stream and then suddenly it has no bytes for 20 checks and then 1500 bytes appear, which causes this problem.

Any ideas ?

igrr, 2016-05-13:
Well 1460 is max TCP packet size, so data can come in a chunk that large. But I wonder why this causes an exception. Can you post the full code?

PhilColbert, 2016-05-13:
Hi

Its a function with just that in really, forgive the poor coding, still learning :) ?

Like I said it can run this same code for over 10 hours, or one hour, and then suddenly throws out the exception.....

What happens if the data is over 1460 ?

#include <Arduino.h>
#include <WiFiClient.h>
#include "AirWare.h"
#include "WifiTools.h"
#include <ESP8266WiFi.h>
#include <DNSServer.h>

extern "C" {
#include <user_interface.h>
}

String readAllDataOnWifi(WiFiClient LK8000Client, String TCPBuffer )
{

  size_t wlen=LK8000Client.available();

  if (wlen!=0)
  {
  
    DEBUG_SERIAL_UART_MAX("[%d] - TCP data bytes available :- %d \r\n", millis(),wlen);
     
    last_data_received=millis();
    uint8_t sbuf[wlen];
    char cbuf[wlen];
    
    LK8000Client.readBytes(sbuf, wlen);        
          
    memset(cbuf, 0, wlen);
     
    memcpy(&cbuf,&sbuf, wlen);
   
    TCPBuffer=TCPBuffer+String(cbuf).substring(0,wlen);
  }
  
  int i;
   
  if ( TCPBuffer.length()<5)
  {
    return TCPBuffer;
  }

  // make 100% sure length>0

  int len=TCPBuffer.length()-4;

  if (len<0)
  {
    len=0;
  }
  
  for (i=0;i<len;i++)
  {
  
    if (TCPBuffer.charAt(i)=='$' && ( TCPBuffer.charAt(i+3)== 'G' || TCPBuffer.charAt(i+3)== 'R' ) && ( TCPBuffer.charAt(i+4)== 'G' || TCPBuffer.charAt(i+4)== 'M' ) )
    {
      break;
    }
  }  

    return TCPBuffer.substring(i);

 }

PhilColbert, 2016-05-17:
Seems for some reason, its to with too many bytes in the wifi client buffer, on some devices when data is being sent over TCP, the data transmission stalls.

The socket is polled every 50ms but client.available reports 0 bytes, then suddenly after many many attempts to read the data , the socket reports many thousands of bytes ( 3000,4000,5000 ) , which then crashes the system....

I think its the fault of the device sending the data as on some devices the data is sent reliably with no stalled data but unfortunately I can not think of a way to solve this problem without just flushing the buffer reporting the thousands of bytes so the Esp doesnt read them..... any ideas?

Thanks

igrr, 2016-05-18:
The issue is not with the amount of data received. The issue is with the way you allocate memory for the received data. When you use a VLA, memory gets allocated on the stack. However you don't have a way of checking whether such an allocation results in a stack overflow. So don't use VLA, use `new` or `malloc` instead. Even better, use statically allocated arrays with constant size and read and data in chunks of array size.

closed 2016-05-18 by igrr