esp8266/Arduino #2025: <bits/cxxabi_tweaks.h> file not found

kriswiner, 2016-05-14:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			custom
Core Version:      	2.2.0

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8285 Module
Flash Size: 		1 MB
CPU Frequency:	80 MHz
Flash Mode:		DOUT
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck 

### Sketch

const uint8_t blue = 15;

void setup() 
{

  Serial.begin(9600);
  pinMode(blue, OUTPUT);
}

void loop() 
{
  digitalWrite(blue, 1);
  delay(100);
  
  Serial.print("test");
  
  digitalWrite(blue, 0);
  delay(900);
}

### Debug Messages

Arduino: 1.6.8 (Windows XP), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (512K SPIFFS), ck, Disabled, None"

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files\Arduino\hardware" -hardware "C:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages" -hardware "C:\Documents and Settings\Kris\My Documents\Arduino\hardware" -tools "C:\Program Files\Arduino\tools-builder" -tools "C:\Program Files\Arduino\hardware\tools\avr" -tools "C:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages" -built-in-libraries "C:\Program Files\Arduino\libraries" -libraries "C:\Documents and Settings\Kris\My Documents\Arduino\libraries" -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=1M512,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10608 -build-path "C:\DOCUME~1\Kris\LOCALS~1\Temp\build4556d5d3442285d468cdd3eb1eaa5bd0.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "C:\Documents and Settings\Kris\My Documents\Arduino\BlinkESP8285\BlinkESP8285.ino"
C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files\Arduino\hardware" -hardware "C:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages" -hardware "C:\Documents and Settings\Kris\My Documents\Arduino\hardware" -tools "C:\Program Files\Arduino\tools-builder" -tools "C:\Program Files\Arduino\hardware\tools\avr" -tools "C:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages" -built-in-libraries "C:\Program Files\Arduino\libraries" -libraries "C:\Documents and Settings\Kris\My Documents\Arduino\libraries" -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=1M512,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10608 -build-path "C:\DOCUME~1\Kris\LOCALS~1\Temp\build4556d5d3442285d468cdd3eb1eaa5bd0.tmp" -warnings=all -prefs=build.warn_data_percentage=75 -verbose "C:\Documents and Settings\Kris\My Documents\Arduino\BlinkESP8285\BlinkESP8285.ino"
"C:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0/tools/sdk/include" "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0/tools/sdk/lwip/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266" "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\variants\generic" "C:\DOCUME~1\Kris\LOCALS~1\Temp\build4556d5d3442285d468cdd3eb1eaa5bd0.tmp\sketch\BlinkESP8285.ino.cpp" -o "nul"
"C:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0/tools/sdk/include" "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0/tools/sdk/lwip/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266" "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\variants\generic" "C:\DOCUME~1\Kris\LOCALS~1\Temp\build4556d5d3442285d468cdd3eb1eaa5bd0.tmp\sketch\BlinkESP8285.ino.cpp" -o "nul"
"C:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0/tools/sdk/include" "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0/tools/sdk/lwip/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266" "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\variants\generic" "C:\DOCUME~1\Kris\LOCALS~1\Temp\build4556d5d3442285d468cdd3eb1eaa5bd0.tmp\sketch\BlinkESP8285.ino.cpp" -o "C:\DOCUME~1\Kris\LOCALS~1\Temp\build4556d5d3442285d468cdd3eb1eaa5bd0.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files\Arduino\tools-builder\ctags\5.8-arduino5/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\DOCUME~1\Kris\LOCALS~1\Temp\build4556d5d3442285d468cdd3eb1eaa5bd0.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0/tools/sdk/include" "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0/tools/sdk/lwip/include" -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L    -DARDUINO=10608 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266" "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\variants\generic" "C:\DOCUME~1\Kris\LOCALS~1\Temp\build4556d5d3442285d468cdd3eb1eaa5bd0.tmp\sketch\BlinkESP8285.ino.cpp" -o "C:\DOCUME~1\Kris\LOCALS~1\Temp\build4556d5d3442285d468cdd3eb1eaa5bd0.tmp\sketch\BlinkESP8285.ino.cpp.o"
"C:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0/tools/sdk/include" "-IC:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0/tools/sdk/lwip/include" -c -g -x as [... cut ...]
In file included from C:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\abi.cpp:23:0:

c:\documents and settings\kris\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\cxxabi.h:50:32: fatal error: bits/cxxabi_tweaks.h: No such file or directory

 #include <bits/cxxabi_tweaks.h>

                                ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

kriswiner, 2016-05-14:
OK, I restarted my computer after going back to 2.0.0 and not i can compile this program without error. Is there something that changed in 2.2.0 that is causing this weird error message?

igrr, 2016-05-16:
Users who had reported similar looking errors had either spaces or non-ASCII characters in their home folder path.

kriswiner, 2016-05-16:
So, you mean that this path:

In file included from C:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\abi.cpp:23:0:

which works on 2.0.0 now has to be changed to:

C:\DocumentsandSettings\Kris\LocalSettings\ApplicationData\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\abi.cpp 

???

igrr, 2016-05-16:
I haven't got a slightest idea why this works for you with 2.0.0 and doesn't work w/ 2.2.0. We haven't changed the toolchain between these versions. I just happened to notice that *usually* this kind of error indicates that there is something wrong with the path. Also you may want to check if the file which it complains about (`cxxabi_tweaks.h`) is actually present. Maybe something was not unpacked correctly?

kriswiner, 2016-05-16:
There seems to be some file confusion in the core. I found the "missing" file here:

C:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\xtensa-lx106-elf\bits

and added it here:

C:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits

and now the blink program compiles under 2.2.0.

Looks like a misplaced or forgotten path error?

BTW, I have been using the newer ESP8285 and noticed that it also required flashing using DOUT instead of the default DIO or QIO. Is there some way we can add to the default boards platform options for DOUT and QOUT?

igrr, 2016-05-16:
I don't think so, lots of people are using this core on Windows and i don't think like everyone has to copy that file, otherwise we would surely had an issue w.r.t. this already. 2.2.0 core has been out for a few months. Also, `Documents and Settings`? Is this Windows XP? I think the folder is called `Users` since Vista.

igrr, 2016-05-16:
Could you please try "portable installation" method into some directory which doesn't contain spaces in its name?
https://github.com/esp8266/Arduino/issues/1631#issuecomment-197800101

igrr, 2016-05-16:
Regarding DOUT/QOUT, yes, I will add those.

kriswiner, 2016-05-16:
Thanks, it's an odd choice Espressif made for an embedded flash to only use
DOUT. Why not QIO, the full quad spi. I wonder then if some of the SPI_Flash
pins are available for GPIO use?

<https://github.com/notifications/beacon/AGY1qv5yusLy12h9Ldb4LYSs5ole4P4Tks5
qCB6FgaJpZM4Iefgm.gif>

kriswiner, 2016-05-16:
Yes, I am using XP. My laptop is 13 years old and still going strong!

<https://github.com/notifications/beacon/AGY1qtBxvk2IPHewLLMoT9BoEvq8NjUGks5
qCB4EgaJpZM4Iefgm.gif>

kriswiner, 2016-05-16:
That seems to work, but I have to reinstall all of my preferences and boards
now!

#1631 (comment)

-
<https://github.com/notifications/beacon/AGY1qnCsZQKgN8WCoizdgz1-JVULUmtwks5
qCB5fgaJpZM4Iefgm.gif>

kriswiner, 2016-05-17:
FYI

Here's a useful dev board for the ESP8266 I designed.

https://www.tindie.com/products/onehorse/esp8285-development-board/

Kris

<https://github.com/notifications/beacon/AGY1qv5yusLy12h9Ldb4LYSs5ole4P4Tks5

closed 2016-06-01 by igrr