esp8266/Arduino #2034: CAN't program the esp8266

abbiy, 2016-05-16:
I am using CP2102 programmer for my ESP8266 project (weather widget). I have checked all the connections & power is also 3.3 volt using voltage converter AMS1171 but still getting the same error -

**warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed**

Please someone help me out with this .

thewhiterabbit, 2016-05-16:
From my experience, make sure you have the correct drivers installed for your programmer. That is the most likely cause. Also I discovered this type of error can occur if you have a bad USB cable.

abbiy, 2016-05-17:
I am using cp2102 & i guess it is functioning ok. When i connect it with my windows 7 (**RXD & POWER LED glows up**) The ESp8266 blue led also blinks .
**what to do now?? :-(**  
 [image]
**HELP**

igrr, 2016-05-17:
Please verify that you have bootstrapping pins connected according to the following schematic:
http://esp8266.github.io/Arduino/versions/2.2.0/doc/boards.html#esp-to-serial

Your CP2102 board lacks DTR and RTS pins, so you either need to place ESP8266 into download mode manually, or solder wires directly to CP2102 DTR and RTS pins.

abbiy, 2016-05-17:
Its my first experience with ESP , can you please elaborate how can i place **ESP8266 into downloading** **mode** manually & which pins i have to solder ? 
**DTS(CP2102)-------RTS(ESP8266)** did u mean this?

igrr, 2016-05-17:
Please check one of many tutorials on the web on how to get started with the ESP8266 :)
For instance, https://hackaday.io/project/5150-arduino-ide-for-esp8266-quickstart-guide
or this one: http://www.whatimade.today/esp8266-easiest-way-to-program-so-far/ (see section titled "Writing program to the ESP8266".

abbiy, 2016-05-17:
**:'(** 
 I had gone through your tutorial and make the circuit just like that -
**GPO0------10K-----SWITCH---GROUND  & REST pin-----RESISTOR-------SWITCH------GRD**
[image]
Anything which i missed ?

igrr, 2016-05-17:
You missed a pull-up resistor on GPIO0, but that shouldn't matter if you are trying to upload code.

Please open Arduino serial terminal, set baud rate in the terminal to 74880 baud, and reset the ESP using S1. S2 should be grounding GPIO0. Then copy&paste here the output you get.

abbiy, 2016-05-17:
74880 baud rate is not showing in my IDE..

igrr, 2016-05-17:
In that case you may need to upgrade your IDE to some recent version. Or use another program which allows such baud rate to be set.
[image]

closed 2016-06-27 by igrr