esp8266/Arduino #2035: TCP  sever   can't  write data to the client

liangzelang, 2016-05-17:
### Basic Infos

#### Hardware
Hardware:		ESP12e
Core Version:      	 

### Description

taday when I try to set my ESP8266 as the TCP server ,
I find  WiFiServer::write(const uint8_t *buffer, size_t size)  hasn't implement in the relate file :WiFiserver.cpp .  so I wonder i must  write the part by myself if I want to use the TCP mode.

 
### Sketch

#include <Arduino.h>
#include "ESP8266WiFi.h"

int status = WL_IDLE_STATUS;
void printWifiStatus();
WiFiServer server(23);

boolean alreadyConnected = false; // whether or not the client was connected previously

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }

  // start the server:
  server.begin();
  // you're connected now, so print out the status:
  printWifiStatus();
 }
 void loop() {
   // wait for a new client:
   WiFiClient client = server.available();

   // when the client sends the first byte, say hello:
   if (client) {
     if (!alreadyConnected) {
       // clead out the input buffer:
       client.flush();
       Serial.println("We have a new client");
       client.println("Hello, client!");
       alreadyConnected = true;
     }

     if (client.available() > 0) {
        Serial.print("get the data of client");
       // read the bytes incoming from the client:
       char thisChar = client.read();
       // echo the bytes back to the client:
       server.write(thisChar);               
       // echo the bytes to the server as well:
       Serial.write(thisChar);
     }
   }
 }

### Debug Messages

```
messages here
```

igrr, 2016-05-17:
Please check WiFiWebServer example of ESP8266WiFi library, it shows how to use WiFiServer.
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiWebServer/WiFiWebServer.ino#L52-L93
You need to use server.available() method to get a WiFiClient object representing single TCP connection.
Then you need to use methods of this client object to send and receive data.

closed 2016-05-17 by igrr

liangzelang, 2016-05-17:
Yeah ,Thank you very muck and I solve this problem .
BUT  ,,,    the connection of server and client will automatically disconnected when the end of loop .I hope the connection will last until closed by myself.