esp8266/Arduino #2036: OTA over own serial implementation?

tuxedo0801, 2016-05-17:
Hi there,

I just stumbled over the idea to do OTA updates not via wifi, but via serial... I guess this might sound strange to you, but here's the idea:

The ESP is running as a KNX home automation node in an existing KNX system and is powered by the KNX bus itself (means: <50mA@3.3V, therefore I need to disable wifi completely).
The ESP communicates via it's UART with an "Knx Transceiver", which let's the ESP access the KNX bus.

The idea now is, to transfer the new firmware over KNX bus telegrams to the ESP, which will then use the received firmware for OTA.

With help of that, one would be able to apply firmware-update via existing KNX bus system (and some special software, which we already have in place)

The OTA documentation just explains the wifi-way of OTA, but keeps the "Stream" interface undocumented. 

My questions are:

* Is it in principle possible to receive with a custom protocol via UART new firmware an apply it to the ESP? (the first part of the question should be possible, then 2nd part is the main-question)
* What about the stream-class?! Can you point me to some code showing/explaining the OTA mechanism so that I might adapt it to receive the new firmware via custom protocol via UART?!

br,
Alex

igrr, 2016-05-17:
If you can write a class derived from `Stream` which will handle reading from Serial and decoding your custom protocol, then you can update the ESP by doing, for example,
```c++
Update.begin(firmwareSize);
Update.writeStream(yourCustomStream);
Update.end();
```
Please check the comments in [Updater.h](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Updater.h).

tuxedo0801, 2016-05-17:
thx, I will checkt this.

tuxedo0801, 2016-05-17:
One more question:

the used protocol cannot transfer the complete firmware as a whole. It's received in smaller parts, with some protocol-overhead between the blocks. 
So it would be easier to write the firmware first to the flash, and the apply the received firmware-file from flash via a kind of "file-stream".

Would this be okay, or does this sound more like a "work-around"?!

igrr, 2016-05-17:
I think you can probably write a wrapper class which will receive these smaller parts and make more data `available` as it gets unpacked. It will read some data from Serial, remove protocol headers, and place that data into some buffer. When UpdaterClass calls `available` or `read` methods of your custom Stream class, you can return the amount of data currently received and unpacked.

tuxedo0801, 2016-05-17:
That sounds like a good/better idea.
Hopefully now the final question:
How about broken data/bad transferred data... What happens if the size of the file is correct, but one or another byte is wrong?! I guess in this case the ESP needs to be flashed again with a programmer to fix it?!
If file is "on the fly" provided to the update class, how about checking a kind of firmware-crc?!

igrr, 2016-05-17:
UpdaterClass has `setMD5` method, which you can use to tell it what MD5 the firmware image is supposed to have. You can pass this data in some initialization packet of your custom protocol.
See this part of ESP8266httpUpdate for an example:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.cpp#L345-L383

tuxedo0801, 2016-05-17:
Thanks a lot for the details. 
--> solved for me.

closed 2016-05-17 by tuxedo0801