esp8266/Arduino #2038: Problem with WIFI WPA personal TKIP

mkgez, 2016-05-17:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			NodeMCU 1 ESP-12E
Core Version:      	        2.2.0

### Description

I have used my Arduino IDE with NodeMcu v1.0 and I have installed esp8266 v2.2.0 but with this version I have some problems to connect to my WIFi that use a WPA personal TKIP encryption, but if I install esp8266 v2.0.0 I can connect without problems to my WIFI.

With version 2.1.0 and 2.2.0 my Wifi.status is always:
Status WIFI -> WL_DISCONNECTED

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		        4M (3M SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	        ?40Mhz?
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

 #include <ESP8266WiFi.h>

char ssid[] = "SSID";     //  your network SSID (name) 
char pass[] = "PASSWORD";  // your network password
int status = WL_IDLE_STATUS;     // the Wifi radio's status

void WIFI() {
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  int j = 1;
  
 // attempt to connect to Wifi network:
  while (WiFi.status() != WL_CONNECTED && j<3) { 
    Serial.print("Attempting to connect to WPA SSID: ");
    Serial.print(ssid);
    // Connect to WPA/WPA2 network:    
    WiFi.begin(ssid,pass);
  int i = 0;
  while (WiFi.status() != WL_CONNECTED && i<10) { 
    Serial.print(".");
    delay(6000);
    i++;
  }
  j++;
  }  
  Serial.println(".");
  
  
  // you're connected now, so print out the data:
  if (WiFi.status() == WL_CONNECTED){
  Serial.print(F("You're connected to the network: "));
  Serial.println(ssid);
    Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());
  }
  else{
  Serial.print(F("It hasn't been possible to connected to the network: "));
  Serial.println(ssid);
  }
}

void setup() {
  // put your setup code here, to run once:
    Serial.begin(9600); 
  while (!Serial) {}
  WIFI();
}

void loop() {
  // put your main code here, to run repeatedly:

}

### Debug Messages

Status WIFI -> WL_DISCONNECTED

igrr, 2016-05-18:
Could you please switch to "Generic ESP8266" module, set reset method to NodeMCU, enable debug output in tools menu (Core+WiFi, Serial), and add `Serial.setDebugOutput(true);` after `Serial.begin`. Then run the sketch and copy&paste here the output from Serial monitor. Thanks.

sramilli, 2016-05-18:
May it be the same problem i have with two different NodeMCU boards?
I get WL_DISCONNECTED the whole time trying to connect to my wifi with different sketches using ESP8266WiFi.h (i also tried to update the ESP8266WiFi lib from the latest on Github).

My debug stat looks like this:
Connecting to sandalo
sta config unchanged....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
wifi evt: 1
STA disconnect: 203
..reconnect
f 0, ....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
wifi evt: 1
STA disconnect: 203
..reconnect
f -180, ....scandone
no sandalo found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
f r0, .....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
wifi evt: 1
STA disconnect: 203
..reconnect
f 0, ....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
wifi evt: 1
STA disconnect: 203
..reconnect
f -180, ....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
wifi evt: 1
STA disconnect: 203
..reconnect
f r0, .....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
wifi evt: 1
STA disconnect: 203
..reconnect
f 0, ....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
wifi evt: 1
STA disconnect: 203
..reconnect
f -180, ....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
wifi evt: 1
STA disconnect: 203
..reconnect
f r0, ....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
wifi evt: 1
STA disconnect: 203
..reconnect
f 0, .....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
wifi evt: 1
STA disconnect: 203
..reconnect
f -180, ....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
..state: 3 -> 0 (4)
reconnect
f r0, wifi evt: 1
STA disconnect: 4
....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
wifi evt: 1
STA disconnect: 203
..reconnect
f 0, .....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
wifi evt: 1
STA disconnect: 203
..reconnect

I used the WiFiClient sketch this time

Board settings:
Board: Generic ESP8266 Module
Flash Mode: DIO
Flash Freq: 40Mhz
CPU Freq: 80 Mhz
Flash Size 4M (3M SPIFFS)
Debug Level: all
Reset M: nodemcu
Programmer: AVRISP mkII

Using ESP8266 Boards v 2.2.0 (tried also 2.1.0) and Arduino 1.6.9

I tried the blinking sketch, and that one works just fine, so i presume my boards settings are correct (is there any documentation about it?).

I tried also  ESP.eraseConfig()  WiFi.persistent(false) in my journey trying to find the solution to the problem, but that was invain.

I also tried to disable the "WPA2-PSK - Wi-Fi Protected Access 2 Pre-Shared Key" on my WiFi, but i couldnt connect anyway.

I hope there is a solution :)

mkgez, 2016-05-19:
I am sorry @igrr because at moment I have changed my router (a newer in which I can't set for encryption type **just** "WPA personal TKIP") so now I can't repeat the problem and show my debug because it works well with WPA2. If you know how to set encryption type on the NodeMCU board I will able to try.

For  @sramilli  I solved temporally my problem down-grounding my version of esp8266 with **2.0.0**. Just with this version I can connect without problems

igrr, 2016-06-02:
@mkgez Okay, feel free to reopen once you reproduce this.

closed 2016-06-02 by igrr

kyberkiller, 2017-05-18:
Why is this closed?