esp8266/Arduino #2039: Problem with ESP-01- FIREBASE ARDUINO - DHT22

SebasPinto, 2016-05-18:
### Basic Infos

#### Hardware
Hardware:			ESP-01

### Description
Im using Arduino Firebase to upload some DHT22 reads, everything works perfect, but after 5 mins more or less, i get this error
Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffda0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff40:  00000000 00000000 00000300 40100705  
3fffff50:  3fffdc00 00000000 3ffee880 40107730  
3fffff60:  40105b6a 05f815c1 3ffef7b8 4020c6fa  
3fffff70:  4021e344 3ffef7b8 4020cbba 00000002  
3fffff80:  4021e36a 3fffdab0 4020ddbb 3fffdab0  
3fffff90:  00000000 3fffdab0 3fff0918 402073ef  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,0)

**I've been trying different configurations, in software and in hardware, adding and removing things, but the only different result is another error:**

    Soft WDT reset

    ctx: cont sp: 3fff12e0 end: 3fff1810 offset: 01b0

            stack>>> 3fff1490: e8b3c608 2b4b0704 6e02165b 28e606bd
            3fff14a0: 00000630 5c5c5c5c 5c5c5c5c 40100537
            3fff14b0: 000005b4 00000000 00000624 401006c2
            3fff14c0: 5c5c5c5c 5c5c5c5c 00000000 40107690
            3fff14d0: 82d9ac62 41216d6e 04687c31 4022deb6
            3fff14e0: 0000005a 36363636 40230920 3fff2d34
            3fff14f0: 00000000 00000000 43db763c 8f8d3bde
            3fff1500: 00000135 3fff1520 40230b41 0b0b0b0b
            3fff1510: 00000001 d3a93166 43db763c 8f8d3bde
            3fff1520: 3b440000 00000000 00000000 de3b8d8f
            3fff1530: 3b445559 6631a9d3 3fff30c4 00000000
            3fff1540: 00000001 402035c7 3fff2e3c 00000000
            3fff1550: 00000135 00000135 00000135 00000135
            3fff1560: 00000000 00000000 00000000 00000001
            3fff1570: 00000135 8f8d3bde 00000120 00000000
            3fff1580: 3fff2ffc 00000000 00000135 40222198
            3fff1590: 01020317 0000a000 00001410 00001410
            3fff15a0: 00000100 00000017 00000010 3fff5104
            3fff15b0: 00000110 3fff1628 3fff1628 00000000
            3fff15c0: 3fff2ffc 3fffb8fc 00000100 4022221f
            3fff15d0: 00000100 00000711 00000711 4010020c
            3fff15e0: 00000001 3ffe8fdd 3fff50fc 3fff5164
            3fff15f0: 00000001 3ffe8fdd 3fff2c84 40203050
            3fff1600: 00000001 3ffe8fdd 00000000 40205419
            3fff1610: 00000000 00000000 00000000 00000000
            3fff1620: 00000000 00000000 3fffb8fc 0000010f
            3fff1630: 00000100 00000005 3fff80e4 401008c8
            3fff1640: 3fff7b9c 00000005 3fff1660 4020644c
            3fff1650: 00000001 00000005 3fff29a0 402059f0
            3fff1660: 00000000 00000000 00000000 00000000
            3fff1670: 00000000 00000000 3fff16c0 402064f7
            3fff1680: 3fff7b9c 00000000 3fff16c0 40206529
            3fff1690: 3ffe8dc9 3fff299c 3fff16c0 00000000
            3fff16a0: 3ffe8dc9 3fff299c 3fff1760 40203ffe
            3fff16b0: 00000030 3fff05b0 3fff1760 40203a15
            3fff16c0: 3fff7b9c 0000000f 00000003 40203ccb
            3fff16d0: 42013333 0000001c 00000000 00000001
            3fff16e0: 00000000 3fff173c 3f000200 4020468f
            3fff16f0: 00000000 3fff8a9c 0000005f 00000052
            3fff1700: 3fff1778 3fff176c 3fff17c0 3fff05a4
            3fff1710: 3fff17a0 3fff173c 40203fe4 402047a3
            3fff1720: 0000002c 3fff17c0 3fff05a4 42013333
            3fff1730: 3fff17d8 3fff17c0 3fff1760 40203bd3
            3fff1740: 3fff299c 000038e8 0000071d 0000071d
            3fff1750: 3ffe8df8 3fff17c0 3fff05a4 40203e81
            3fff1760: 3ffe8df8 3fff299c 00000000 3fff2b94
            3fff1770: 0000000f 00000000 3fff7f1c 0000000f
            3fff1780: 00000000 3ffe8de8 3fff17c0 00000000
            3fff1790: 00000100 0000000a 3fff17c0 402068b6
            3fff17a0: 3fff5164 0000000f 00000005 40206885
            3fff17b0: 425b3333 0000000a 3fff05a4 40202292
            3fff17c0: 3fff861c 0000001f 0000001e 3fff897c
            3fff17d0: 0000001f 00000014 0000000a 42013333
            3fff17e0: 00000000 00000000 00000001 3fff07e8
            3fff17f0: 3fffdad0 00000000 3fff07e0 40206e28
            3fff1800: feefeffe feefeffe 3fff07f0 40100958
            << 

    ets Jan 8 2013,rst cause:2, boot mode:(1,0)

    ets Jan 8 2013,rst cause:4, boot mode:(1,0)

    wdt reset

I don't have any idea what could be the problem. I have tried with example sketch from Firebase Arduino library, but same result, so i don't think my sketch is the problem

### Settings in IDE

Module:  			
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

#include <ESP8266WiFi.h>
#include <FirebaseArduino.h>
#include <DHT.h>

#define DHTPIN 2
#define DHTTYPE DHT22

DHT dht(DHTPIN, DHTTYPE, 30);

void setup() {

  Serial.begin(115200);

  dht.begin();
  // connect to wifi.
  WiFi.begin("UNE_HFC_0E77", "X98N7JAG");
  Serial.print("connecting");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("connected: ");
  Serial.println(WiFi.localIP());

  Firebase.begin("birdmaker.firebaseio.com", "DtbC2DzxSrbmST9qzzqOAvDtRNtSI7F1UxzjN6TP");
}
bool incuID = 0;

void loop() {
  float h = dht.readHumidity();

  float t = dht.readTemperature();

  String name = Firebase.push("data", 0);
  Serial.print("pushed: /logs/");
  delay(1000);

  Serial.println(name);

  if (Firebase.failed()) {
    Serial.print("setting /number failed:");
    Serial.println(Firebase.error());
    return;
  }

  if (!incuID) {
    Firebase.set("data/" + name + "/spTemp", 37.5);
    delay(1000);
    Firebase.set("data/" + name + "/spHum", 65);
    delay(1000);
    Firebase.set("data/" + name + "/day", 7);
    delay(1000);
    Firebase.set("data/" + name + "/eclDay", 65);
    delay(1000);
    Firebase.set("data/" + name + "/spec", "Pollo");
    delay(1000);
    Firebase.set("data/" + name + "/eggQ", 30);
    delay(1000);
    incuID = true;
    if (Firebase.failed()) {
      Serial.print("setting /number failed:");
      Serial.println(Firebase.error());
      return;
    }

  } else {
    Firebase.set("data/" + name + "/temp", t);
    Serial.println(t);
    delay(1000);
    Firebase.set("data/" + name + "/hum", h);
    Serial.println(h);
    delay(1000);
    if (Firebase.failed()) {
      Serial.print("setting /number failed:");
      Serial.println(Firebase.error());
      return;
    }
  }
  yield();  //this are tries, but with or whithout this, same result
  delay(1000);  //this are tries, but with or whithout this, same result

igrr, 2016-05-18:
Could you please decode the stack dump and paste the result here? Thanks.
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

proppy, 2016-05-19:
Asked the same thing on https://github.com/googlesamples/firebase-arduino/issues/158, I'm happy to diagnose the problem there first (I suspect it's an issue on our end) and escalate if we identify an issue with the core.

mkeyno, 2016-05-19:
I have same problem but when I use the logic level converter , problem is gone (sensor vcc 5 volt )

SebasPinto, 2016-05-19:
@mkeyno So the problem is the Voltage?, even if i'm using and independent source with 3.3V?

mkeyno, 2016-05-19:
yes, but yet not figure that out why the ESP module has problem with my all old fashion Arduino 5 volt supply modules , I might add the ESP pin so fragile regards the old Arduino board (electrically)