esp8266/Arduino #2041: arduino OTA network port doesn't show up

lorenzotacconi, 2016-05-19:
### Basic Infos

#### Hardware
Hardware:		ESP-12E
Core Version:      	2.2.0

### Description

I have download through serial the sketch BasicOTA and at reset the board is conneting to AP :
Booting
Ready
IP address: 192.168.1.121

On Arduino Tool menu never shows up a network port even restarting Arduino IDE

Running espota.py from command line OTA software update is performed corrrectly.
The problem seems to be limited to the discover of the network port

### Settings in IDE
Arduino 1.6.9
Module:  			NodeMCU 1.0 
Flash Size: 		4MB (3M SPIFFS)
CPU Frequency:	80Mhz
Upload Using:		OTA

igrr, 2016-05-19:
Most of the time, this means that there is something blocking mDNS packets on the network.
- try disabling the firewall on the PC
- check WiFi router configuration, it may have some option like "client isolation", try disabling it
- you may try using some app like "Bonjour browser" for Android. If the ESP is not visible in Bonjour browser, then the issue is likely not on the PC side (i.e. either router is blocking mDNS or there's something wrong on the ESP side).

lorenzotacconi, 2016-05-19:
Thanks for the help, I have solved the problem, it was related to router configuration filtering MDNS responses

closed 2016-05-19 by lorenzotacconi