esp8266/Arduino #2043: PubSubClient Slow Publishing

AndyKirby, 2016-05-19:
Guys

First off great work on all of this as ever, and I think this may be
more of a feature than bug.

I am working on a NoT app specifically a sensor node disguised as a
table lamp, nothing too world moving. Based on an ESP8266-07.

I have got aways into the prototype code and am seeing something I need
to fix, probably in the code I am writing.

I have a scheduler doing multitasking and the scheduler slots are
currently pegged at 10mS (10,000 uS):-

#define Interval 10000UL // Number of microseconds that each scheduler
slot should take to complete

The MQTT code is currently:-

https://github.com/knolleary/pubsubclient

The current state of my prototype (very much work in progress) code is
here :-

https://github.com/AndyKirby/Firmware/tree/master/NoTLamp

I put it out there as I go along.

Although it is currently all giving the results I would expect, watching
the serial diagnostics I can see some of the scheduler slots over
running and it is usually always the slots that are doing the Publish :-

Overrun: slot 3 by 35157
Tick
Tock
Tick
Tock
Tick
Tock
Tick
Tock
Tick
Tock
Tick
Tock
Tick
Tock
Tick
Overrun: slot 107 by 31004
Tock
Tick
Tock
Tick
Tock
Overrun: slot 3 by 17025
Tick
Tock
Tick
Tock
Tick
Tock
Tick
Tock
Tick
Tock
Overrun: slot 3 by 34070
Tick
Tock
Tick
Tock
Tick
Tock
Tick
Tock
Tick
Tock
Tick
Tock
Tick
Tock
Tick
Overrun: slot 107 by 35157
Tock

Tick and Tock are 1 second markers so I can see the code is running. The
over run times are in uS as this is what I time the slots in. The slot
number matches up to the switch statement for the very simple DIY
scheduler. These over run times sometimes become as large as 602 mS. But
are never less than 14mS or so on a 10mS slot giving a best performance
of 24mS, per publish. Variability could be down to network/server
latency but is so big on an unloaded network plus very unbusy server
that I suspect something else is going off here and the result is
multiples of something slow.

I suspect that the publish code is blocking until the message has been
published and this is messing with my scheduler timings. Probably a
feature then. TBH I would have expected that it was handing off the
packet/s to the wifi layer and not blocking so much

Questions, is there a better way for me to be using this library, or is
there an MQTT library already out there which handles this better ??
(Better for my application rather than anything else).

Apologies about asking awkward questions, but I figured you guys with
your in depth knowledge of all that is going off might have some ideas
to share on this.

Thanks in advance.

Cheers

Andy Kirby

pajama, 2016-06-17:
Also experiencing the same issue. Would love to know if you found any resolve to this!

AndyKirby, 2016-06-19:
TBH I have not.

Project time is a touch over tight, but my next attempt will be to try
Ilroy's library and see if it performs better.

You have been the first response I have received to the question.

I suspect that the PubSub library is waiting for the wifi interaction to
complete before it returns and this is creating the delay, rather than
passing it to the wifi layer and getting on with something else and
letting the wifi bit take care of itself.

It is a touch awkward for the project I am working on as it is a multi
tasking project which this interferes with. A lot of time is being taken
up rewriting or tweaking libraries to get rid of hidden internal delays
so that the multitasking will work right.

I guess example libs are not written with multitasking in mind.

Currently just coming to the end of pillaging the DHT22 library.

Cheers

Kirbs

igrr, 2016-06-20:
PubSubClient uses WiFiClient, so it blocks execution on every write. If you need non-blocking behaviour, there is another MQTT library which works via callbacks:
https://github.com/i-n-g-o/esp-mqtt-arduino

closed 2016-06-23 by igrr

AndyKirby, 2016-06-23:
Just seen your reply Iggr, will check it out. Many thanks.

Kirbs

igrr, 2016-06-23:
You may also check git version of the core with the normal PubSubClient — `WiFiClient::write` just got improved, it should not block for small writes any more. I haven't done benchmarks on MQTT performance, but in a test which does `client.print` a bunch of times delays went down significantly.

AndyKirby, 2016-06-23:
Cool

I will try this first, although having started reading the INGo mqtt
stuff looks to be very promising for multitasking too.

The help is very much appreciated as is this project.

Cheers

Kirbs

Mattioli, 2018-06-13:
Hey, @AndyKirby, did you have any success on this matter? We're experiencing the same issue on our project.