esp8266/Arduino #2046: Arduino OTA network port doesn't show up anymore after 1.6.9 / 2.2.0 update

obstler42, 2016-05-20:
#### Hardware
Hardware:			ESP-12e
Core Version:      	2.2.0

### Settings in IDE
Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4M (3M SPIFFS)
CPU Frequency:		80Mhz
Port:    COM3 (not showing Network!)

### Sketch
BasicOTA

I had OTA updates working fine with Arduino IDE 1.6.7 and esp8266 code 2.0.0 setup. Now I've updated to IDE 1.6.9 and code 2.2.0, and the network port just won't show up in the IDE anymore..?

Serial monitor shows the board IP, I can ping it, and bonjour browser sees it, just the Arduino IDE doesn't.

Is there any way to debug the discovery process in the IDE, to figure out what's going on?

obstler42, 2016-05-20:
After a lot more fiddling around I think I found the cause for that problem: setting the hostname with underscores _ causes it not to show up in the IDE anymore with 1.6.9/2.2.0. I can perfectly reproduce that. This works:

    //Hostname defaults to esp8266-[ChipID]
    ArduinoOTA.setHostname("esp8266-OTA-40");

and this doesn't:

    ArduinoOTA.setHostname("esp8266_OTA_40");

When the hostname contains _ it simply will not show up in IDE 1.6.9 with core 2.2.0, while it worked just fine with 1.6.7/2.0.0.

For now I've just switched to using - in the hostname

igrr, 2016-05-20:
This may have something to do with filtering of ports list in the IDE. At
some point between 1.6.8 and 1.6.9 my CP2102 ports disappeared from ports
list. They had underscores in device name. I opened an issue on Arduino
tracker and this was fixed a few days later.

Perhaps network ports are triggering a different code path which still
exhibits the issue.

hallard, 2016-05-25:
Guys,

I spend some time yesterday because I had the same problem with _ (I just saw this issue) and I checked RFC, and guess what? underscore may works (or not) depending on device and code you use but are not allowed in DNS hostname, so library works as it should, but good to know it's not allowed and it's not working with Arduino MDNS.

See [RFC1912](https://www.ietf.org/rfc/rfc1912.txt)

```
   Allowable characters in a label for a host name are only ASCII
   letters, digits, and the `-' character.  Labels may not be all
   numbers, but may have a leading digit  (e.g., 3com.com).  Labels must
   end and begin only with a letter or digit.  See [RFC 1035] and [RFC
   1123].  (Labels were initially restricted in [RFC 1035] to start with
   a letter, and some older hosts still reportedly have problems with
   the relaxation in [RFC 1123].)  Note there are some Internet
   hostnames which violate this rule (411.org, 1776.com).  The presence
   of underscores in a label is allowed in [RFC 1033], except [RFC 1033]
   is informational only and was not defining a standard.  There is at
   least one popular TCP/IP implementation which currently refuses to
   talk to hosts named with underscores in them.  It must be noted that
   the language in [1035] is such that these rules are voluntary -- they
   are there for those who wish to minimize problems.  Note that the
   rules for Internet host names also apply to hosts and addresses used
   in SMTP (See RFC 821).
```

closed 2016-06-01 by igrr

Nakul93, 2017-08-22:
Run the espota.exe script in cmd this resolved the issue for me.

JiriBilek, 2018-03-01:
Thanks guys,
this issue resolved my problem, too.
Just posting to say thank you.

MHz000, 2018-03-19:
I'm too have this problem with IDE 1.8.5 altought my hostname doesn't contain an underscore it siply reads "esp8266No1". The chip is assigned an IP-address and is visible by the WLAN router.
Any idea?

atary75, 2018-03-19:
I'm also having this issue on 1.8.5 portable, but it seems not related with the IDE bits, as I can have them to work, just by copying the files into another PC also with Win10 (meaning not the Arduino version or Win10). Any idea, how can I debug this issue? when Arduino IDE looks for Network Ports availability? Any debug flag to trace it?

MHz000, 2018-03-19:
this is the serial monitor output:
...
Booting
Connection Failed! Rebooting...

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
Booting
Connection Failed! Rebooting...
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
...
Any hit how to deal with this message, how to ackt?

Btw. I'm on a WIN 10 system how to do this:
"Run the espota.exe script in cmd"

Thank-you

MHz000, 2018-03-19:
I can ping the ESP:
Ping wird ausgeführt für 192.168.178.63 mit 32 Bytes Daten:
Antwort von 192.168.178.63: Bytes=32 Zeit=3ms TTL=255
Antwort von 192.168.178.63: Bytes=32 Zeit=21ms TTL=255
Antwort von 192.168.178.63: Bytes=32 Zeit=11ms TTL=255
Antwort von 192.168.178.63: Bytes=32 Zeit=189ms TTL=255

Ping-Statistik für 192.168.178.63:
    Pakete: Gesendet = 4, Empfangen = 4, Verloren = 0
    (0% Verlust),
Ca. Zeitangaben in Millisek.:
    Minimum = 3ms, Maximum = 189ms, Mittelwert = 56ms

So it looks like an IDE problem to me but not sure of.

atary75, 2018-03-19:
After a few hours trying, I've finally made it to work! It seems my switch was blocking this type of traffic between my laptop and the access point where the ESP was speaking... Connecting my laptop directly to the same access point fixed the issue to me. In summary, the wired connection between my laptop and the router (access point) has a switch in the middle, that it seems blocking these packages. Lesson of the day: PC should be directly connected to the same AP as ESP :) hope this helps others!

MHz000, 2018-03-19:
Thank-you for sharing your finding however as my router (fritz.box) see eg
listen the MAC address and the allocated IP address of the ESP it looks
like both router and ESP are in the same WLAN. Thats the way I set up the
ESP. I too can ping the ESP at the given IP address with success.
Any idea?
Thanks

devyte, 2018-03-19:
In pretty much all cases I've seen, there's something blocking the port. In the above comment it was a switch, in your case it could be a windows firewall thing, or an AV thing (e.g. mcafee), or similar.

MHz000, 2018-03-19:
Develo *bingo!*
It looks like you hit the nail on the head :-))
open port 8266 by following your excellent instruction the port showed up
in arduino IDE  and uploading sketches works fin AGAIN!

Thank-you so much Develo