esp8266/Arduino #2047: More like a doubt

lucifer43, 2016-05-21:
What is the difference between ( first connecting the esp8266 to wifi as a station and then building a server as an access point ) and ( directly building a server using esp8266 as access point)
Please elaborate

WereCatf, 2016-05-22:
Why do you think there is a difference?

altishchenko, 2016-05-22:
lucifer43,
I think it is a term mismatch of sorts... In wifi-speak, access point is a
'Wi-Fi access point', i.e. wifi router, giving wifi to its clients. Not a
TCP server. ESP can be easily a wifi station (a.k.a. client) and provide
TCP services - your first scenario, and it can be WiFi-router providing
wifi services and TCP services to its clients - your second scenario.

С уважением,
Александр Тищенко
+7 (916) 704-6157
atishch@gmail.com

closed 2016-06-01 by igrr