esp8266/Arduino #2049: ESP Huzzah board hang after deep sleep and  wake up ( pin 16 to rst connected)

Zyxmn, 2016-05-23:
I have been testing the deepsleep function on the huzzah board and the board crashes average after 9 to 14 consecutive deepsleep and wakeup cycles.

Sometimes it crashes with only 3 to 4 deepsleep and wakeup cycles and sometimes goes to about 20 plus cycles before crashing.

Here is the sketch I use:

#include <Esp.h>

void setup() 
{
  Serial.begin(115200);
  Serial.println();
  Serial.println(ESP.getResetReason());
  Serial.println(); 
  delay(100);
  ESP.deepSleep(100000, WAKE_RF_DEFAULT); // Sleep for 1 second
}

void loop() 
{

}

igrr, 2016-05-23:
Could you please decode the stack dump and paste the result here? Thanks.
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

Zyxmn, 2016-05-23:
How can I get the stack dump printed on the serial monitor?

Zyxmn, 2016-05-23:
The device hangs ( normal operation ceases and the led on gpio 0 glows like
when in programming mode)  but no stack dump on the serial monitor occurs
during the hang.

igrr, 2016-05-24:
The chip has a watchdog timer, so it shouldn't "hang" for more than 6 seconds. Can you check if there is any output on the serial monitor at 74880 baud?

Zyxmn, 2016-05-24:
This is output on the serial at 74880 baud

²j¤øtail 0
chksum 0x0f
csum 0x0f
~ld
   ",vërAer§ê "üm?ýk5êõh?üiõ(e|Zôj1ËÈ
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x0f
csum 0x0f
~ld
   "jÅjÒdpªòNXÍmÍ+ýë%êåhQ,üiõ(u|Ztj1ËÈ
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x0f
csum 0x0f
~ld
   ¢êÅjRdpºòNXÍm?ýk5êõhQ¬üéõ(e|Jôj3KÈ
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x0f
csum 0x0f
~ld
   "êÅêÒdpºrNXÍmÍ+ýë%jåhQ,üéõ(e|Jôj3ËÈ
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x0f
csum 0x0f
~ld
   "êÅêÒdpªrNXÍm?ýk5êõh?üéõ(e|Jôj3KÈ
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x0f
csum 0x0f
~ld
   ¢êÅjÒdpºrNXÍmÍ+ýë%êåhQ,üéõ(e|Zôj3KÈ
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

igrr, 2016-05-24:
Does it end like this? I.e. the last line before it hangs is `ets Jan 8 2013 ...`?

Zyxmn, 2016-05-24:
yes that is the last line

igrr, 2016-05-24:
In that case this is most likely a hardware issue. This indicates that the ROM bootloader didn't load anything from the flash chip.

Zyxmn, 2016-05-24:
tested on another board just now. having the same issue . can you possibly replicate this on your end?

Zyxmn, 2016-05-24:
on a second run in the new board, the device is chugging away nicely. will post the results again after an hour of testing

Zyxmn, 2016-05-24:
it seems to be a bug. The device hangs after a few consecutive deep sleep cycles even on the new board.

protodude, 2016-05-30:
Hello, 
I have exactly the same issue as Zyxmn described. (Also using the Adafruit Huzzah Feather)
Also here: "rst cause:2, boot mode:(3,6)" is last serial output.
I was also able to extend the number of working wake up cycles by adding a 4.7k pull up each for pin 0 and 2 and pull down for pin 15. 
I was even able to increase the cycles by adding a 2200uF Cap on the the 3.3V power net. 

Still it crashes after something between Xhours and a few days....

igrr, 2016-06-01:
I've set up a board to run deep sleep in a loop, will see if this can be reproduced.

igrr, 2016-06-02:
The board ran 13517 deep sleep cycles, no issues so far. Here's the log:
http://arduino.esp8266.com/deepsleep.log.zip

@protodude @Zyxmn the fact that crashes become less frequent when you fix hardware issues indicates that this is hardware-related. As I have mentioned above, if it hangs right after the `ets Jan 8 2013` line, it means that the program wasn't even loaded from the flash chip. So there isn't anything which you could change with software here. Be sure to have pull up resistors on RST, CH_PD, GPIO 0 and 2, capacitor on RST and adequate filtering caps on supply lines. You can find correct component values in the reference designs posted [here](http://espressif.com/en/support/download/documents), look for "Hardware resources".

closed 2016-06-02 by igrr