esp8266/Arduino #2052: Ethernal AP

kxmatejka, 2016-05-24:
#### Hardware
Hardware:		ESP-12

### Description

Hello, after uploading example ESP8266Wifi/WifiAccessPoint I have constantly accessible AP even if I upload empty or blink sketch. Everything else working fine, but I'm worried about some deep issue.

### Settings in IDE

Arduino IDE:            1.6.7
Module:  			NodeMcu 1.0 (ESP-12E Module)
Flash Size:               4MB
CPU Frequency:      80Mhz

WereCatf, 2016-05-24:
That's how it works, the WiFi-settings are stored in EEPROM and by default they're used from there at boot unless you supply new settings, ie. it's working as intended. Use WiFi.mode(WIFI_OFF) or WiFi.mode(WIFI_STA) if you don't want an AP running. To turn the WiFi-portion of the chip off completely and thus reduce power-consumption you can use the following: ```
WiFi.mode(WIFI_OFF);
WiFi.forceSleepBegin();
delay(1);
```

igrr, 2016-05-24:
Also using `WiFi.persistent(false);` before any other `WiFi.` calls will prevent configuration from being saved in EEPROM.

kxmatejka, 2016-05-24:
Thank you guys, I have not seen this feature anywhere described.

closed 2016-05-24 by kxmatejka