esp8266/Arduino #2053: esp8266/Arduino update for multicast bug?

Tinker1, 2016-05-24:
Igrr,
Over in the ESP8266 forum I was discussing with another user what others have been experiencing, a constant watchdog reboot when using this library with the Wemos D1 board setting the wifi to multicast mode.

It has been a number of weeks, and he mentioned that your were working on a resolution to this issue, so I was wondering if you have made any progress.

I apologize if this is the wrong area to post this question, but I didn't see any other way to contact you about this, and they said that you were not active over in the ESP8266 forum. Thanks for your time :)

igrr, 2016-05-24:
This issue should be fixed in [2.3.0-rc1](https://github.com/esp8266/Arduino/releases/tag/2.3.0-rc1), see #1826.

Tinker1, 2016-05-24:
Wonderful! Thank you so much for addressing this issue :)

closed 2016-06-01 by igrr