esp8266/Arduino #2055: FYI -- a fast i2c implementation in Assembly (brzo i2c)

pasko-zh, 2016-05-25:
Hello!

I would like to share this with you: I have implemented an i2c master in (inline) assembly for the esp8266 called **brzo i2c**. It can be used as an alternative to the wire library. It features:

- Robust implementation with SCL cock stretching support and handling of all those nasty i2c errors
- Clean and exact timing for SCL and SDA, e.g., removal of ACK spike
- Fast mode plus SCL speeds: 
800 KHz @ 80 MHz CPU
1 MHz @ 160 MHz CPU 
- Introduction of i2C transactions: Group any combination of i2c writes or reads together, either with or without repeated start

Further infos are available on the [wiki](https://github.com/pasko-zh/brzo_i2c/wiki) and the library is available [here](https://github.com/pasko-zh/brzo_i2c). 

The wiki and the code contains many infos/hints around i2c, so I thought it could be useful to you guys. And of course you may use the library :-)

cheers
paško

danxn, 2016-05-26:
BTW, as I can see onReceive handler disabled in ESP Wire library (https://github.com/esp8266/Arduino/blob/master/libraries/Wire/Wire.cpp#L203-L246) so it is impossible to use nodeMCU in multimaster mode on I2C bus. It will be great if this feature will be available again.

FWeinb, 2016-05-27:
Thanks for building this library I made a little comparison video showing the performance of driving a ssd1306 oled display using this library instead of Wire library. Have a look [here](https://vimeo.com/168362918).

pasko-zh, 2016-05-27:
**Wow, this looks very nice!! Very smooth!** So, I am happy it was useful to you!

btw: Hmm, how do you get the current time? Are you using the Time library or the native SDK sntp_ functions? Are you able to adjust for "Sommerzeit" aka daylight saving time? (For another project I want to display the current time, but of course not on such a fance display like you did, instead on an old Split Flap Display from the swiss railways (SBB))

FWeinb, 2016-05-27:
Thank you. For this I am using `configTime` so there is no adjustment for "Sommerzeit".

hallard, 2016-05-28:
@pasko-zh, my godness, nice works, thanks for this
@FWeinb nice comparison, would you mind sharing the code of the example ? I love OTA progress bar ;-)

igrr, 2016-06-01:
@pasko-zh awesome work, do you think it's possible to make a variant of Wire library based on bzro-i2c? I.e. make API compatible with Wire.

FWeinb, 2016-06-01:
I had the same idea it is tracked in pasko-zh/brzo_i2c#1

pasko-zh, 2016-06-01:
@igrr Thanks :-) 
Yes, it should be possible -- but please allow some time for that to happen.

d-a-v, 2018-10-12:
@pasko-zh any news regarding integration in the core ?

5chufti, 2018-10-14:
> @igrr Thanks :-)
> Yes, it should be possible -- but please allow some time for that to happen.

I think his question relates to your message quoted above

devyte, 2018-11-27:
Given recent I2C slave support integrated into the currennt Wire lib, this requires further investigation.
Removing milestone for now.

devyte, 2018-11-27:
@suculent is there any chance you could help with testing this?

suculent, 2018-11-27:
Well, I know Brzo and I'm already using it in some project as a replacement of the current I2C lib.
However I don't have much of a suitable testing hwardware (oscilloscope). This is nice work, but I'm currently very busy with drivers for APDU-9900, Network Interface Manager (allowing multiple network connections like WiFi/GSM at once; seamless connectivity switching and shared MQTT client)... not speaking of maintaining whole THiNX.

devyte, 2018-11-27:
@suculent The fact that you're using it in one of your projects tells me what I wanted to know :laughing: 
So it's just slave support in brzo before it can be wrapped and integrated.

suculent, 2018-11-27:
Indeed, slave support is now required. What we've merged from @bjoham can be happily replaced with Brzo, expecting it will share existing Wire interface.

Reason why I don't use Brzo in all projects is merely interoperability issue with all existing libraries based on Wire. It's great with displays where you need speed, however it gets complicated when you I 60+ I2C devices to manage.

suculent, 2018-11-27:
Also, I'd look into the Wire.begin() thing. For slave mode it needs both pin and address setting in the initializer... and most of the libs call Wire.begin() itself, which can be an issue.

Wire should be able to act properly, in case it has been already initialized, so the libraries calling just Wire.begin(); do not reset SDA/SCL pin settings or existing callbacks.

pasko-zh, 2018-11-27:
Well, I just followed the discussion here and I am of course happy to support you 😄 

My other job (aka _the_ job) keeps me quite busy doing many many many power point presenations :tired_face: thus, my motivation for these topics here is huge... but allow some time for me to answer your questions, it wont be always as fast as `brzo_i2c` is...

devyte, 2018-11-27:
@pasko-zh I'm glad you're here 😄 
Essentially, what I would like to do is integrate your great work into our Arduino core as a subrepo, wrap it with a Wire lib interface, and have it accessible to everyone. That would give the best of all worlds, I think, by getting your code to more users and having those users get the fastest implementation.
However, before that can proceed, at least one functional gap has to be closed: I2C slave support, which was added only recently to our current I2C implementation.
One thing that is imminent are some examples for I2C master and I2C slave to have 2 ESPs communicate. With that, we can test using P2P ESPs without external hardware.
Is there any chance you can implement slave support in your code? There is no hurry, so it's ok between powerpoints 😆 
Then, your repo is currently licensed as GPL-3.0. That is inherently incompatible with apps that are to be distributed in any way due to the closed source nature of Espressif's libs. In other words, any app that makes use of your code in conjunction with the closed Espressif libs can only be used personally and can't be distributed to others. Is there any chance you can change the license to a more permissive one, or at least provide an exception to re-license your code under our license when built in conjunction with our code?
I'm available on [gitter ](https://gitter.im/esp8266/Arduino) in case you would like to discuss directly.

pasko-zh, 2018-11-29:
@devyte Somewhere in between powerpoints this should be possible 😄 
Concerning GPL, I have to think again. I have to find a compromise between as free as possible (in the sense of free software) and giving as few "restrictions" as possible. LGPL is certainly a good option.

d-a-v, 2019-10-01:
I am closing this issue.
GPL [is still ongoing with this library](https://github.com/pasko-zh/brzo_i2c) so there's nothing we can do here.

However documentation on how to use it has been available for more than one year: 
https://github.com/pasko-zh/brzo_i2c/wiki#how-to-migrate-from-wire-library-to-brzo-i2c

closed 2019-10-01 by d-a-v

suculent, 2019-10-01:
Hi,

I don't have currently time to do that (releasing the project I was using it in).
I've also dropped software I2C from my project because of low realiability/scalability
and no forward compatibility with multi-core MPUs (ESP32) / I've exchanged that with CAN bus.

M.