esp8266/Arduino #2061: Exception (9) with 2.2.0

PyBerger, 2016-05-26:
ESP8266 running as TCP Server with 3 clients connected.

After some emission from the the server to the 3 clients,  exception is happening

Exception (9):
epc1=0x402031c7 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00017ce3 depc=0x00000000

ctx: sys
sp: 3ffffde0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff80:  401011ae 00000000 00000002 3fffdcb0
3fffff90:  3fffdc10 00000000 3fff3c6c 40226570
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49
<<<stack<<<

The Exception decoder doesn't give much information....

With 2.1.0 the problem doesn't seem to exist. With 2.2.0 and my setup it only takes a few seconds to be reproduced...

igrr, 2016-06-01:
The issue description doesn't really give me enough information to reproduce this. Maybe you could post and example sketch?

PyBerger, 2016-06-14:
Hi, 

Sorry for the late reply, was caught on some other activites. 

I have just made another try today with the latest version (master) and the result is somewhat different as I still have a crash with an Exception (9) (got once an Exception (4)) [again wasn't happening with the official 2.1.0 release].

The new stack trace is as follows:

`Exception (9):
epc1=0x402033a0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff195d depc=0x00000000

ctx: sys
sp: 3ffffdd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff70:  3fff6e74 4010117b 00000000 00000000
3fffff80:  401011b6 00000000 00000002 3ffee368
3fffff90:  3fffdc10 00000000 3fff6e44 40226c24
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49
<<<stack<<<`

> 
0x402033a0: ESP8266WiFiGenericClass::_eventCallback(void*) at hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2/functional line 2439
0x4010117b: ppProcessTxQ at ?? line ?
0x401011b6: ppProcessTxQ at ?? line ?
0x40226c24: wifi_set_status_led_output_level at ?? line ?
<

The sketch is pretty complex, so here is a summary of the working ...

There is a running interrupt that handles some external event...

every 200ms, if a buffer is not empty, data is sent over TCP using those calls:

`
  for (int i = 0; i < TCP_MAX_CLIENTS; i++) {
     if (tcpTransponderClient[i]) {
          int err = tcpTransponderClient[i].write((uint8_t *) wifi_buf, wifi_buf_idx);
          if (err != wifi_buf_idx) {
        debug_print("Error : Sending TCP packet: ");
        debug_println(err);
        if (err > 0)
        {
          debug_print("Error: New trial resending - ");
          err = tcpTransponderClient[i].write((uint8_t *) wifi_buf, wifi_buf_idx);
          if (err != wifi_buf_idx) { 
            debug_print("KO (");debug_println(err);debug_println(")");  
          } else
            debug_println("OK");
        }
      }
    }
  }
`
If needed I can send you the complete sketch in private.

PyB

PyBerger, 2016-07-18:
Hi, 

I still have this random exception (9) with 2.3.0

Igrr, I can pm you the code if you want to check....

PyB

devyte, 2017-10-17:
@PyBerger pin ISRs, as well as all functions called from within the ISRs, require the IRAM attribute.
This is not the right place to request help for your particular project. Closing per #3655 .

closed 2017-10-17 by devyte