esp8266/Arduino #2063: Serial.write(0) does not give output

gmag11, 2016-05-27:
### Basic Infos
Serial.write(0) does not produce any output in serial port.

#### Hardware
Hardware:			ESP-12
Core Version:      	Current git

### Description
I'm trying to generate serial output with a binary protocol where 0x00 byte value is possible. I use Serial.write(byte) to write values. It works until I get a 0 value. It gives no output from UART.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp

void setup()
{
	Serial.begin(115200);
}

void loop()
{
	Serial.write((int)1);
	delay(1000);
	Serial.write((int)0);
	delay(1000);
}
```

MacroYau, 2016-05-28:
Have you tried `Serial.write((byte) 0x00)`?

gmag11, 2016-05-28:
Yes, I've tried that. And use a variable too.

MacroYau, 2016-05-28:
Just tried on NodeMCU DevKit v1.0 hardware with the latest Git version of the Arduino core. It worked nominally to give `0x00` in Serial output, no matter I coded `(int) 0`, `0x00` or `(byte) 0x00`. Can you give further details to the problem, perhaps a screenshot of your Serial console?

gmag11, 2016-05-28:
Did you analyze output from serial port? Code worked and gave no error, but no char is on serial port when `write(0)`.
Console output is what I put on my first post, at the end. It should write `0x00` and `0x01` every second, but only `0x01` are written, every 2 seconds.
Thank you for testing.

gmag11, 2016-05-28:
It worked. Sorry and thank you. I was using Eltima Serial port Monitor. It did not show 0x00. Using PuTTY to log output let me notice that sequence is 0x00 0x01 actually.

closed 2016-05-28 by gmag11