esp8266/Arduino #2065: MDNS doesnt work when u switch the wifi modes

salilddd, 2016-05-28:
Hi,
In my program, the prom moves from WIFI_STA mode to WIFI_AP and then back to WIFI_STA.

but when I move back from WIFI_AP to WIFI_STA and try to declare the IP on MDNS, it doesnt seem to work. Example code below:
>> ping test.local doesnt work after the setup is completed.

`/*
  ESP8266 mDNS responder sample
 */

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiClient.h>

const char* ssid = "Dhawan-2G";
const char* password = "esujgvxp";

// TCP server at port 80 will respond to HTTP requests
WiFiServer server(80);

void setup(void)
{  
  Serial.begin(9600);
  
  // Connect to WiFi network
  WiFi.softAPdisconnect();
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");  
  
  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  // Set up mDNS responder:
  if (!MDNS.begin("test")) {
    Serial.println("Error setting up MDNS responder!");
    while(1) { 
      delay(1000);
    }
  }
  Serial.println("mDNS responder started");
    // Add service to MDNS-SD
  MDNS.addService("http", "tcp", 80);

delay(2000);
Serial.print("1");
//--------------------------------Now go to AP mode
 WiFi.disconnect();
  delay(500);
  WiFi.softAP("test");
  Serial.println(String("softap started - "));
  WiFi.mode(WIFI_AP);
delay(10000);
Serial.print("2");
  //-------------------------------------------------------------------
    // Connect to WiFi network
  WiFi.softAPdisconnect();
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");  
  
  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  // Set up mDNS responder:
  if (!MDNS.begin("test")) {
    Serial.println("Error setting up MDNS responder!");
    while(1) { 
      delay(1000);
    }
  }
  Serial.println("mDNS responder started");
    // Add service to MDNS-SD
  //MDNS.addService("http", "tcp", 80);
  //MDNS.update();

Serial.print("3");

  

  server.begin();
  Serial.println("TCP server started");
  

}

void loop(void)
{
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  Serial.println("");
  Serial.println("New client");

  // Wait for data from client to become available
  while(client.connected() && !client.available()){
    delay(1);
  }
  
  // Read the first line of HTTP request
  String req = client.readStringUntil('\r');
  
  // First line of HTTP request looks like "GET /path HTTP/1.1"
  // Retrieve the "/path" part by finding the spaces
  int addr_start = req.indexOf(' ');
  int addr_end = req.indexOf(' ', addr_start + 1);
  if (addr_start == -1 || addr_end == -1) {
    Serial.print("Invalid request: ");
    Serial.println(req);
    return;
  }
  req = req.substring(addr_start + 1, addr_end);
  Serial.print("Request: ");
  Serial.println(req);
  client.flush();
  
  String s;
  if (req == "/")
  {
    IPAddress ip = WiFi.localIP();
    String ipStr = String(ip[0]) + '.' + String(ip[1]) + '.' + String(ip[2]) + '.' + String(ip[3]);
    s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html>Hello from ESP8266 at ";
    s += ipStr;
    s += "</html>\r\n\r\n";
    Serial.println("Sending 200");
  }
  else
  {
    s = "HTTP/1.1 404 Not Found\r\n\r\n";
    Serial.println("Sending 404");
  }
  client.print(s);
  
  Serial.println("Done with client");
}
`

igrr, 2016-06-01:
Closing as duplicate of #1828

closed 2016-06-01 by igrr

salilddd, 2016-06-01:
Hey Ivan,

Is this issue fixed now?

Or what can be done to fix this bug?

Regards,
Salil