esp8266/Arduino #2068: Lost Wifi Connect and Memory

wuwx, 2016-05-30:
### Basic Infos

#### Hardware
Hardware:			NODEMCU1.0
Core Version:      	2.2.0

### Description

Problem description

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <SoftwareSerial.h>

SoftwareSerial mySerial(14, 12, false, 256);

void setup() {
  Serial.begin(115200);
  WiFi.begin("SSID", "PASSWORD");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  Serial.println(WiFi.localIP());
  
  mySerial.begin(9600);
}

void loop() {
  mySerial.println("0000");
  while (mySerial.available()) {
    Serial.write(mySerial.read());
  }
  
  if (millis() % 1000 == 0) {
    Serial.print(millis());
    Serial.print(" ESP.getFreeHeap: ");
    Serial.println(ESP.getFreeHeap());
  }
  
}

```

### Debug Messages

```
wifi evt: 0
wifi evt: 3
192.168.254.242
5000 ESP.getFreeHeap: 38560
wifi evt: 7
wifi evt: 7
wifi evt: 7
7000 ESP.getFreeHeap: 38560
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
18000 ESP.getFreeHeap: 38016
LmacRxBlk:1
LmacRxBlk:1
20000 ESP.getFreeHeap: 38016
LmacRxBlk:1
LmacRxBlk:1
22000 ESP.getFreeHeap: 38016
LmacRxBlk:1
wifi evt: 7
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
27000 ESP.getFreeHeap: 37672
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
38000 ESP.getFreeHeap: 37672
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
46000 ESP.getFreeHeap: 37328
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1

```

```
$ ping 192.168.254.242
PING 192.168.254.242 (192.168.254.242): 56 data bytes
ping: sendto: Host is down
ping: sendto: Host is down
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
Request timeout for icmp_seq 3
Request timeout for icmp_seq 4
Request timeout for icmp_seq 5
64 bytes from 192.168.254.242: icmp_seq=6 ttl=128 time=163.704 ms
64 bytes from 192.168.254.242: icmp_seq=7 ttl=128 time=1.972 ms
64 bytes from 192.168.254.242: icmp_seq=8 ttl=128 time=5.644 ms
64 bytes from 192.168.254.242: icmp_seq=9 ttl=128 time=2.556 ms
64 bytes from 192.168.254.242: icmp_seq=10 ttl=128 time=3.523 ms
64 bytes from 192.168.254.242: icmp_seq=11 ttl=128 time=3.042 ms
64 bytes from 192.168.254.242: icmp_seq=12 ttl=128 time=8.355 ms
64 bytes from 192.168.254.242: icmp_seq=13 ttl=128 time=5.272 ms
Request timeout for icmp_seq 14
Request timeout for icmp_seq 15
Request timeout for icmp_seq 16
Request timeout for icmp_seq 17
Request timeout for icmp_seq 18
Request timeout for icmp_seq 19
Request timeout for icmp_seq 20
Request timeout for icmp_seq 21
Request timeout for icmp_seq 22
Request timeout for icmp_seq 23
Request timeout for icmp_seq 24
Request timeout for icmp_seq 25
Request timeout for icmp_seq 26
Request timeout for icmp_seq 27
```

Running NODEMCU and ping Wifi.localIP() Address
some times later, we can't receive icmp response and
Wifi Connection is lost, and memory be leak

igrr, 2016-06-01:
Does the same happen when you add `WiFi.mode(WIFI_STA);` to your `setup` method?

wuwx, 2016-06-01:
oh yes, if we set WIFI_STA, it's normal
but if set WIFI_AP_STA, it's broken
how can we fix it ?

closed 2016-06-02 by wuwx