esp8266/Arduino #2071: multiple char [] , last defined char[] would affect the previous one (pointer issue ?)

mohsh86, 2016-05-30:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos
this sketch let ESP8266 to connect to energy meter PZEM-004T to retrieve energy usage data, in this instance, the PZEM-004T was disconnected for testing purposes, which results in -1 return for all the values by the PZEM-004T library, this demo was created from a combo of esp8266 wifi connect + mqtt pubsub + PZEM examples.

since there is no float to char []* in esp8266, it was converted to string first.

#### Hardware
Hardware:			Wemos D1 Mini
Core Version:      	2.2

### Description
defining multiple char [] arrays: msg, vchar, ichar, pchar, echar, rchar
once i introduced rchar (of size 1) *programming wise might be wrong but for the sake of the bug* i noticed that the energy reading (watthour generated by echar[]) is outputting blank on println instead of -1. [swapping the position of pchar and echar in decleration, pchar got affected]
increasing the size of rchar (to say rchar [10] like the others) would fix the issue, it looks like array size pointer index is referring to the previous declared ones ? i've abandoned programming ages ago so pardon me for bad programming in this sketch or not explaining the bug very well

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

/*
 Basic ESP8266 MQTT example

 This sketch demonstrates the capabilities of the pubsub library in combination
 with the ESP8266 board/library.

 It connects to an MQTT server then:
  - publishes "hello world" to the topic "outTopic" every two seconds
  - subscribes to the topic "inTopic", printing out any messages
    it receives. NB - it assumes the received payloads are strings not binary
  - If the first character of the topic "inTopic" is an 1, switch ON the ESP Led,
    else switch it off

 It will reconnect to the server if the connection is lost using a blocking
 reconnect function. See the 'mqtt_reconnect_nonblocking' example for how to
 achieve the same result without blocking the main loop.

 To install the ESP8266 board, (using Arduino 1.6.4+):
  - Add the following 3rd party board manager under "File -> Preferences -> Additional Boards Manager URLs":
       http://arduino.esp8266.com/stable/package_esp8266com_index.json
  - Open the "Tools -> Board -> Board Manager" and click install for the ESP8266"
  - Select your ESP8266 in "Tools -> Board"

*/

#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <PZEM004T.h>

// Update these with values suitable for your network.
const char* ssid = "Meshx86-Zone2";
const char* password = "xxx";
const char* mqtt_server = "192.168.86.1";

// Static IP Configs
IPAddress ip(192,168,86,11);
IPAddress gw(192,168,86,1);
IPAddress nm(255,255,255,0);

WiFiClient espClient;
PubSubClient client(espClient);
long lastMsg = 0;
char msg[50];
int value = 0;

// Readings & Reset Vars
float v,i,p,e;
int r=0;
String v_str,i_str,p_str,e_str,r_str;
char vchar[10],ichar[10],pchar[10],echar[10];
char rchar[1];

// The PZEM004T stuff
PZEM004T pzem(1,3);  // RX,TX
IPAddress pzem_ip(192,168,1,1);

void setup() {
  pinMode(BUILTIN_LED, OUTPUT);     // Initialize the BUILTIN_LED pin as an output
  Serial.begin(115200);
  pzem.setAddress(pzem_ip);
  setup_wifi();
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);
  digitalWrite(BUILTIN_LED, HIGH);
  
}

void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);  
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  WiFi.config(ip,gw,gw,nm);
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();

  // Switch on the LED if an 1 was received as first character
  if ((char)payload[0] == '1') {
    digitalWrite(BUILTIN_LED, LOW);   // Turn the LED on (Note that LOW is the voltage level
    // but actually the LED is on; this is because
    // it is acive low on the ESP-01)
  } else {
    digitalWrite(BUILTIN_LED, HIGH);  // Turn the LED off by making the voltage HIGH
  }

}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect("ESP8266Client")) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      client.publish("outTopic", "hello world");
      // ... and resubscribe
      client.subscribe("/energy/mainsreset");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}
void loop() {
  v = pzem.voltage(ip);
  i = pzem.current(ip);
  p = pzem.power(ip);
  e = pzem.energy(ip);

  v_str=String(v);
  i_str=String(i);
  p_str=String(p);
  e_str=String(e);
  r_str=String(r);

  v_str.toCharArray(vchar, v_str.length() + 1);
  i_str.toCharArray(ichar, i_str.length() + 1);
  p_str.toCharArray(pchar, p_str.length() + 1);
  e_str.toCharArray(echar, e_str.length() + 1);
  r_str.toCharArray(rchar, r_str.length() + 1);
  
  
  
  
  if (!client.connected()) {
    reconnect();
  }
 // client.loop();

    Serial.print("Publish message voltage ");
    Serial.println(vchar);
    client.publish("/energy/voltage", vchar);
    
    Serial.print("Publish message current ");
    Serial.println(ichar);
    client.publish("/energy/current", ichar);

    Serial.print("Publish message watt ");
    Serial.println(pchar);
    client.publish("/energy/watt", pchar);

    Serial.print("Publish message watthour ");
    Serial.println(echar);
    client.publish("/energy/watthour", echar);

    Serial.print("Publish message reset ");
    Serial.println(rchar);
    client.publish("/energy/mainsreset", rchar);

    delay(1000);
}
### Debug Messages

```
���� 
Connecting to Meshx86-Zone2
.
WiFi connected
IP address: 
192.168.86.11
���V y���V z���V {���V |Attempting MQTT connection...:ref 1
:wr
:sent 27
:ww
:rn 4
:c0 1, 4
connected
:wr
:sent 23
:ww
:wr
:sent 25
:ww
Publish message voltage -1.00
:wr
:rn 5
:sent 24
:ww
Publish message current -1.00
:wr
:sent 24
:ww
Publish message watt -1.00
:wr
:sent 21
:ww
Publish message watthour 
:wr
:sent 20
:ww
Publish message reset 0
:wr
:sent 23
:ww
:rch 5, 23

```

mohsh86, 2016-05-30:
updated code to reproduce the problem, without all the wifi / serial / mqtt stuff

`float v=-1,i=-1,p=-1,e=-1;
int r=0;
String v_str,i_str,p_str,e_str,r_str;
char vchar[10],ichar[10],pchar[10],echar[10];
char rchar[1];

void setup() {
Serial.begin(115200);
}

void loop() {
  v_str=String(v);
  i_str=String(i);
  p_str=String(p);
  e_str=String(e);
  r_str=String(r);

  v_str.toCharArray(vchar, v_str.length() + 1);
  i_str.toCharArray(ichar, i_str.length() + 1);
  p_str.toCharArray(pchar, p_str.length() + 1);
  e_str.toCharArray(echar, e_str.length() + 1);
  r_str.toCharArray(rchar, r_str.length() + 1);

  
    Serial.print("Publish message voltage ");
    Serial.println(vchar);
    
    
    Serial.print("Publish message current ");
    Serial.println(ichar);

    Serial.print("Publish message watt ");
    Serial.println(pchar);

    Serial.print("Publish message watthour ");
    Serial.println(echar);

    Serial.print("Publish message reset ");
    Serial.println(rchar);

    delay(1000);

}`

WereCatf, 2016-05-30:
> once i introduced rchar (of size 1) programming wise might be wrong but for the sake of the bug i noticed that the energy reading (watthour generated by echar[]) is outputting blank on println instead of -1. [swapping the position of pchar and echar in decleration, pchar got affected]

Buffer-overflow, it's as simple as that. The bug is not with the ESP8266 codebase, the bug is simply in your own code: increase the size of rchar so it doesn't overflow.

mohsh86, 2016-05-31:
rchar is an int that will always have either 1 or 0 in my program, my question is, setting the size of rchar would result in affecting the previous array (echar)

WereCatf, 2016-06-01:
I just told you, it's a buffer-overflow. You are not reserving enough space for the character and the terminating NULL, so the NULL ends up in echar instead. Increase the size of rchar.

closed 2016-06-01 by igrr