esp8266/Arduino #2073: Arduino max() min() functions doesn't work when I load ESP Libraries

jamilbn, 2016-05-31:
#### Hardware
Hardware:			ESP-12
Core Version:      	2.2.0

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPUpdateServer.h>
#include <ESP8266mDNS.h>
#include <FS.h>

Tried to load even only the FS.h but I get the same error

error: 'min' was not declared in this scope
error: 'max' was not declared in this scope

When I comment those libraries my sketch works just fine with the functions max an min as they should work.

https://www.arduino.cc/en/Reference/Max

WereCatf, 2016-05-31:
It's an issue I stumbled upon myself, too. I just used _min() and _max() in my code instead.

igrr, 2016-06-01:
Closing as duplicate of #398

closed 2016-06-01 by igrr