esp8266/Arduino #2075: ESP8266WebServer, WiFiClientSecure and https

kenpeter, 2016-05-31:
### Basic Infos

#### Hardware
Hardware:			ESP-12e
Core Version:      	not sure

### Description

Problem description
I set up a json api: https://toothfi.com/device/device/abc/get_fake_device_info

I follow this sample code: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/HTTPSRequest/HTTPSRequest.ino

and make my own version:
https://github.com/kenpeter/arduino_https_request

The real problem is:
In https_request.ino, if I comment out "ESP8266WebServer g_web_server(80);" in line 18. The code is working. If I leave "ESP8266WebServer g_web_server(80);", the error is

```
Decoding 27 results
0x40224443: more_comps at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1072
0x402245dc: alloc at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1106
0x4022488e: regular_multiply at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 822
0x402253e6: bi_barrett at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1287
0x40225641: bi_mod_power at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1412
0x402244a8: trim at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1197
0x402267ec: RSA_public at /Users/igrokhotkov/e/axtls/e1/crypto/rsa.c line 227
:  (inlined by) RSA_encrypt at /Users/igrokhotkov/e/axtls/e1/crypto/rsa.c line 260
0x402228ca: send_client_key_xchg at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 342
0x40222cc9: do_clnt_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 105
0x4022dcbf: pbuf_free at ?? line ?
0x4022dccd: pbuf_free at ?? line ?
0x40202c93: ClientContext::_consume(unsigned int) at /home/kenpeter/.arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 497
:  (inlined by) ClientContext::read(char*, unsigned int) at /home/kenpeter/.arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WiFi/src/include/ClientContext.h line 169
0x40222700: do_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1470
:  (inlined by) basic_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1348
0x40222a64: do_client_connect at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 148
0x40221418: ssl_new at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 576
0x40222836: ssl_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 262
0x402030e8: SSLContext::connect(ClientContext*, unsigned int) at /home/kenpeter/.arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 497
:  (inlined by) WiFiClientSecure::_connectSSL() at /home/kenpeter/.arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 262
0x402025ce: WiFiClient::connect(IPAddress, unsigned short) at /home/kenpeter/.arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WiFi/src/WiFiClient.cpp line 327
0x402069c8: esp_yield at /home/kenpeter/.arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/core_esp8266_main.cpp line 43
0x4020327a: WiFiClientSecure::connect(IPAddress, unsigned short) at /home/kenpeter/.arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 246
0x4020254d: WiFiClient::connect(char const*, unsigned short) at /home/kenpeter/.arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WiFi/src/WiFiClient.cpp line 327
0x40206348: Print::println() at /home/kenpeter/.arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/Print.cpp line 76
0x40203290: WiFiClientSecure::connect(char const*, unsigned short) at /home/kenpeter/.arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 252
0x40201ee9: get_device_info() at /tmp/buildcd64e0125f5c3fda1d6d44e3409e6705.tmp/sketch/device.cpp line 36
0x40202020: setup at /var/www/html/test/testme/igloo/misc/https_request/https_request.ino line 23
0x40206a0c: loop_wrapper at /home/kenpeter/.arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/core_esp8266_main.cpp line 43
0x40100718: cont_norm at /home/kenpeter/.arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/cont.S line 109
```
I am not sure what the errors mean. 

### Settings in IDE

Module:  			nodemcu 1.0
Flash Size: 		4MB
CPU Frequency:		160Mhz
Flash Mode:			?
Flash Frequency:	?
Upload Using:		SERIAL
Reset Method:		?

### Full Sketch to have a look
https://github.com/kenpeter/arduino_https_request

igrr, 2016-06-01:
Having ESP8266WebServer and HTTPS client and the same time likely means you are going to run out of memory.

supersjimmie, 2016-06-03:
Looks like I have the same:
It looks like Stable 2.2.0 still has this https problem.
When my code repeats an https call, after a number of calls I receive:
```
Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3fff0c40 end: 3fff1220 offset: 01a0
```
The stack point to a number of https stuff:
```
0x40222497: more_comps at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1072
0x40222630: alloc at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1106
0x402228e2: regular_multiply at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 822
0x4022343a: bi_barrett at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1287
0x40223695: bi_mod_power at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1412
0x402224fc: trim at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1197
0x40224840: RSA_public at /Users/igrokhotkov/e/axtls/e1/crypto/rsa.c line 227
:  (inlined by) RSA_encrypt at /Users/igrokhotkov/e/axtls/e1/crypto/rsa.c line 260
0x4022091e: send_client_key_xchg at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 342
0x40220d1d: do_clnt_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 105
0x4022bd13: pbuf_free at ?? line ?
0x4022bd21: pbuf_free at ?? line ?
0x40202edf: ClientContext::_consume(unsigned int) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 497
:  (inlined by) ClientContext::read(char*, unsigned int) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src\include/ClientContext.h line 169
0x40220754: do_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1470
:  (inlined by) basic_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1348
0x40220ab8: do_client_connect at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 148
0x4021f46c: ssl_new at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 576
0x4022088a: ssl_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 262
0x40203364: SSLContext::connect(ClientContext*, unsigned int) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 497
:  (inlined by) WiFiClientSecure::_connectSSL() at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 262
0x40202752: WiFiClient::connect(IPAddress, unsigned short) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x40204bec: esp_yield at C:\...\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x4020363a: WiFiClientSecure::connect(IPAddress, unsigned short) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 246
0x4010020c: _umm_free at C:\...\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1285
0x402026d1: WiFiClient::connect(char const*, unsigned short) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x40202fcc: WiFiClientSecure::~WiFiClientSecure() at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 497
0x40203650: WiFiClientSecure::connect(char const*, unsigned short) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 252
0x40203924: HTTPClient::connect() at C:\...\esp8266\2.2.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 308
0x40204454: String::~String() at C:\...\esp8266\2.2.0\cores\esp8266/WString.cpp line 720
0x40203da0: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at C:\...\esp8266\2.2.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 308
0x40204f48: Print::write(unsigned char const*, unsigned int) at C:\...\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40204239: Print::write(char const*) at C:\...\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40203e46: HTTPClient::GET() at C:\...\esp8266\2.2.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 308
0x4020211c: loop at C:\Users\username\Documents\Arduino\octoblu-stream/octoblu-stream.ino line 70
0x40202268: ESP8266WiFiAPClass::softAPdisconnect(bool) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/ESP8266WiFiAP.cpp line 276
0x40204356: Print::print(unsigned long, int) at C:\...\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40204f48: Print::write(unsigned char const*, unsigned int) at C:\...\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40238a28: sntp_getservername at ?? line ?
0x40204239: Print::write(char const*) at C:\...\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x402042b0: Print::println(Printable const&) at C:\...\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40238a28: sntp_getservername at ?? line ?
0x40201fb4: setup at C:\Users\username\Documents\Arduino\octoblu-stream/octoblu-stream.ino line 45
0x40204c38: loop_wrapper at C:\...\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100718: cont_norm at C:\...\esp8266\2.2.0\cores\esp8266/cont.S line 109

```

igrr, 2016-06-03:
Yes, your stack trace also indicates that you are running out of RAM, or the heap is getting fragmented.

supersjimmie, 2016-06-03:
It's 'clearly' related to https, doing the same on http just runs fine for hours.
```
//    http->begin(STREAMHOST, 443, streamurl, streamhttpsfingerprint);
    http->begin("http://" + String(STREAMHOST) + streamurl);
```

igrr, 2016-06-03:
This is related to the fact that TLS needs plenty of RAM, whereas plaintext transport (one used for HTTP)ndoesnt need any

igrr, 2016-06-03:
He

supersjimmie, 2016-06-07:
Might some kind of (fixed) reservation of that large amount or RAM be an option?
It works fine for a couple of times, so I would say it's not because there's really not enough memory.

SloMusti, 2016-12-03:
Running in the same issue here, that there is insufficient RAM for both to be used at the same time. Any suggested solutions?

Palatis, 2016-12-21:
how much ram does it need anyway?
I have ~20kb free heap and it still fails.

me-no-dev, 2016-12-21:
When I tested TLS on the ESP, I could manage to do basic requests with not much data in the response. I got some sort of TLS server to run as well, but again with not much more content than a couple of packets. Everything above that requires more ram than we have. If you have more stuff going on in your sketch, chances are that that costs you RAM already so running TLS is even harder

Palatis, 2016-12-21:
I'm probably only need to send & receive < 0.5kb of data... :-/

pburgio, 2017-03-30:
Hello, I am running in the same problem. Anyone found a way to solve this?

Paolo

Jeroen88, 2017-07-16:
Got a (not ideal) solution running both. I dynamically create both in loop() using new and delete. In setup() I create a webServer = new ESP8266WebServer(80); In loop() when I need the httpClient, I delete webServer, call a function that has httpClient as a local variable, and recreate the webServer after the call. It works, but the disadvantage is that I client sometimes can not connect, because the web server is not present.

devyte, 2017-10-17:
A TLS transaction requires just under 20KB of RAM.
Closing, as there isn't much that can be done at this time.

closed 2017-10-17 by devyte