esp8266/Arduino #2076: Update ESP8266 library on Arduino IDE

odilonafonso, 2016-05-31:
Please, how can I update the esp8266 librarys on Ubuntu.

I did it (in January) :

<"git clone https://github.com/esp8266/Arduino.git esp8266">

Now I want update, but git does not clone over an existing directory.

WereCatf, 2016-05-31:
`git pull`

WereCatf, 2016-05-31:
Oh, and don't forget to run the get.py in the tools-directory to get the latest tools.

odilonafonso, 2016-05-31:
Hi @WereCatf .. thanks for help.

I tried git pull. But....

```
git pull https://github.com/esp8266/Arduino.git esp8266
fatal: Not a git repository (or any of the parent directories): .git
```

For now, i did :
```
mv esp8266 esp8266old
git clone https://github.com/esp8266/Arduino.git esp8266

```

I know, is not the best solution !!!

WereCatf, 2016-05-31:
I said, inside the esp8266-directory. You tried to run `git pull` outside the directory, that's why you got an error. It's a lot easier and faster than doing a full clone every time.

closed 2016-06-01 by igrr