esp8266/Arduino #2080: ESP8266Wifi sample problems with client browser GET /favicon.ico

odilonafonso, 2016-06-01:
### Basic Infos

#### Hardware
Hardware:		ESP-12
Core Version:      	2.2.0

### Description

I'm testing the sample:
WiFiWebServer from ESP8266WiFi samples
and I came across the following problem:
When requesting a client page connected by Chromium (Ubuntu 15.10), after receiving the page and respond accordingly, I get one (or more) new request(s) from the client:
`/ GET favicon.ico HTTP / 1.1`
thereby causing a failure in the execution of the sketch.
Can anyone tell me why and how to deal with this problem?
Grateful

### Settings in IDE

Module:  			ESP-12
Flash Size: 		4MB/3MB
CPU Frequency:	80Mhz
Upload Using:		SERIAL

### Debug Messages

```
GET /gpio/2 HTTP/1.1
Client disconnected
new client
GET /favicon.ico HTTP/1.1
new client
GET /favicon.ico HTTP/1.1
new client
GET /favicon.ico HTTP/1.1
```

odilonafonso, 2016-06-01:
The problem occurs when I use a port other than 80.
Using port 80 the problem does not occur.
Can anyone tell me why?

igrr, 2016-06-02:
Please clarify what the problem is, i.e. "thereby causing a failure in the execution of the sketch" is pretty vague.
Also please provide the code to reproduce the issue.

odilonafonso, 2016-06-02:
Hi @igrr 
Thanks for help.
The code is the sample WiFiWebServer from the ESP8266 repository (this). Code above with the modifications made marked with **"//change Odi"**

```
/*
 *  This sketch demonstrates how to set up a simple HTTP-like server.
 *  The server will set a GPIO pin depending on the request
 *    http://server_ip/gpio/0 will set the GPIO2 low,
 *    http://server_ip/gpio/1 will set the GPIO2 high
 *  server_ip is the IP address of the ESP8266 module, will be 
 *  printed to Serial when the module is connected.
 */

#include "ESP8266WiFi.h"

const char* ssid = "xxxxxxxxx";        //changed Odi
const char* password = "yyyyyyyyy";   //changed Odi

// Create an instance of the server
// specify the port to listen on as an argument
WiFiServer server(81);           //changed Odi

#define GPIO2 5                        //included Odi

void setup() {
  Serial.begin(115200);
  delay(10);

  // prepare GPIO2
  pinMode(GPIO2, OUTPUT);              //changed Odi
  digitalWrite(GPIO2, 0);              //changed Odi
  
  // Connect to WiFi network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  
  // Start the server
  server.begin();
  Serial.println("Server started");

  // Print the IP address
  Serial.println(WiFi.localIP());
}

void loop() {
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  
  // Wait until the client sends some data
  Serial.println("new client");
  while(!client.available()){
    delay(1);
  }
  
  // Read the first line of the request
  String req = client.readStringUntil('\r');
  Serial.println(req);
  client.flush();
  
  // Match the request
  int val;
  if (req.indexOf("/gpio/0") != -1)
    val = 0;
  else if (req.indexOf("/gpio/1") != -1)
    val = 1;
  else {
    Serial.println("invalid request");
    client.stop();
    return;
  }

  // Set GPIO2 according to the request
  digitalWrite(GPIO2, val);            //changed Odi
  
  client.flush();

  // Prepare the response
  String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html>\r\nGPIO is now ";
  s += (val)?"high":"low";
  s += "</html>\n";

  // Send the response to the client
  client.print(s);
  delay(1);
  Serial.println("Client disonnected");

  // The client will actually be disconnected 
  // when the function returns and 'client' object is detroyed
}

```

Here the dump from serial:

```
Connecting to xxxxxxxxx
..........
WiFi connected
Server started
192.168.2.103
new client
GET /gpio/1 HTTP/1.1
Client disonnected
new client
GET /favicon.ico HTTP/1.1
invalid request
new client
GET /favicon.ico HTTP/1.1
invalid request

```
The hardware is ESP-12e V2.
Arduino IDE 1.6.7
ESP8266 SDK 2.2.0
Ubuntu 15.10
Chromium Version 50.0.2661.102 Ubuntu 15.10 (64-bit)

Here dump when called from Firefox - with the same problem today...

```
new client
GET /gpio/1 HTTP/1.1
Client disonnected
new client
GET /favicon.ico HTTP/1.1
invalid request
new client
GET /favicon.ico HTTP/1.1
invalid request
new client
GET /favicon.ico HTTP/1.1
invalid request
new client
GET /favicon.ico HTTP/1.1
invalid request
new client
GET /gpio/0 HTTP/1.1
Client disonnected
new client
GET /favicon.ico HTTP/1.1
invalid request
new client
GET /favicon.ico HTTP/1.1
invalid request
new client
GET /favicon.ico HTTP/1.1
invalid request

```

I hope that helps identify the problem.

WereCatf, 2016-06-02:
Um, the browser requests a favicon.ico, but your sketch doesn't reply with a 404 not found? Of course it's going to try again, then. Reply with a 404.

odilonafonso, 2016-06-02:
This sketch works a little better:

```
/*
 *  This sketch demonstrates how to set up a simple HTTP-like server.
 *  **** na verdade é um TCP-Server !!!!!
 *  The server will set a GPIO pin depending on the request
 *    http://server_ip/gpio/0 will set the GPIO2 low,
 *    http://server_ip/gpio/1 will set the GPIO2 high
 *  server_ip is the IP address of the ESP8266 module, will be 
 *  printed to Serial when the module is connected.
 */

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>

const char* ssid = "xxx";
const char* password = "yyy";

// helper macro
#define LINE(name,val) Serial.print(name); Serial.print("\t"); Serial.println(val);
#define _FILENAME_ String(__FILE__).substring(String(__FILE__).lastIndexOf('/')+1)

#define SERVER_PORT 81
// Create an instance of the server
// specify the port to listen on as an argument
WiFiServer server(SERVER_PORT);

#define GPIO 5
#define STATUS_LED 10
void setup() {
  Serial.begin(115200);
  LINE("\nStarting:", _FILENAME_);
  delay(10);

  // prepare GPIO2
  pinMode(GPIO, OUTPUT);
  digitalWrite(GPIO, LOW);
  pinMode(STATUS_LED, OUTPUT);
  digitalWrite(STATUS_LED, LOW);
  
  // Connect to WiFi network
  Serial.println();

  WiFi.printDiag(Serial);

  String xSSID = WiFi.SSID();
  if ( !(WiFi.SSID().equals("")) ){
    Serial.println("Trying to connect with credentials found...");
    WiFi.begin();
  }
  else {
    Serial.print("Connecting to "); Serial.println(ssid);
    Serial.println("Connecting with new credentials..");
    WiFi.begin(ssid, password);
  }
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  // Print the IP address
  Serial.println(WiFi.localIP());

  // Set up mDNS responder:
  // - first argument is the domain name, in this example
  //   the fully-qualified domain name is "esp8266.local"
  // - second argument is the IP address to advertise
  //   we send our IP address on the WiFi network
  if (!MDNS.begin("esp8266")) {
    Serial.println("Error setting up MDNS responder!");
    while(1) { 
      delay(1000);
    }
  }
  Serial.println("mDNS responder started");
  
  // Start the server
  server.begin();
  Serial.println("Server started");

  // Add service to MDNS-SD
  MDNS.addService("http", "tcp", SERVER_PORT);
}

void displayStatusLed() {
   for (int i=0; i<4; i++) {
      digitalWrite(STATUS_LED, HIGH);
      delay(111);
      digitalWrite(STATUS_LED, LOW);
      delay(111);
   }
}

void loop() {
   while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
   }
   // Check if a client has connected
   WiFiClient client = server.available();
   if (!client) {
      return;
   }

   int nTimes=0;
   #define LIMIT 50
   while(!client.available()) {
      if (nTimes++ > LIMIT) {
//         Serial.println("*");
         return;
      }
//      Serial.print(".");
      delay(3);
   }

/*
 * Read the first line of the request
 * The first line contains all the necessary information;
 * Url, arguments passed
 * The others are supplementary information, the connected client,
 * How kind of browse (Firefox, Chrome, etc ..)
 * Content-type, refresh, etc ...
 * Necessary only if we implement a strict webserver as
 * The HTTP protocol
 */
   String req = client.readStringUntil('\r');
   if (req.length() < 1) {
      return;
   }
   // read the rest of the submitted request, despising
   while (client.available())
      client.read();
               
   // Prepare the response
   String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html><h1>\r\n";

   // Match the request
   int val;
   if (req.indexOf("/gpio/0") != -1)
      val = 0;
   else if (req.indexOf("/gpio/1") != -1)
      val = 1;
   else if (req.indexOf("favicon.ico") != -1) {
      client.stop();
      delay(11);
      return;
   }
   else
      val = -1;

   displayStatusLed();
   Serial.println("new client");
   Serial.println(req);
   client.flush();
  
   String strStatus;
   if (val < 0) {
      strStatus = "invalid request";
   }
   else {
      // Set GPIO2 according to the request
      digitalWrite(GPIO, val);
      strStatus = "GPIO is now ";
      strStatus += (val)?"high":"low";
   }

   s += strStatus + "</h1></html>\r\n";

   // Send the response to the client
   client.print(s);
   delay(1);
   client.flush();
   Serial.println(strStatus);
   Serial.println("Client disconnected");

  // The client will actually be disconnected 
  // when the function returns and 'client' object is detroyed
}

```

Because I despise the request:
`GET /favicon.ico`
and hide the problem
But it's coming ... commenting the code as follows:
```
   // Match the request
   int val;
   if (req.indexOf("/gpio/0") != -1)
      val = 0;
   else if (req.indexOf("/gpio/1") != -1)
      val = 1;
//   else if (req.indexOf("favicon.ico") != -1) {
//      client.stop();
//      delay(11);
//      return;
//   }
   else
      val = -1;
```
The problem becomes visible again.

and here the debug:
(With hidden problem)

```
Starting:	WiFiWebServerMDNS.ino

Mode: STA
PHY mode: N
Channel: 1
AP id: 0
Status: 1
Auto connect: 1
SSID (9): xxx
Passphrase (10): yyy
BSSID set: 0
Trying to connect with credentials found...
..........
WiFi connected
192.168.2.103
mDNS responder started
Server started
new client
GET /gpio/1 HTTP/1.1
GPIO is now high
Client disconnected
new client
GET /gpio/0 HTTP/1.1
GPIO is now low
Client disconnected
new client
GET /gpio/2 HTTP/1.1
invalid request
Client disconnected

```

igrr, 2016-06-02:
Sorry, we the code, but it still eludes me, what exactly is the problem?
The fact that browser sends a request for favicon? It's hardly a problem, that is just something browsers happen to do. If you don't have a favicon, reply with 404, as @WereCatf has suggested.

odilonafonso, 2016-06-02:
Okay @WereCatf, thanks for your help again.

Why the browser is asking favicon.ico?
If I answer the first request with 404, I will solve the remaining requests, but the first request will continue to be made.
The question remains:

_**Why the browser is requesting favicon.ico?**_

I made some googling and found a lot of people complaining about this problem, but found no solution.

Well, making the change for you suggested, things improved. I changed a few lines of code by:

```
   // Match the request
   int val;
   if (req.indexOf("/gpio/0") != -1)
      val = 0;
   else if (req.indexOf("/gpio/1") != -1)
      val = 1;
   else if (req.indexOf("favicon.ico") != -1) {
      s = "HTTP/1.1 404 WTF Favicon !!\r\n";
      val = -2;
   }
   else
      val = -1;

   displayStatusLed();
   Serial.println("new client");
   Serial.println(req);
   client.flush();
  
   String strStatus;
   if (val < 0) {
      strStatus = ( (val == -1) ? "invalid request" : "WTF Favicon !!" );
   }
```
Still I get some "dirt":
```
new client
GET /gpio/2 HTTP/1.1
invalid request
Client disconnected
new client
GET /favicon.ico HTTP/1.1
WTF Favicon !!
Client disconnected
new client
GET /favicon.ico HTTP/1.1
WTF Favicon !!
Client disconnected
new client
GET /gpio/2 HTTP/1.1
invalid request
Client disconnected

```
**_But much less than before !!!!_**

Thank you ! 
I'll try to figure out how to prevent browse send this nonsense, but if not, I will keep this code.

odilonafonso, 2016-06-02:
OK @igrr , thanks for your help too.

I will follow the recommendation of @WereCatf , as well as yours.
But I imagined it as a TCP (not HTTP) server the protocol requirements were lower.
Anyway, my question is answered:
I do not see any need to use the library "WiFiServer.h" because "ESP8266WebServer.h" library is much more complete.

And would like to use this piece of code checking for a request for favicon.ico by the browser.

Thank you guys !!

closed 2016-06-02 by odilonafonso

odilonafonso, 2016-06-02:
Perhaps this issue could be raised in this (and perhaps all) of the examples with webserver.
Like me, most people who use the examples are beginners and probably go through the same kind of situation. They will probably lose many hours with it.

What is the effective need to send a favicon.ico from an IoT device?
It is possible to create a small favicon to be sent by IoT webserver to the browser? Or send any other information to the browser?

This is a question that I will search with patience.

WereCatf, 2016-06-02:
I don't understand why you think this is an issue in the first place. Browsers often make requests for stuff that the server doesn't have, the servers reply with a 404 and it's up to the browsers to handle things from there on.

Also, there is no "effective need to send a favicon.ico" if you don't want to, it's up to you. All modern browsers simply request for it by default as it's the little icon you get for that website when you bookmark a page or when you open a page in a new tab -- see this little picture I cut out of my Firefox-browser here: https://dl.dropboxusercontent.com/u/11811685/favicons.png It all works just fine even if you reply with a 404, the browser just won't show an icon, then.

Also, yes, it is possible to send favicons from the ESP8266 if you like. There are several example-sketches that show you how to serve files from the internal Flash.

odilonafonso, 2016-06-02:
I thought it was a problem why do not quite understand, in this matter I am a beginner.
I thought it was a problem for not being provided for in the sample, not being treated.
And since you are an expert and telling me that there is no actual need to send a favicon.ico, I believe you. But in my application I will handle the browser request, because I did not know it was necessary.
Probably who made the example did not bother with it, but he knew what he was doing.
Finally, I appreciate your suggestion as an expert on the subject and I will look for some example of how to send a file from the flash memory.
Thanks again for your attention and help.

electronicsguy, 2016-12-09:
@odilonafonso   Thanks for asking this! @WereCatf  I too was flumoxxed with this, since it was the first time I was seeing it, and the server seemed to output a message as if it's an error. I too didn't know that the browser makes these secondary requests which we could simply drop with a 404 response.

BabarLatif, 2017-03-17:
Hi Odilonafonso,

I am a newbie hobbiyst. I want learn HTML for Arduino IDE for nodemcu etc. What my problem is that I don't know how to plug standard html statements into IDE code. Where and when to insert "\" or modify the syntax. Is there a good source to learn html for Arduino IDE.

Appreciate you help

dipeshpatidar, 2017-10-11:
can any one please explain me what these line is for please help me.....i am a beginner
  String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html>\r\nGPIO is now ";

WereCatf, 2017-10-11:
@dipeshpatidar It tells the browser that its request succeeded, then tells the browser what type of content is incoming from the server and the rest is the actual contents.

dipeshpatidar, 2017-10-12:
@WereCatf thnxz for your pricious rply....... But can you please explain me what is the work of each element or what is the meaning of each element in these .......like here it is \r\n\r\n\r\n.....so what does these mean

electronicsguy, 2017-10-12:
@dipeshpatidar   please read up here: [SO-link](https://stackoverflow.com/questions/7372918/whats-the-use-of-r-escape-sequence)

HTTP protocol requires header/body to be terminated CRLF. read here: [SO-link2](https://stackoverflow.com/questions/5757290/http-header-line-break-style)

suomi35, 2017-12-01:
You can also reply with a 301 and source favicon.ico, etc from somewhere else:
client.print("HTTP/1.1 301 Moved Permanently\r\nLocation: http://a.website.com/favicon.ico\r\n\r\n");

That way your browser is happy and you can have a favicon :smile:

alwynallan, 2018-01-10:
This thread shows up first on Google for "esp8266webserver send favicon.ico" so I'll share the solution to that problem, even if it's off-topic.

1. Use 'xxd -p -c1 favicon.ico' and a text editor to create some C like 'PROGMEM const char favicon[] = {0x00, 0x00, 0x01, ...'
2. In setup() add 'webserver.on("/favicon.ico", handleFavicon);'
3. Above that add
`void ICACHE_FLASH_ATTR handleFavicon() {
  webserver.send_P(200, "image/x-icon", favicon, sizeof(favicon));
}`

maikonayres, 2018-01-23:
void handleFavicon() {
  server.send(404, "text/html; charset=iso-8859-1","<html><head> <title>404 Not Found</title></head><body><h1>Not Found</h1></body></html>");
}

SomeGuyNamedJay, 2018-01-28:
add this to your html. Change the url. <link rel="shortcut icon" type="image/png" href="http://eg.com/favicon.png"/>

rohit1269, 2018-06-19:
while((!(WiFi.status() == WL_CONNECTED)){
    delay(300);
    Serial.print(".");

  } 
my complier showing 
......................................... infinity why this loop is not completed

devyte, 2018-06-20:
@rohit1269 there is a know  issue with status() due to an issue in the glue code with lwip2. You can install latest git,and rebuild lwip2 from latest lwip2 master (i.e.: run one additional command after installing core from latest git), or you can wait until 2.4.2, which is currently targeted for August.