esp8266/Arduino #2081: softAPConfig IP Conflict

craigfoo, 2016-06-01:
I'm using a Huzzah ESP8266 and my code fires up the ESP8266 Access Point so I can connect to it via TCP to send Wifi credentials over. When I set the default IP with softAPConfig, the HTTP Client fails to work. This only happens on certain routers and it fails on the current router it's trying to connect to which has an IP range of 192.168.1.100 to .199 The default gateway is 192.168.1.254. 

When I set the softAPConfig IP to 192.168.1.1, the HTTP client fails to connect. If I set it to anything else; i.e. 192.168.0.1, the HTTP client works fine. Here is the test code to recreate the problem:

```
/**
 * BasicHTTPClient.ino
 *
 *  Created on: 24.05.2015
 *
 */

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

IPAddress apIP(192, 168, 1, 1);
IPAddress apGateway(192, 168, 1, 1);
IPAddress apSubmask(255, 255, 255, 0);

void setup() {

    USE_SERIAL.begin(115200);
    WiFi.mode(WIFI_AP_STA);
    WiFi.softAPConfig(apIP, apGateway, apSubmask);
    WiFi.softAP("ESP8266");

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

    WiFiMulti.addAP("[SSID]", "[PASS]");

}

void loop() {
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {

        Serial.print("Local IP Address: "); Serial.println(WiFi.localIP());
        Serial.print("SoftAP IP Address: "); Serial.println(WiFi.softAPIP());

        HTTPClient http;

        USE_SERIAL.print("[HTTP] begin...\n");
        http.begin("www.google.com", 80, "/"); 

        USE_SERIAL.print("[HTTP] GET...\n");
        // start connection and send HTTP header
        int httpCode = http.GET();
        if(httpCode) {
            // HTTP header has been send and Server response header has been handled
            USE_SERIAL.printf("[HTTP] GET... code: %d\n", httpCode);

            // file found at server
            if(httpCode == 200) {
                String payload = http.getString();
                USE_SERIAL.println(payload);
            }
        } else {
            USE_SERIAL.print("[HTTP] GET... failed, no connection or no HTTP server\n");
        }
    }

    delay(5000);
}
```

igrr, 2016-06-02:
The issue comes from the fact that LwIP network stack doesn't handle the case when two network interfaces have the same subnet IPs. In fact, I'm pretty sure that if you do the same on Linux, you will get the same result.
What LwIP does is, it walks the list of network interfaces, looking for an interface which has a route for the given IP address. This happens two times:
- when you resolve the domain name, and LwIP looks for the interface to use for talking to DNS server
- when the IP address is resolved, and you do an HTTP request, LwIP looks for the interface again

The list of interfaces is built in the order they are brought up. SoftAP interface goes up immediately, while STA interface goes up once connection to an AP is established. So the list of interfaces has SoftAP first, STA second. Normally the order of interfaces doesn't matter because they are in different subnets. 
But in this case SoftAP interface looks perfectly okay when LwIP tries to talk to DNS server, which of course doesn't work... DNS request packet is sent to SoftAP network, but DNS server is on STA network. So the domain name doesn't get resolved.

There is no good and robust solution for this issue other than to pick a really random IP address for your SoftAP interface.

A workaround, which I personally don't find really useful, but still... is to disable SoftAP interface while doing an HTTP request. You can do it at network interface level without affecting WiFi connection like this:

```c++
// add declarations of a few useful functions near the beginning of the sketch:
extern "C" void* eagle_lwip_getif(uint8_t);
extern "C" void netif_set_up(void*);
extern "C" void netif_set_down(void*);

// before doing HTTP request, disable SoftAP interface:
netif_set_down(eagle_lwip_getif(1));

// do http request and get the response

// now enable SoftAP interface again:
netif_set_up(eagle_lwip_getif(1));
```

See the source of [ip_route function](https://github.com/esp8266/Arduino/blob/master/tools/sdk/lwip/src/core/ipv4/ip.c#L123-L156) to see what is happening under the hood.

craigfoo, 2016-06-02:
@igrr, this makes sense, thanks for summarizing. I think my best option is to pick an IP address that a "private network" wouldn't sit on, https://en.wikipedia.org/wiki/Private_network.

closed 2016-06-03 by igrr