esp8266/Arduino #2085: ESP12 + "Minimal Hardware Setup for Bootloading and Usage" --> does not work

tuxedo0801, 2016-06-03:
### Basic Infos

#### Hardware
Hardware:			ESP-12E, plain module, minimal setup
Core Version:      	2.2.0

### Description

I have setup a minimal setup with ESP12 module, according to https://github.com/esp8266/Arduino/blob/master/doc/boards.md#minimal-hardware-setup-for-bootloading-and-usage

Probem is: reset mode "ck" is not able to write the sketch to the ESP12 module.

I also tried to use the nodemcu style by using two transistors (only had BC547 + 10k resistors) + reset mode "nodemcu".. Does also not work.
I tried with an FTDI232 Adapter that has of course DTR and RTS pins available.
OS is Debian Testing AMD64.

Exactly the same sketch with the same module settings works fine with a real nodemcu module. 

If I add two buttons with reset+flash to the ESP module and press them manually when doing the upload, it works.
So it's just the auto-program-mode that isn't working.

Power-supply is an regulated power supply (max 30V/2,5A) which is setup to 3.3V.

Are there any known traps that one has to pay attention for?
Are there any hints that I can follow to find the root-cause for this issue?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			QIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		tried both: ck / nodemcu

### Sketch

nothing special, just a simple "blink sketch"

igrr, 2016-06-03:
> Are there any hints that I can follow to find the root-cause for this issue?

- Attach second UART to ESPs TX pin, open a terminal at 74880 baud, and see what ESP says when it is being reset for programming.
- Attach a scope to GPIO0, RST, and RX of the ESP, see what's happening.

tuxedo0801, 2016-06-03:
2nd UART on same TX pin sounds like a good idea. I will check this.

I don't have a scope, but can attach a logic analyzer and capture the situation.

I will come back with more details later ...

tuxedo0801, 2016-06-03:
I captured the data with the logic analyzer (saleae compatible):

http://root1.de/public/esp8266.logicdata.zip (about 29MB, extracted about 200MB)

If I set the baud-rate in the analyzer to 115k2, I get frame errors. If I set it to 78440, things get readable (no frame errors), but I have no idea whats wrong...

What I can see: PC is sending data to ESP, but ESP is not answering at all.
No activitiy on DTR/GPIO0, just a few times some logic on RTS/Reset..

Would be great if someone can have a look at it.

tuxedo0801, 2016-06-03:
With connected reset/rts gpio0/dtr pins, bootloader says:

>  ets Jan  8 2013,rst cause:1, boot mode:(7,7)
> 
> waiting for host
>

tuxedo0801, 2016-06-03:
the boot-mode (7,7) was giving the the root-cause... GPIO15 was not on GND... Had a problem on my breadboard (broken contact).

Now upload is working as expected...

igrr, 2016-06-03:
That's the way to troubleshoot upload issues 👍

closed 2016-06-03 by igrr