esp8266/Arduino #2086: how to communicate two esp8266 each other with arduino

dkumariet, 2016-06-03:
### Basic Infos

#### Hardware
Hardware:			ESP-12
## Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512k(64k)
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload speed:		115200
Reset Method:		nodemcu

hi, 
i want to communication betweent two esp8266 module, "serial to wifi to wifi to serial" data send & recive both side,
 i try with some code but not done, only communicate with mobile browser 
any one help me?

igrr, 2016-06-03:
Please check this project, it does exactly what you need: https://github.com/jeelabs/esp-link

closed 2016-06-03 by igrr

dkumariet, 2016-06-03:
thanks for reply, 
but, i want to send data from hyperterminal to esp8266 to esp8266 to hyperterminal

E-systemsTechno, 2016-11-28:
Hi sir at present i am also doing same project, I think you completed the project "hyperterminal to esp8266 to esp8266 to hyperterminal" so can you please guide me to completew this project.