esp8266/Arduino #2089: Upload firmware program in ESP 12-E

Afaquecs1, 2016-06-03:
I want to upload program in esp8266mod (vendor AI thinkers) via Arduino IDE 1.6.8 with esp expansion(from github) in Windows 10. I have done following connection in my ESP 12-E. 
GND->GND
GIO0->GND
GIO15->GND
VS->Vcc(3.3V)
GIO2->Vcc
EN->Vcc
Tx->Rx
Rx->Tx
I am using USB to serial converter. When I upload my blink program to the ESP 12-E following error occurs 
Warning: Espcom_sync failed 
When I unplug the GIO2 pin the blue led blinks while uploading the program. But still show the same error. 
Please suggest me how to overcome thiproblemem
When I try to flash with ESPEasy. I got following message 
trying to connect
        flush start
        setting serial port timeouts to 1 ms
        setting serial port timeouts to 1000 ms
        flush complete
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        read 0, requested 1
warning: espcomm_sync failed
error: espcomm_open failed
Each time message is repeated the blue led blinks.

igrr, 2016-06-03:
Please see this comment for tips on how to troubleshoot upload issues:
https://github.com/esp8266/Arduino/issues/2085#issuecomment-223506059

devyte, 2017-10-17:
Closing per #3655 .