esp8266/Arduino #2090: WebUpdater problem

pauloeduardosp, 2016-06-03:
### Basic Infos

#### Hardware
Hardware:			ESP-07 and ESP-01
Core Version:      	

### Description

Problem description
I'm try to use webupdater sketch exemple, but didn't work.
After I do webupdate with WebUpdate.cpp.bin and to module didn't work and freeze with this information

After I do webupdate with WebUpdate.cpp.bin and to module didn't work and freeze with this information

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x0f
csum 0x0f

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB/64KB
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	44Mhz
Upload Using:		OTA
Reset Method:		

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char* host = "esp8266-webupdate";
const char* ssid = "anonymous2";
const char* password = "manuela.";

ESP8266WebServer server(80);
const char* serverIndex = "<form method='POST' action='/update' enctype='multipart/form-data'><input type='file' name='update'><input type='submit' value='Update'></form>";

void setup(void){
  Serial.begin(115200);
  Serial.println();
  Serial.println("Booting Sketch...");
  WiFi.mode(WIFI_AP_STA);
  WiFi.begin(ssid, password);
  if(WiFi.waitForConnectResult() == WL_CONNECTED){
    MDNS.begin(host);
    server.on("/", HTTP_GET, [](){
      server.sendHeader("Connection", "close");
      server.sendHeader("Access-Control-Allow-Origin", "*");
      server.send(200, "text/html", serverIndex);
    });
    server.on("/update", HTTP_POST, [](){
      server.sendHeader("Connection", "close");
      server.sendHeader("Access-Control-Allow-Origin", "*");
      server.send(200, "text/plain", (Update.hasError())?"FAIL":"OK");
      ESP.restart();
    },[](){
      HTTPUpload& upload = server.upload();
      if(upload.status == UPLOAD_FILE_START){
        Serial.setDebugOutput(true);
        WiFiUDP::stopAll();
        Serial.printf("Update: %s\n", upload.filename.c_str());
        uint32_t maxSketchSpace = (ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000;
        if(!Update.begin(maxSketchSpace)){//start with max available size
          Update.printError(Serial);
        }
      } else if(upload.status == UPLOAD_FILE_WRITE){
        if(Update.write(upload.buf, upload.currentSize) != upload.currentSize){
          Update.printError(Serial);
        }
      } else if(upload.status == UPLOAD_FILE_END){
        if(Update.end(true)){ //true to set the size to the current progress
          Serial.printf("Update Success: %u\nRebooting...\n", upload.totalSize);
        } else {
          Update.printError(Serial);
        }
        Serial.setDebugOutput(false);
      }
      yield();
    });
    server.begin();
    MDNS.addService("http", "tcp", 80);
  
    Serial.printf("Ready! Open http://%s.local in your browser\n", host);
  } else {
    Serial.println("WiFi Failed");
  }
}
 
void loop(void){
  server.handleClient();
  delay(1);
} 
### Debug Messages

```
messages here
```

me-no-dev, 2016-06-04:
does it freeze after the update is done? Also what happens if you reset the ESP manually after you update and then try again?

pauloeduardosp, 2016-06-04:
Yes, freeze after update. I make a manual reset because when they reset
after up load I was without serial, than a reset manually to get a msg.
I'll try again to check if a msg is the same

pauloeduardosp, 2016-06-05:
follow the entire message.
Don't have a error at serial after webupdate

Booting Sketch...
192.168.1.31
Ready! Open http://esp8266-webupdate.local in your browser
Update: webupdater.ino.bin
sleep disable
ERROR[0]: No Error
ERROR[0]: No Error
ERROR[0]: No Error
Update Success: 251760
Rebooting...
¢@*rjrA(!‹SËu‹ÐšvXKAú~‡êÑ*!Â½a€�þ"@ªrjrA(!‹SËu‹Ðšö›X%¨ý«Ðyêeh)"át‡mÝªNø

They freeze here, after a make a manual reset happen this

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
@cp:0
ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld
Fatal exception (28): 
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0003bff0, depc=0x00000000
Fatal exception (28): 
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0003bff0, depc=0x00000000
Fatal exception (28): 
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0003bff0, depc=0x00000000
Fatal exception (28): 
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0003bff0, depc=0x00000000
Fatal exception (28): 
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0003bff0, depc=0x00000000
Fatal exception (28): 
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0003bff0, depc=0x00000000
Fatal exception (28): 
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0003bff0, depc=0x00000000
Fatal exception (28): 
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0003bff0, depc=0x00000000

bakroistvan, 2016-11-21:
I am having the same exception: amica nodemcu with latest master||dev image from nodemcu-build.com
Right after flashing.

MoegerleStephan, 2017-03-21:
Having the same issue..
Are there any news?:)

MoegerleStephan, 2017-03-24:
Ok I think I've found the problem..
The issue occurs everytime the update failes and the user restart the update without rebooting the esp8266 in the mean time. Because the esp didn't reset the attributes eg startAdress, currentAddress, size,.. before beginning the update it will continue where it ended the last time when it failes.
Therefore line 44 in Updater.cpp should be removed. After this fixture the update starts once again from the beginning:
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Updater.cpp#L44

With this fix I didn't had any more freezing problems after update with one client. 
I didn't checked what happened if several clients try to update the esp at the same time. I guess this could also lead to unsightly problems..

devyte, 2017-10-17:
@pauloeduardosp is this issue still valid in latest git? I can't tell which line you mean by line 44.

MoegerleStephan, 2017-10-17:
The path changed now to line 45.
Please remove this return command. 
Otherwise if a firmware upload fails and you upload this file once again you are not able to update the esp correctly.

earlephilhower, 2019-01-16:
Summary of the bug:
Updater::begin() checks that `_start==0`, but if you had a failed upload before, _start would be the address of the last program attempted to upload.  So after one failed upload, you need to reset the ESP to try another upload.

Solution A:
* Remove the check that start==0 in ::begin, because the structure is __reset() to all 0s later on in the method.

Solution B:
* Keep the check, but on any failure in the method call __reset() before `return false;`.

I like sol'n B.

PR #6325 (2019-07-21): Clear updater state on any error
Fixes #2090

The Updater checks that an update isn't already in progress on ::begin,
but when an error happens in the middle of an upload it's impossible to
actually reset this flag w/o a reboot.

Reset the state members (esp. _size) on any error condition so
that you can restart the transfer with a new ::begin.  Any error
condition is fatal, anyway, so there is no reason not to clear the
current state at that point.
commit 5d5cd1d4 (2019-07-21): Clear updater state on any error (#6325)

Fixes #2090

The Updater checks that an update isn't already in progress on ::begin,
but when an error happens in the middle of an upload it's impossible to
actually reset this flag w/o a reboot.

Reset the state members (esp. _size) on any error condition so
that you can restart the transfer with a new ::begin.  Any error
condition is fatal, anyway, so there is no reason not to clear the
current state at that point.
closed 2019-07-21 by earlephilhower