esp8266/Arduino #2092: GDBStub not usable

boguslawb, 2016-06-04:
Can you prepare a tutorial (video of actual working task would be ideal) how exactly to use GDBStub to catch and backtrace exception ? I followed your description and found the same issues as here http://blog.squix.org/2016/04/esp8266-offline-debugging-with.html
gdb  cannot attach to remote process, if I try directly attach to redirected I got invalid packet error in gdb, if I try to add line  ((int*)0) = 0 then I cannot run device - it stuck on reset

tijnkooijmans, 2016-06-05:
Same issue here, this is my output from xtensa-lx106-elf-gdb: 

Reading symbols from /Users/tijn/Develop/minibrew-firmware/ESP8266/MiniBrewWifi/Builds/embeddedcomputing.elf...done.
Remote debugging using :9980
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response

earlephilhower, 2019-01-08:
This is done in PR #5559 where we have a new GDBStub and a writeup in the documentation directory.  We'll track progress in that pull.

closed 2019-01-08 by earlephilhower