esp8266/Arduino #2095: boot mode always (7,0) ??

tuxedo0801, 2016-06-04:
Hi,

this issue is related to #2085 ...

On my breadboard, programming is now working (see linked issue).

Now I have soldered a ESP12 module to a custom PCB, which reflects the same schematic part as described here: https://github.com/esp8266/Arduino/blob/master/doc/boards.md#minimal-hardware-setup-for-bootloading-and-usage

I check every wire three times... It's not working as in my breadboard.

I have an FTDi232 adapter attached, with RTS + DSR connected.
If a do a board-reset by hand, the bootloader output shows:

>  ets Jan  8 2013,rst cause:2, boot mode:(7,0)
> 
> waiting for host
> 

So, with my above linked issue, I also had boot mode = 7... so GPIO15 is not on GND. On my breadboard I had a connection issue on the GND-line... fixed it --> worked.

On my PCB all wires are correct. I'm using the same type of resistors and the same power-source.
While I measure 0,0V on GPIO15 on my breadboard, the identical setup on my PCB shows about 0,5V on GPIO15...

I'm running out of ideas what could this 0,5V on GPIO15 ... Vcc is exactly 3.3V and the used resistor is 10kOhm (desoldered, measured 9,98kOhm, resoldered).... My current guess is: ESP12 module is broken?!

Any ideas? As I said: The PCB is, regarding the programming-circuit part the same as described in the "mini hardware setup" from the link above. 

More ESP12 modules are ordered, but still on delivery... So, if someone has a hint on what else I could check... I would be pleased. Otherwise I have to wait for my order to replace the ESP12 module.

tuxedo0801, 2016-06-04:
Hmm,

I replaced the 10k resistor between GPIO15 and GND with a wire, to get GPIO15"hard to GND", and bootloader now says?!

>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x40100000, len 1396, room 16 
> tail 4
> chksum 0x89
> load 0x3ffe8000, len 776, room 4 
> tail 4
> chksum 0xe8
> load 0x3ffe8308, len 540, room 4 
> tail 8
> chksum 0xc0
> csum 0xc0
> 
> 2nd boot version : 1.4(b1)
>   SPI Speed      : 40MHz
>   SPI Mode       : DIO
>   SPI Flash Size & Map: 8Mbit(512KB+512KB)
> jump to run user1 @ 1000
>

tuxedo0801, 2016-06-04:
Flashing the ESP is now possible, and sketch seems to work. But boot-loader output is still confusing.

tuxedo0801, 2016-06-04:
And also confusing:

> Directly connecting a pin to VCC or GND is not a substitute for a PullUp or PullDown resistor, doing this can break upload management and the serial console, instability has also been noted in some cases.

In my case pulldown brouht me 0.5V instead of 0V ... 
Is there any recommendation that works in ANY case?! or is it safe to have GPIO15 hard-wired to GND?

devyte, 2017-10-17:
Closing per #3655 .

closed 2017-10-17 by devyte