esp8266/Arduino #2097: WiFiClientSecure and WiFi disconnects

tijnkooijmans, 2016-06-05:
Hello Ivan,

I'm experiencing crashes with WiFiClientSecure whenever my WiFi disconnects. Basically any use of the client object will crash the ESP.

When I try to delete the client I get this Exception (28) immediately:

0x402203da: ax_port_write at /Users/tijn/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:541
0x40100700: free at /Users/tijn/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/cores/esp8266/umm_malloc/umm_malloc.c:1733
0x4024dc1c: send_raw_packet at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c:990
 (inlined by) send_packet at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c:1123
0x40218380: Print::write(unsigned char const*, unsigned int) at /Users/tijn/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/cores/esp8266/Print.cpp:38
0x4024e112: DISPLAY_ALERT at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c:2174 (discriminator 3)
 (inlined by) send_alert at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c:1610 (discriminator 3)
0x40218524: Print::println() at /Users/tijn/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/cores/esp8266/Print.cpp:140
0x40218590: Print::println(char const*) at /Users/tijn/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/cores/esp8266/Print.cpp:152
0x4024e16e: ssl_free at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c:236
0x40220666: ~SSLContext at /Users/tijn/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:77
 (inlined by) SSLContext::unref() at /Users/tijn/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:92
0x4021fd74: WiFiClientSecure::~WiFiClientSecure() at /Users/tijn/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:217
0x4021fdab: WiFiClientSecure::~WiFiClientSecure() at /Users/tijn/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:221

When I don't delete it any use of client.available() will Exception (28) it:

0x4022034b: ax_port_read at /Users/tijn/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:524
0x40104731: ppFetchTxQFirstAvail at ??:?
0x4024e277: basic_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c:1231
0x402328ae: ets_vsnprintf at ??:?
0x40218358: Print::write(unsigned char const*, unsigned int) at /Users/tijn/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/cores/esp8266/Print.cpp:38
0x4024e6bc: ssl_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c:265
0x4022068c: SSLContext::_readAll() at /Users/tijn/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:184
0x402206cd: SSLContext::available() at /Users/tijn/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:164

I am using the WiFi.onEvent handler to handle WiFi disconnects, but I'm not managing to find a way to gracefully stop the client. I also tried client.stop() and client.flush(), but keep seeing the same crashes.

Please let me know if you need more details.

Thanks! 
Tijn

igrr, 2016-06-05:
Good point! Will fix that.

tijnkooijmans, 2016-06-05:
Great, thanks a lot!

igrr, 2016-06-12:
Created #2139 which addresses this issue.
@tijnkooijmans could you please try this change? I have tested it with HTTPS client and MQTT client, but there may be some failure mode which I didn't check.

tijnkooijmans, 2016-06-13:
I'm trying to compile this branch, but unfortunately the linker doesn't accept the use of <list> in ESP8266WiFiGeneric. Any idea how to fix this?

tijnkooijmans, 2016-06-13:
These are all the linker flags I use: xtensa-lx106-elf-gcc -Os -w -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -lm -lgcc -lhal -lphy -lnet80211 -llwip -lwpa -lmain -lpp -lsmartconfig -lwps -lcrypto -laxtls -Wl,--end-group -LBuilds

igrr, 2016-06-13:
It should work if you use arduino-builder. Are you using something else to compile your code?

igrr, 2016-06-13:
I've rebased this on master, now it should work with makeEspArduino as well, if you add `-lstdc++` to the list of libraries.

tijnkooijmans, 2016-06-13:
yes I'm using embedxcode, since it's a huge project I need a powerful IDE. maybe I should migrate to platform IO though.

tijnkooijmans, 2016-06-13:
Ok added that linkerflag worked. And the good news is that it doesn't crash anymore on wifi-reconnects, so great job!

closed 2016-06-23 by igrr