esp8266/Arduino #2102: Review/remove usages of Access-Control-Allow-Origin in libraries and examples

igrr, 2016-06-06:
There are a bunch of [instances](https://github.com/esp8266/Arduino/search?utf8=%E2%9C%93&q=Access-Control-Allow-Origin) of `Access-Control-Allow-Origin` header in libraries and examples. The ones which are in libraries are hidden form sight and are potential security issues.

Need to review the uses of this header and document, remove, or make them optional as appropriate.

me-no-dev, 2016-06-06:
Appart from the SSDP lib, for which I am not positive, the rest can go for sure. I believe some of those were added in testing of some scenarios back then, when we were trying to get all connections closed at the end of the response. The rest are probably copy/paste

yoursunny, 2018-11-18:
As of commit 2f4380777ef3f1a32a7d539371d910820d0c6f6a, the only occurrence of "Access-Control-Allow-Origin" is in SSDP library. Thus, this issue should be closed.

devyte, 2018-11-18:
Ok, closing per previous comment.

closed 2018-11-18 by devyte