esp8266/Arduino #2109: not able to create webserver using esp8266MOD AI-thinker

pankaj031196, 2016-06-07:
hey...please help me
i am struggling with my above mentioned wifi module from last 4-5 days. the sketch can be uploaded to it...and the blink sketch is working on it...but when i tried webserver or any other codes like the one to show its IP address....or use at commands.....it doesnt respond.....AT commands are not working.....when i write AT in serial moniter and press enter....it doesnt comes in the output area and just vanishes.....output area of serial moniter doesnt show anything...sketches can be uploaded but serial moniter doesnt respond
my connections are..
vcc-3.3v
gnd-gnd
rx-tx
tx-rx
ch_pd-high
gpio-gnd(ground)

please help someone

WereCatf, 2016-06-07:
If you flash new firmware on it with this software then there is no AT-firmware anymore, so of course AT-commands won't work either -- you have replace the firmware with your own!

You have to write your own code to handle input from serial-port or roll back to AT-firmware.

pankaj031196, 2016-06-11:
sir i am very novice....can u please guide me step by step what to do now ?

pankaj031196, 2016-06-11:
i could not g

closed 2016-06-11 by pankaj031196