esp8266/Arduino #2111: Problems with ESP12f without WiFi on 50mA power source

tuxedo0801, 2016-06-07:
### Basic Infos

#### Hardware
Hardware:			ESP-12F
Core Version:      	2.2.0

### Description

I want to drive my ESP12F module from a 50mA@3.3V source. For this application I don't need WiFi at all. So I disabled it.
If I measure the current consumption with my multimeter, I get about 14mA@3.3V. That would be okay for my power source.
But as soon as I connect the ESP to the meant power source, the 3.3V line drops to 1.8V and I can measure a current consumption of about 30..35mA. The blue LED on the ESP12F starts to flicker continuously.
My current theory is:
The ESP8266 has a too high starup current consumption my power source cannot handle. I added an 100µF capacitor, but this does not solve the issue.
I don't have a scope to measure the exact startup current. 

Question is:
* Is my code okay (wrt. disabling WiFi completely)? Of course the consumtion I can measure is just about 14mA, but maybe there is a "faster way" of disabling it to reduce startup current consumption?!
* Does anybody know about the startup power consumption without wifi or can do a measurement with a scope?!
* Are there known issues with this scenario (no wifi, low current power source)?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp

#include <ESP8266WiFi.h>

void setup() {
    // turn off wifi completely   
    WiFi.disconnect(); 
    WiFi.mode(WIFI_OFF);
    WiFi.forceSleepBegin();
    delay(10);
    
    Serial.begin(19200);  

}

void loop() {
   
  Serial.println("HIGH");
  
  delay(3000);
  
  Serial.println("LOW");

  delay(3000);
  
} 
```

igrr, 2016-06-07:
Can you try adding `RF_MODE(RF_DISABLED);` line between `include` and `void setup`?
After that calling WiFi.xxx functions should not be needed.
Note that this requires a pretty recent version, like staging 2.3.0-rc1 or git version.

tuxedo0801, 2016-06-07:
Thanks.I will test it this evening. 

But I'm still interested in a scope-screenshot showing the current consumption on startup. So If someone can help ...

igrr, 2016-06-07:
Bummer, `RF_MODE(RF_DISABLED);` works only after deep sleep wakeup, not after normal power-up.
I've got some scope pics, will post a bit later.

igrr, 2016-06-07:
Some scope pics, scale 100 mV = 40 mA.

Test 1: Your sketch. You can see that RF calibration is causing large current spikes
[image]

krzychb, 2016-06-07:
@igrr,
Do you mean one division on the screen = 40mA and max spikes = 80mA?

igrr, 2016-06-07:
I have updated my previous comment with some numbers.
The mentioned hack with an extra hook in `user_init` can make its way into the core, i guess. Would be useful to turn off wifi depending on certain conditions.

tuxedo0801, 2016-06-07:
Thanks for the screenshots. I will test it and report back.

>can make its way into the core, i guess

What does this mean? How would one disable wifi in future with this change?

tuxedo0801, 2016-06-07:
100mA for 0,8sec... still too much. tested 100µF, 200µF and 1000µF.

Isn't there a way to start with wifi disabled instead of start with and then disable it?

tuxedo0801, 2016-06-07:
200µF gives me on console:

>  ets Jan  8 2013,rst cause:1, boot mode:(3,6)
> 
> load 0x4010f000, len 1264, room 16 
>

igrr, 2016-06-08:
Looks like you aren't reaching the point when ROM bootloader finishes reading second stage bootloader from flash. How much of a voltage drop does your power supply have at 50mA?

tuxedo0801, 2016-06-08:
It's not a "real" power supply.
The ESP12 is attached to a "KNX Transceiver" which is connected to the (wired) KNX home automation bus. The KNX transceiver IC (www.onsemi.com/pub_link/Collateral/NCN5120-D.PDF) is able to provide about 40..50mA at 3.3V.
There is no spec/figure/graph showing the voltage drop over current consumption. I can try to measure the voltage drop when applying a big enough resistor to simulate constant 50mA current consumption.

tuxedo0801, 2016-06-12:
Good news... with all your suggestes changes (test1+2+3) in addition to an 470µF capacitor on Vcc, ESP now now able to start with my luw current power source...

Thanks a lot.

tuxedo0801, 2016-06-12:
btw: 

Would be great if we could add those code-changes to the next release?!

@igrr 
Should I make a PR or do you add those changes yourself?

igrr, 2016-06-12:
Awesome, glad that this worked for you.
Created #2133 to track changes necessary in the core.

closed 2016-06-12 by igrr

wb8wka, 2017-07-10:
Is there a reason RF_MODE(RF_NO_CAL);  would not be working?  2.3.0 with a Wemos D1 mini.  All I can do is either turn the RF off or RF_CAL is on. with with the call to RF_NO_CAL

MrD2, 2018-02-07:
This reduction of power consumption doesn't seem to work on 2.4.0-rc2

using user_init()