esp8266/Arduino #2112: ESP-LAUNCHER

finoconv, 2016-06-07:
### Basic Infos

#### Hardware
Hardware:			ESP-LAUNCHER
Core Version:      	?2.1.0-rc2? (I don't know)

### Description

Problem description
Hi I'm new to develop on ESP8266.
I'm trying to upload an empty FW to validate the communication between ESP8266 and PC.
I'm using the ESP-LAUNCHER that uses the FTDI USB transceiver

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB? (I don't know what is the correct setting)
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu? (I don't know what is the correct setting)

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here

Lo sketch usa 219.263 byte (50%) dello spazio disponibile per i programmi. Il massimo è 434.160 byte.
Le variabili globali usano 31.164 byte (38%) di memoria dinamica, lasciando altri 50.756 byte liberi per le variabili locali. Il massimo è 81.920 byte.
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

How can I fix this problem?
![immagine](https://cloud.githubusercontent.com/assets/19795038/15852218/aa0487c4-2ca0-11e6-871a-7a667f418d18.png)

igrr, 2016-06-23:
Are you sure the port is COM4? I just tried connecting ESP-LAUNCHER to a Windows computer and it ended up on port COM16, whereas port COM4 was some built-in serial port.

finoconv, 2016-06-23:
Probably the error was due to the device configuration. When I attempt to program the device I should clear the IO0 by a user switch. In this way the device works well

closed 2016-06-23 by finoconv

DrewOJensen, 2016-07-21:
@finoconv I just got the launcher. Could you give a screen shot of your Arduino IDE setting along with a walk through of how you do a sketch upload?

Thanks!

tranquangvu0651, 2016-10-06:
Just easy ! pull-down to Gnd pin GPIO0 using a 330 Ohm resistor, 

It's Work ! :+1: