esp8266/Arduino #2114: RF Module (RFM83) and ESP 8266 could it work?

Banzai00, 2016-06-07:
I have a RF module RFM83, i want it to work with ESP8266 Wroom 02.
From what i've seen in the internet it works with Arduino and i've found RadioHead library that provides an example for esp 12. (http://www.airspayce.com/mikem/arduino/RadioHead/classRH__ASK.html#details)
SO, my questions :  
Could the RFM83C work with Wroom 02 ??
For Arduino the data pin of the receiver is always connected to D11. For the Wroom 02, do i connect data pin of the receiver to specific pin or any I/O can do the work ???

supersjimmie, 2016-06-08:
The page you mention says it all:

> ESP8266
> This module has been tested with the ESP8266 using an ESP-12 on a breakout board ESP-12E SMD Adaptor Board with Power Regulator from tronixlabs http://tronixlabs.com.au/wireless/esp8266/esp8266-esp-12e-smd-adaptor-board-with-power-regulator-australia/ compiled on Arduino 1.6.5 and the ESP8266 support 2.0 installed with Board Manager. CAUTION: do not use pin 11 for IO with this chip: it will cause the sketch to hang. Instead use constructor arguments to configure different pins, eg:
*  RH_ASK driver(2000, 2, 4, 5);
*  
Which will initialise the driver at 2000 bps, recieve on GPIO2, transmit on GPIO4, PTT on GPIO5. Caution: on the tronixlabs breakout board, pins 4 and 5 may be labelled vice-versa.

Banzai00, 2016-06-08:
I know my friend but why use 3 pins when i only need one ?? (i only need the receiver to work with ESP 8266, the transmitter is a remote) and can i use another pin to receive ?? because for the Wroom 02 GPIO2 is always connected to 3.3V.

supersjimmie, 2016-06-09:
If I look at the rh_ask library, you will need to fill in all 4 parameters, so 3 pins and if you don't there are (unwanted) defaults: 
[RadionHead RH_ASK library](https://github.com/PaulStoffregen/RadioHead/blob/master/RH_ASK.h#L183)
You can try to change the library (both in the .h as in the .cpp) to make it possible to make it only use the pins that are filled in. Offcourse that includes every next step in the source.

Banzai00, 2016-06-11:
supersjimmie when i try to upload the example of ask-receiver in Radiohead Library. I get this compiling error.
`ESP8266 SPI support not yet available. Please feel free to submit a patch`

So it seems that it's impossible to use RFM83 with wroom 02.

supersjimmie, 2016-06-13:
I believe that's not an error but a warning?
[See Warning here](https://github.com/Yveaux/RadioHead/blob/master/RadioHead/RHHardwareSPI.cpp#L282-L283)
But are you using SPI then?  (On an esp8266 the SPI pins are 'hardcoded' to pins 12-15.)

Banzai00, 2016-06-14:
Thank you supersjimmie for your time.
This is the code i'm trying to implement in the Wroom 02
`// ask_receiver.pde
// -*- mode: C++ -*-
// Simple example of how to use RadioHead to receive messages
// with a simple ASK transmitter in a very simple way.
// Implements a simplex (one-way) receiver with an Rx-B1 module

#include <RH_ASK.h>
#include <SPI.h> // Not actualy used but needed to compile

//RH_ASK driver;
 RH_ASK driver(2000, 12, 4, 5); // ESP8266: do not use pin 11

void setup()
{
    Serial.begin(115200);	// Debugging only
    if (!driver.init())
         Serial.println("init failed");
}

void loop()
{
    uint8_t buf[RH_ASK_MAX_MESSAGE_LEN];
    uint8_t buflen = sizeof(buf);

    if (driver.recv(buf, &buflen)) // Non-blocking
    {
	int i;

	// Message with a good checksum received, dump it.
	driver.printBuffer("Got:", buf, buflen);
    }
}`

In this example they say that SPI is not really needed but it's only there to make the compilation possible. But when i try to compile get the error compiling that i told you about : 
`Arduino: 1.6.5 (Windows 7), Board: "Generic ESP8266 Module, Serial, 160 MHz, 80MHz, QIO, 115200, 4M (1M SPIFFS), nodemcu, Disabled, None"

C:\Users\Admin\Desktop\arduino-1.6.5-r5\libraries\RadioHead\RHHardwareSPI.cpp:283:2: error: invalid preprocessing directive #warn
 #warn "ESP8266 SPI support not yet available. Please feel free to submit a patch"
  ^
Error compiling.`

I tried to decrypt the signal send by the data pin. I managed to do that but it's not stable enough sometimes it works sometimes not ( because i'm reading pulses in microseconds ).

Thank again my friend for your help and could you give me an idea or an advise so i can make this code works which will make easy for me to get the RFM83 and the Wroom 02 to work together.

supersjimmie, 2016-06-14:
ah, :-) Just remove that line, the compiler does not like ```#warn ... ``` at all.
(open file RHHardwareSPI.cpp and remove line 283, or put // in front of it)

hallard, 2016-06-24:
Aa far as I remember, RadioHead need some tweak on SPI and Interrupts to work on ESP8266, I've done this when I was working on RFM69 module. I would love doing Pull Request for this, but only one day if official radiohead repo is released on github and authorized pubic community doing PR.

Banzai00, 2016-06-26:
yes, hallard it seems that RadioHead Library as it's now doesn't work with ESP8266. I hope that some day they make the modifications needed.

devyte, 2017-10-17:
Closing per #3655 .

closed 2017-10-17 by devyte