esp8266/Arduino #2115: wifiMulti (or WiFi.printDiag) not handling passphrase correctly on unprotected APs

SteveToulouse, 2016-06-07:
### Basic Infos

#### Hardware
Hardware:			NodeMCU 1.0 Amica ESP-12E
Core Version:      	2.2.0

IDE Arduino             1.6.8

### Description

Either `wifiMulti` or `WiFi.printDiag` are setting or printing the wrong  (but not random) value for passphrase on an open (unprotected) AP.

I added a series of 4 APs (see code) of which the last is an open AP. When connected to that AP (`wifiMulti.run`) the WiFi.printDiag prints out the passphrase of the last previously-added AP instead of nothing/empty string. In the example below it printed "passphrase3", but after inverting the adding order, it always prints the passphrase of the preceding AP by order of `addAP`.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4M (3M SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			n/a
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		n/a

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

ESP8266WiFiMulti wifiMulti;

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.setDebugOutput(true);

  wifiMulti.addAP("Stevebox", "passphrase1"); // my ADSL gateway router
  wifiMulti.addAP("AndroidAP", "passphrase2"); // my android phone
  wifiMulti.addAP("SteveAP", "passphrase3"); // my secondary wifi router
  wifiMulti.addAP("FreeWifi"); // a neighbour's ADSL gateway router with public AP
//  wifiMulti.addAP("FreeWifi","");   has the same effect
//  wifiMulti.addAP("FreeWifi"," ");   works (passphrase is printed as length 1)
}

void loop() {
  Serial.println();
  if (int stat = wifiMulti.run() != WL_CONNECTED) {
    Serial.print("WiFi not connected! run returned: ");
    Serial.println(stat);
  }

  WiFi.printDiag(Serial);
  delay(5000);
}
```

### Debug Messages

```
scandone

f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1

connected with FreeWifi, channel 2
dhcp client start...
cnt 
ip:10.1.125.162,mask:255.248.0.0,gw:10.7.255.254
Mode: STA
PHY mode: N
Channel: 2
AP id: 0
Status: 5
Auto connect: 1
SSID (8): FreeWifi
Passphrase (11): passphrase3
BSSID set: 1
pm open,type:2 0
```

SteveToulouse, 2016-06-24:
Hi again,

I took a look at the code in ESP8266WiFiMulti.h and .cpp and I'm wondering if this is not a toolchain problem ? I'm not proficient enough to test the hypothesis myself, unfortunately:

In the .h, we have:

`        bool addAP(const char* ssid, const char *passphrase = NULL);
`

If I understand correctly how this is supposed to work, it should put a NULL pointer (value = 0) into the space reserved on the stack for the passphrase parameter, so that if no passphrase parameter is provided in the call, addAP won't receive a meaningless pointer.

But what seems to be happening is that

`      ESP8266WiFiMulti.addAP(mySSID);
`
is receiving the pointer which it just happens to find on the stack at the place expected for passphrase. With consecutive calls to addAP, it will find the previous occurrence of "passphrase".

This doesn't explain how passing an empty string as the passphrase should exhibit the same behaviour.

In the .cpp, this test at line 187 should protect us from the absent passphrase parameter (although we are depending on the compiler not to try to evaluate after the && if passphrase is NULL):

`       if(passphrase && *passphrase != 0x00) {
`

but apparently not. So what *is* addAP receiving ?

igrr, 2016-06-24:
Toolchain works fine, the bug is in https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp#L187-L194
As you can see `passphrase` field of newAP structure only gets set if passphrase is not null and not an empty string. Otherwise `newAP.passphrase` is not initialized (i.e. contains whatever garbage happens to be on the stack). Unsurprisingly, if you call `addAP` two times in a row, `newAP` structure will end up in the same place on the stack, so `newAP.passphrase` will have the value from previous call.

SteveToulouse, 2016-06-24:
Ahhhh yes! Should have been obvious - I'll have to take a beginners programming course again ;-)

commit 86067333 (2016-07-08): Fix2115 (#2244)

* Update ESP8266WiFiMulti.cpp

Fix #2115

* Update ESP8266WiFiMulti.cpp

Fix #2115. Cleaner version (catch strdup("") fail).
closed 2018-01-02 by igrr